Koenkk/zigbee2mqtt #9546: zigbee2mqtt doesn't start with hassio

iamkarlson, 2021-11-07:
## What happened
z2m doesn't start with zzh stick: (Error: SRSP - SYS - ping after 6000ms) or diffirent similar error. logs: https://pastebin.com/ztr9yb4c

## What did you expect to happen
z2m works as expected

## How to reproduce it (minimal and precise): get new zzh stick, flash firmware, run z2m

## Debug info
Zigbee2MQTT version: 1.22.0-2
Adapter hardware: CC2652R1 (zzh)
Adapter firmware version: [CC2652R_coordinator_20210708](https://github.com/Koenkk/Z-Stack-firmware/blob/afd5ac4666732a7822611cbacef9ed571af99e01/coordinator/Z-Stack_3.x.0/bin/CC2652R_coordinator_20210708.zip)
homeassistant:
```Core
Version
core-2021.11.1```
hassio:
```
Supervisor
Version
supervisor-2021.10.8
```

iamkarlson, 2021-11-07:
another logs where another step is failing but visible that stick is being detected with correct firmware: https://pastebin.com/z5ktTxbm

tomdudu38, 2021-11-08:
hi, did you read that:
https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#error-srsp-sys-ping-after-6000ms

iamkarlson, 2021-11-08:
@tomdudu38 I just got a message from electrolama guys and they said it might be an error associated with kernel. here's my kernel version:
```
$ uname -a
Linux chaos 5.10.0-9-amd64 #1 SMP Debian 5.10.70-1 (2021-09-30) x86_64 GNU/Linux
```

tomdudu38, 2021-11-12:
i think you should begun to check if you have the good port configured.

ls -l /dev/serial/by-id
and:
test -w [PORT] && echo success || echo failure
and check the right on the port.

(read the link i gave you befor)
and you should check if you have "priviliged mode" AND the good usb port
 setting for your container.

i already see zigbee2mqtt, thinking my adapter was pluged but, it was another usb device....

you send a link... that talk about: " the zzh. This causes strange behavior, e.g. delay when receiving messages from devices."
it doen't seems to look like your problem

tomdudu38, 2021-11-12:
if you unplug the key the port may change so you have to configure it by-id ...

iamkarlson, 2021-12-09:
Hi @tomdudu38 I check it all you advised me to just before I created the issue and it yet didn't work

iamkarlson, 2022-01-09:
It's not stale.

closed 2022-02-17 by github-actions[bot]