Koenkk/zigbee2mqtt #9551: No Action sent After OTA Upgrade of Tradfri Shourcat butto E1812 to firmware 2.3.080

Phelton, 2021-11-07:
Hello everyone,
i have launched the OTA Upgrade on one of my Tradfri Shortcut button to last firmware (2.3.080) ... after some hours of wait the process was terminated, but after that the button was stop of work.
Now when i try to click nothing happens.
I have tried to remove e re-pair the button many times but the situation doesn't change.

I´m using:
- Home Assistant OS 6.6
- Zigbee2mqtt 1.22.0-2
- Mosquitto broker 6.0.1

I have tried to pair a new Tradfri button from my stash and a new button with previous firmware (2.3.015) work perfectly... only button with new firmware (2.3.080) doesn't work.

Sorry but i don´t expert and don´t known what other information i can give for help and where i can find that information.
If i can execute some debug on my system please tell me how and i will happy to retrive all information that is necessary for understand because the button stopped working after the update.

thanks a lot.
bye from Italy

P.S. Sorry for my bad English.

malaakso, 2021-11-08:
See #7716 . In short

1. press the small button four times to reset the switch
2. restart zigbee2mqtt
3. re-join the button by enabling "permit join" and pressing the small button four times

Now actions should be again visible. If some clusters of the button are bound to a device (e.g. lamp), you also need to bind them to the coordinator to get the actions to appear.

Phelton, 2021-11-08:
thanks for your reply.
I'm tried many times the procedure that you have explaned, but nothing is changed.. Action reamain in state "None" .
I can see all information but don't change nothing when i click button.
I don't known if is important, but with thi nuwe firmware ikea introduce double click mode at plus to the one click and the hold click actions.
The interview was completed... but don't function nothing

[image]
[image]

I'm tried also to re-pair in proximity of dongle ZigBee for improve quality of link, but nothing is changed.
Other possible solution?

thanks again for your help.
bye

Phelton, 2021-11-08:
of course..
following, wh all happens when i  bind the button:

```
Info Device '0x680ae2fffe322a6c' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x680ae2fffe322a6c","ieee_address":"0x680ae2fffe322a6c"},"type":"device_joined"}'
Info Configuring '0x680ae2fffe322a6c'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x680ae2fffe322a6c"},"type":"device_connected"}'
Info MQTT publish: topic 'homeassistant/sensor/0x680ae2fffe322a6c/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x680ae2fffe322a6c"],"manufacturer":"IKEA","model":"TRADFRI shortcut button (E1812)","name":"0x680ae2fffe322a6c","sw_version":"Zigbee2MQTT 1.22.0"},"device_class":"battery","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x680ae2fffe322a6c","name":"0x680ae2fffe322a6c battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x680ae2fffe322a6c","unique_id":"0x680ae2fffe322a6c_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x680ae2fffe322a6c/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x680ae2fffe322a6c"],"manufacturer":"IKEA","model":"TRADFRI shortcut button (E1812)","name":"0x680ae2fffe322a6c","sw_version":"Zigbee2MQTT 1.22.0"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","json_attributes_topic":"zigbee2mqtt/0x680ae2fffe322a6c","name":"0x680ae2fffe322a6c action","state_topic":"zigbee2mqtt/0x680ae2fffe322a6c","unique_id":"0x680ae2fffe322a6c_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x680ae2fffe322a6c/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x680ae2fffe322a6c"],"manufacturer":"IKEA","model":"TRADFRI shortcut button (E1812)","name":"0x680ae2fffe322a6c","sw_version":"Zigbee2MQTT 1.22.0"},"enabled_by_default":false,"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x680ae2fffe322a6c","name":"0x680ae2fffe322a6c linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x680ae2fffe322a6c","unique_id":"0x680ae2fffe322a6c_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x680ae2fffe322a6c/update_state/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x680ae2fffe322a6c"],"manufacturer":"IKEA","model":"TRADFRI shortcut button (E1812)","name":"0x680ae2fffe322a6c","sw_version":"Zigbee2MQTT 1.22.0"},"enabled_by_default":false,"icon":"mdi:update","json_attributes_topic":"zigbee2mqtt/0x680ae2fffe322a6c","name":"0x680ae2fffe322a6c update state","state_topic":"zigbee2mqtt/0x680ae2fffe322a6c","unique_id":"0x680ae2fffe322a6c_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
Info MQTT publish: topic 'homeassistant/binary_sensor/0x680ae2fffe322a6c/update_available/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x680ae2fffe322a6c"],"manufacturer":"IKEA","model":"TRADFRI shortcut button (E1812)","name":"0x680ae2fffe322a6c","sw_version":"Zigbee2MQTT 1.22.0"},"enabled_by_default":false,"json_attributes_topic":"zigbee2mqtt/0x680ae2fffe322a6c","name":"0x680ae2fffe322a6c update available","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x680ae2fffe322a6c","unique_id":"0x680ae2fffe322a6c_update_available_zigbee2mqtt","value_template":"{{ value_json.update_available}}"}'
Debug Received MQTT message on 'homeassistant/sensor/0x680ae2fffe322a6c/battery/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x680ae2fffe322a6c"],"manufacturer":"IKEA","model":"TRADFRI shortcut button (E1812)","name":"0x680ae2fffe322a6c","sw_version":"Zigbee2MQTT 1.22.0"},"device_class":"battery","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x680ae2fffe322a6c","name":"0x680ae2fffe322a6c battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x680ae2fffe322a6c","unique_id":"0x680ae2fffe322a6c_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Debug Received MQTT message on 'homeassistant/sensor/0x680ae2fffe322a6c/action/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x680ae2fffe322a6c"],"manufacturer":"IKEA","model":"TRADFRI shortcut button (E1812)","name":"0x680ae2fffe322a6c","sw_version":"Zigbee2MQTT 1.22.0"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","json_attributes_topic":"zigbee2mqtt/0x680ae2fffe322a6c","name":"0x680ae2fffe322a6c action","state_topic":"zigbee2mqtt/0x680ae2fffe322a6c","unique_id":"0x680ae2fffe322a6c_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
Debug Received MQTT message on 'homeassistant/sensor/0x680ae2fffe322a6c/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x680ae2fffe322a6c"],"manufacturer":"IKEA","model":"TRADFRI shortcut button (E1812)","name":"0x680ae2fffe322a6c","sw_version":"Zigbee2MQTT 1.22.0"},"enabled_by_default":false,"icon":"mdi:signal","json_attributes_top [... cut ...]
Debug Device '0x680ae2fffe322a6c' requested OTA
Debug Received Zigbee message from '0x680ae2fffe322a6c', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":587728433,"imageType":4550,"manufacturerCode":4476}' from endpoint 1 with groupID null
Debug Received Zigbee message from '0x680ae2fffe322a6c', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":587728433,"imageType":4550,"manufacturerCode":4476}' from endpoint 1 with groupID null
```

muddy79, 2021-11-08:
I believe the issue is related to this ticket: https://github.com/Koenkk/zigbee2mqtt/issues/7716
As a workaround, you can try to bind your device to the coordinator. Please select Bind tab, then select Source Endpoint 1, Destination to Coordinator, select OnOff Cluster (maybe others...) and press Bind button. It might be necessary to try several times, and it's helpful to wake up the device by pressing it's action button just before pressing the bind button on the webpage.

Phelton, 2021-11-08:
> I believe the issue is related to this ticket: #7716 As a workaround, you can try to bind your device to the coordinator. Please select Bind tab, then select Source Endpoint 1, Destination to [...]

Thanks a lot :) 
I have followed your procedure and now the  Shortcut Button device was returned to work perfectly.
I have only a last question... with new firmware, ikea says that the button acquire the  Double click functionality... i don't view in the action double click ... this means that in a future release of zigbee2mqtt will enable a recognize of double click for this device or that double click is not implemented yet whit this firmware release ?

Thanks a lot again for your support, your time and your cordiality.

Regards

Phelton from Italy

bartplessers, 2021-11-14:
Hello @muddy79 ,

I have the same problem with Ikea E1743.
- select on/off
- bind
- error:

[image]

```
Error Failed to bind cluster 'genOnOff' from 'my/n00/p00/switch/R08/voordeur' to 'Coordinator' (Error: Bind 0x14b457fffe6c4c22/1 genOnOff from '0x00124b00215ffe86/1' failed (AREQ - ZDO - bindRsp after 10000ms))
Error Failed to bind
```

any other suggestions?

grtz
B

hoggerz, 2021-11-16:
> Thanks a lot :) I have followed your procedure and now the Shortcut Button device was returned to work perfectly. I have only a last question... with new firmware, ikea says that the button acquire [...]

Hi Phelton,

My updated shortcut button produces 4 actions now:

action “on” for single click
action “off” for double click and 
Another action for the button hold and another for the button release (can’t remember off the top of my head)

hoggerz

Phelton, 2021-11-16:
> Hi Phelton,
>
> My updated shortcut button produces 4 actions now:
>
> action “on” for single click action “off” for double click and Another action for the button hold and another for the button release [...]

It's very interesting.
I still only see click, hold click and release hold click.
When I press twice it does not expose anything.

For better understand why you see the double click action and i not, could you tell me:
1- your version of zigbee2mqtt?
2- the firmware version and exact model of your shourtcat button?
3- if you have changed or not the bind configuration after the update?

Following my information:
1- my zigbee2mqtt: 1.22.0-2
2- ...
[image]

3- my actual bind option (is was necessary for me after the update):
[image]

Thanks you very much for your cooperatio and support.
bye
Phelton

877dev, 2021-11-16:
For info same issue here, with E1812 shortcut button, was working fine on firmware 2.3.015, then updated OTA to 2.3.080 and it will not respond to button presses, nothing happening in logs.

I tried changing battery, stopped Z2M, unplugged ZZH stich for 10secs, restarted Z2M. Press small button 4 times next to the ZZH coordinator:
```
Zigbee2MQTT:info  2021-11-16 16:14:06: Device 'Ikea_2_button_sentry' joined
Zigbee2MQTT:info  2021-11-16 16:14:06: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Ikea_2_button_sentry","ieee_address":"0x680ae2fffe2ec31a"},"type":"device_joined"}'
Zigbee2MQTT:info  2021-11-16 16:14:07: Configuring 'Ikea_2_button_sentry'
Zigbee2MQTT:info  2021-11-16 16:14:07: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"Ikea_2_button_sentry"},"type":"device_connected"}'
Zigbee2MQTT:debug 2021-11-16 16:14:09: Device 'Ikea_2_button_sentry' announced itself
Zigbee2MQTT:info  2021-11-16 16:14:09: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Ikea_2_button_sentry","ieee_address":"0x680ae2fffe2ec31a"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-11-16 16:14:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Ikea_2_button_sentry"},"type":"device_announced"}'
Zigbee2MQTT:debug 2021-11-16 16:14:11: Received Zigbee message from 'Ikea_2_button_sentry', type 'readResponse', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":87}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-11-16 16:14:11: MQTT publish: topic 'zigbee2mqtt/Ikea_2_button_sentry', payload '{"battery":87,"linkquality":228}'
Zigbee2MQTT:info  2021-11-16 16:14:11: Successfully configured 'Ikea_2_button_sentry'
Zigbee2MQTT:debug 2021-11-16 16:14:16: Received Zigbee message from 'Ikea_2_button_sentry', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":587728433,"imageType":4550,"manufacturerCode":4476}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-11-16 16:14:16: Device 'Ikea_2_button_sentry' requested OTA
Zigbee2MQTT:info  2021-11-16 16:14:16: MQTT publish: topic 'zigbee2mqtt/Ikea_2_button_sentry', payload '{"battery":87,"linkquality":228,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:warn  2021-11-16 16:14:18: Device 'Ikea_2_button_sentry' left the network
Zigbee2MQTT:info  2021-11-16 16:14:18: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Ikea_2_button_sentry","ieee_address":"0x680ae2fffe2ec31a"},"type":"device_leave"}'
```

dankocrnkovic, 2021-11-16:
Same here. No actions. Binding to Coordinator fails.
error 2021-11-16 17:24:50: Failed to bind cluster 'genOnOff' from 'Button 1' to 'Coordinator' (Error: Bind 0xec1bbdfffed0a996/1 genOnOff from '0x00124b001cd415d1/1' failed (AREQ - ZDO - bindRsp after 10000ms))

dankocrnkovic, 2021-11-16:
> I assume you are also pressing the button to wake it up beforehand/during? I did and had the same result.

Yes. pressing before, during... no luck....

TheDodger, 2021-11-16:
Okay, so here's the trick:

If you haven't updated to .080 yet, do the binding of endpoint 1 to endpoint 1 of the coordinator while the update is running.
Works on first try and the button is doing its "off" immediately after the update finishes.

If you're already on .080, just click the bind button a couple dozen times to queue enough requests. Then begin pressing the shortcut button every few seconds, 2-3 should fit. After 30+ seconds the binding should succeed, faster if you've got luck.

ocso, 2021-11-17:
> Okay, so here's the trick:
> 
> If you haven't updated to .080 yet, do the OnOff cluster binding from endpoint 1 to endpoint 1 of the coordinator while the update is running. Works on first try and the button is doing its "off" immediately after the update finishes.
> 
> If you're already on .080, just click the bind button a couple dozen times to queue enough requests. Then begin pressing the shortcut button every few seconds, 2-3 should fit. After 30+ seconds the binding should succeed, faster if you've got luck.

Thank you @TheDodger, it worked. I clicked the Bind button approx. 30 times and the shortcut button in every 1-2 seconds to keep alive the device and finally the binding was added and the click "action" was reported to HA again.

877dev, 2021-11-17:
> If you're already on .080, just click the bind button a couple dozen times to queue enough requests. Then begin pressing the shortcut button every few seconds, 2-3 should fit. After 30+ seconds the binding should succeed, faster if you've got luck.

Yes this worked for me too thanks! I probably clicked too many times as I got around 10-20 green ok messages:
[image]

When I press button now I get this:
```
Zigbee2MQTT:info  2021-11-17 14:15:52: MQTT publish: topic 'zigbee2mqtt/Ikea_2_button_sentry', payload '{"action":"on","battery":100,"linkquality":165,"update":{"state":"idle"},"update_available":false}'
```

When I long hold button I get this:
```
Zigbee2MQTT:debug 2021-11-17 14:15:57: Received Zigbee message from 'Ikea_2_button_sentry', type 'commandMoveWithOnOff', cluster 'genLevelCtrl', data '{"movemode":0,"rate":83}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-11-17 14:15:57: MQTT publish: topic 'zigbee2mqtt/Ikea_2_button_sentry', payload '{"action":"brightness_move_up","action_rate":83,"battery":100,"linkquality":168,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:debug 2021-11-17 14:15:58: Received Zigbee message from 'Ikea_2_button_sentry', type 'commandStopWithOnOff', cluster 'genLevelCtrl', data '{}' from endpoint 1 with groupID 0
```

And when releasing button after long hold:
```
Zigbee2MQTT:info  2021-11-17 14:15:58: MQTT publish: topic 'zigbee2mqtt/Ikea_2_button_sentry', payload '{"action":"brightness_stop","battery":100,"linkquality":165,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:debug 2021-11-17 14:16:02: Received Zigbee message from 'Ikea_2_button_sentry', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
```

hoggerz, 2021-11-17:
Hi,

Strangely I didn't have to re-bind mine, I did add it and get it working before the firmware update, so somehow it still works without issue, binding is different to yours too.

[image]
[image]

I haven't tried to remove and re interview it though, so it might break and require the 30 attempts at binding it though if I did.
Another shortcut button I have, went through the update process, had no response when pressed, so when I bound genOnOff it bound straight away to the coordinator, but reports the wrong firmware version, still showing 2.3.015, however, the OTA tab now shows no update for it either, even if I check so that's weird.

SkipperDo, 2021-11-25:
Ok... It seems I'm not that stupid...
I get the same Problem after OTA with my Shortcut Button...

Im using just mosquito, zigbee2mqtt & RedNode

Nothing else..

Hardware: Rasperry Pi3B, Conbee2

Any steps I can try to fix it?

tomasbedrich, 2021-12-18:
@TheDodger thanks man, after a few hours of trying, finally!

Koenkk, 2021-12-19:
This is fixed in the dev branch now. If you already updated the device before updating to the latest dev, wakeup the device by clicking on it and right after that click the yellow reconfigure button in the frontend.

closed 2021-12-19 by Koenkk