Koenkk/zigbee2mqtt #9554: Lights go unresponsive from time to time

hmorkemo, 2021-11-08:
Many of my lights (Ikea and Philips) go unavailable from time to time (several times pr. week), and can't be contolled. After a while, sometimes, they come back online again.
Sometimes a refresh in the z2m device panel brings the light back to life. But usually that doesn't help. Often I have to delete / add it.
It seems like the lights farthest away from the controller suffer the most.
All lights are quite evenly distributed in and around the house. The farthest distance from the ConBee is 20 meters.
They are uncontrollable both from HA and from the Zigbee2mqtt UI.
The same problem occurs with my Ikea Fyrtur blinds from time to time.
Typical error message in the zigbee2mqtt log:
"Publish 'set' 'state' to 'Utebod Lys2' failed: 'Error: Command 0x5c0272fffe840e54/1 genOnOff.off({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'"
I have 55 mains powered devices (Ikea and Philips lights, wall plugs). 48 battery powered devices (Sensors, fyrtur blinds, switches).

What could be the reason for this behaviour, and what can I do to make it stable?
## What happened
Lights stop responding to Zigbee2Mqtt.
## What did you expect to happen
Normal lights on/off for all lights.
## How to reproduce it (minimal and precise)
This happens spontaneoulsy, it seems, with no prior action
## Debug info
Zigbee2MQTT version: 1.22.0-2, Zigbee channel 25. Far away from any local WiFi channels (1, 6 and 8).
Adapter hardware: Conbee II
Adapter firmware version: 26720700, updated 7.nov.
HA Core 2021.11.1, Supervisor 2021.10.8

hmorkemo, 2021-11-09:
I have 3 philips lights: bakgården1, 2 and 3, and a zigbee group "bakgården".
They are quite close to each other.
Lights 1 and 3 are controllable like they should be.
Light 2, located in the middle, is not. When I switch it on/off there's no reaction except for the message:
`Publish 'set' 'state' to 'Bakgården2' failed: 'Error: Command 0x0017880102b354a7/11 genOnOff.on({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
`
BUT: When I switch on/off/dim using the group, all 3 lights behave just fine! And no error messages.
This seems, to me, very strange.  Bug in zigbee2mqtt? or in ConBee II?

rickieyyy, 2021-11-09:
I noticed the same issue, also using a Conbee II.
My network is quite big with 73 devices, but is should be enough for the conbee to handle.
it happends most often when I send commands to a loth of lights at the same time (lets say at the evening i turn on 10 lights at the same time).

Anonymenet77, 2021-11-10:
Hello,

I have the same problem.
my whole hue system is unstable since version 1.21.2.
I am losing the connection of certain bulbs.
I have to restart Z2M, it works again for a few minutes and then I lose the connection to the bulbs again.

Several posts were opened without response.

https://github.com/Koenkk/zigbee2mqtt/discussions/9101
https://github.com/Koenkk/zigbee2mqtt/issues/8815
https://github.com/Koenkk/zigbee2mqtt/issues/9188
https://github.com/Koenkk/zigbee2mqtt/issues/8830
https://github.com/Koenkk/zigbee2mqtt/issues/8993

bartplessers, 2021-11-14:
Same here.
Today I upgraded my system + reboot. 
Again 2 IKEA light bulbs were unresponsive.
What helps for me than is
- enable join
- physically power off bulb 
- power on bulb

So it seems the bulb was still member of my zigbee mesh, but just has forgotten that...

Grtz
B

funkstille, 2021-12-03:
Same here, using a Conbee 2.

Have not had this issue until last weekend, 
when I updated both
stick firmware to 26720700
and z2Mqtt to latest.

After those updates the problems started.

It only happens to Hue lights (even very old Bloom), 
OSRAM Lights are working fine. Xiaomi devices are fine.

The Hue lights are responsive for ~30 seconds after power cycle,
after that I receive only time outs...

Could anyone try to downgrade conbee2 an check?
I do not have an environment available for flashing the stick right now...

funkstille, 2021-12-03:
Downgraded to

Zigbee2MQTT Version
1.21.0 commit: 70891eec
Coordinator Typ
ConBee2/RaspBee2
Coordinator Version
0x26680700
Frontend Version
0.5.18

Problems still occur.

funkstille, 2021-12-03:
Another Observation:

I do have on Hue e14 Bulb with firmware 1.46.13_r26312.
It does not lose connection. Is very responsive. Firmware Update is available but fails.

I also had an old Hue Iris which was responsive until I updated the bulb firmware to 5.127.1.26581.
Since then Iris not answering anymore. All other hue bulbs have firmware 5.127.1.26581 as well.

rickieyyy, 2021-12-03:
For me the issue is that random devices do not respond, it looks like it's happening when i send out multiple commands at the same time, for example if i set that 4 lights should go on, only 3 respond the other give a timeout error.

funkstille, 2021-12-03:
Did another check. Connected the updated Hue Iris to a HUE Bridge. Bulb stays responsive all the time. Checked for further bulb firmware updates with HUE Bridge - nothing available

Koenkk, 2021-12-04:
@funkstille could you provide a sniff of this happening? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

funkstille, 2021-12-04:
@Koenkk Will try later that evening. Never tried before. To get I right: Do I need the flashed cc2531 additional to the CONBEE2 stick?

Koenkk, 2021-12-05:
@funkstille yes, conbee 2 as your coordinator, cc2531 for the sniffing.

funkstille, 2021-12-07:
Since this is the first time I've done this, I don't know what to look for. I have noticed these errors:

```
Frame 337: 57 bytes on wire (456 bits), 57 bytes captured (456 bits) on interface -, id 0
    Interface id: 0 (-)
    Encapsulation type: IEEE 802.15.4 Wireless PAN (104)
    Arrival Time: Dec  6, 2021 13:31:49.323930000 CET
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1638793909.323930000 seconds
    [Time delta from previous captured frame: 0.721275000 seconds]
    [Time delta from previous displayed frame: 0.721275000 seconds]
    [Time since reference or first frame: 140.323930000 seconds]
    Frame Number: 337
    Frame Length: 57 bytes (456 bits)
    Capture Length: 57 bytes (456 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: wpan:data]
IEEE 802.15.4 Data, Dst: 0xffcf, Src: 0x65c1, Bad FCS
    Frame Control Field: 0x8841, Frame Type: Data, PAN ID Compression, Destination Addressing Mode: Short/16-bit, Frame Version: IEEE Std 802.15.4-2003, Source Addressing Mode: Short/16-bit
    Sequence Number: 200
    Destination PAN: 0x1a62
    Destination: 0xffcf
    Source: 0x65c1
    FCS: 0x0000 (Incorrect, expected FCS=0x4678)
    [Expert Info (Warning/Checksum): Bad FCS]
        [Bad FCS]
        [Severity level: Warning]
        [Group: Checksum]
Data (46 bytes)
    Data: 0802fcffc1651ed828e1f0050655f2e7000123170000c021eac0aa2b72ffb901af66289c…
    [Length: 46]
```

Maybe these errors cause the device to be blocked?

Koenkk, 2021-12-07:
@funkstille most of the time this is due to your sniffer being too far away from the transmitting devices, therefore usually not a problem.

funkstille, 2021-12-07:
Ok. I will try again an move the bulb closer to the sniffer ;)

funkstille, 2021-12-13:
If I move the bulb closer to the stay reactive, I was even able to do a firmware update. But If I move the bulb to its origin place (with power off / on) it gets recognized but has the same problem as before: not responsive after a few seconds. I have a lot of sniffing logs. But I could not find any evidence since I am no expert.
 
There are enough routers (inns plugs, osram lights) in the house. It has worked for months without any problem. Only the Hue bulbs with newer firmware "get lost". 1 hue with an old firmware stays responsive even though it has hardly no connection (LQI < 50). 
Attached is a Wireshark file where 0x5c3b becomes unresponsive after I moved it to its origin place.

[wireshark 5c3b.pcapng.txt](https://github.com/Koenkk/zigbee2mqtt/files/7705742/wireshark.5c3b.pcapng.txt)
.

Koenkk, 2021-12-14:
@chrishae do you know if this is a known problem with the conbee II?

ChrisHae, 2021-12-16:
Not really. We have also networks running with > 100 devices. I can just recomend to update the FW to the latest version 26720700 and to try to use an USB extension cable with the ConBee2.
Also there is a new FW available for IKEA devices that address the describted issues:
https://www.reddit.com/r/tradfri/comments/rg77ed/new_firmware_release/

funkstille, 2021-12-16:
Today I flashed deCONZ firmware version 26680700
```
sudo GCFFlasher_internal -t 60 -d /dev/ttyACM0 -f deCONZ_ConBeeII_0x264a0700.bin.GCF
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2214230-if00 (ConBee II)
deCONZ firmware version 26680700
```
and restarted zigbee2mqtt - everything is up and running again with latest zigbee2mqtt version. 
No problems with unresponsive bulbs anymore. 
I will test for a few days and try to update to latest conbee 2 firmware to double check if the problem occours again.

olschm, 2021-12-30:
I had similar issues with the latest firmware for the conbee II mostly with some hue bulbs and ikea TRADFRI drivers.
Also for me, unitl now, the downgrade to 26680700 firmware improved the availability of the devices.

@funkstille  Did you try to update again to a newer fw in the meantime?

funkstille, 2021-12-30:
@olschm
No, I did not try to upgrade so far. I haven’t had a single issue with unresponsive bulbs since downgrading.

Since all other devices also working without problems there is no need to upgrade for me so far.

MGdRomph, 2022-01-02:
I been experiencing the same problem for a while. Some Hue bulbs became unresponsive after a couple of seconds, while similar bulbs, next to the other one, would continue to work fine. 

After reading @funkstille most recent posts I gave it another try and tested with firmware 26720700 and 26680700 of the ConBee II, with both version the problem is there. When I downgraded to version 264a0700, just like @funkstille, the problem was resolved and now everything seems to be running as expected.

If there is anything I can do to help resolving the problem, which seems to be in combination with the ConBee II most recent firmware versions, please let me know.

ChrisHae, 2022-01-03:
@MGdRomph Did the Hue bulbs become unresponsive after pairing or did they already run for a while? And do you have any IKEA bulbs in your network?

Anonymenet77, 2022-01-03:
Hello,

I made you by my experience having a conbee and also problems with my HUE bulbs.

I tested different firmware of the conbee here is the list of firmware that worked

OK -> deCONZ_ConBeeII_0x264a0700.bin.GCF                 27-May-2019 10:03
Not tested -> deCONZ_ConBeeII_0x26570700.bin.GCF                 27-Apr-2020 09:59  
OK -> deCONZ_ConBeeII_0x26580700.bin.GCF                 10-May-2020 13:16
KO -> deCONZ_ConBeeII_0x26650700.bin.GCF                 13-Sep-2020 15:46 
KO -> deCONZ_ConBeeII_0x26660700.bin.GCF                 07-Oct-2020 10:19
KO -> deCONZ_ConBeeII_0x26670700.bin.GCF                 18-Nov-2020 12:51  
KO -> deCONZ_ConBeeII_0x26680700.bin.GCF                 30-Nov-2020 18:03
KO -> deCONZ_ConBeeII_0x26700700.bin.GCF                 11-Aug-2021 07:39
KO -> deCONZ_ConBeeII_0x26720700.bin.GCF                 19-Aug-2021 22:20

I remain available to advance the subject

sorry for my English

MGdRomph, 2022-01-03:
> @MGdRomph Did the Hue bulbs become unresponsive after pairing or did they already run for a while? And do you have any IKEA bulbs in your network?

At that point I was migrating some HUE bulbs from the HUE bridge to ConBee II/Zigbee2Mqtt, after pairing they worked a couple of commands and then became unresponsive. 
I have no IKEA bulbs in my network but the two bulbs that became unresponsive do have the new firmware version 67.91.1, all the other bulbs (that work fine) have version 67.88.1.

Anonymenet77, 2022-01-03:
what's the version of your conbee || ?

MGdRomph, 2022-01-03:
> what's the version of your conbee || ?

At this moment I am using 264a0700, as this was the version metioned by @funkstille that worked for him. 
And thanks for testings all versions of the firmware, I will give version 26580700 and 26650700 a try as well just to verify if indeed the problem arrises between those two versions.

MGdRomph, 2022-01-03:
With the ConBee II firmware version 26650700 both HUE bulbs with firmware version 67.91.1 stopped responding after a few minutes (one almost directly, the other one stayed working for about a minute or two). 

After that I downgraded to ConBee II firmware version 26580700. Bulbs still didn't respond but after a power cycle both bulbs did became responsive again and are working still working fine.

Just like @Anonymenet77 already wrote, with firmware version 26580700 all seems to be working fine from version 26650700 and up my HUE bulbs with firmware version 67.91.1 become unresponsive after some time.

Anonymenet77, 2022-01-03:
> With the ConBee II firmware version 26650700 both HUE bulbs with firmware version 67.91.1 stopped responding after a few minutes (one almost directly, the other one stayed working for about a minute or two).
> 
> After that I downgraded to ConBee II firmware version 26580700. Bulbs still didn't respond but after a power cycle both bulbs did became responsive again (I did turn on "permit join" in Zigbee2Mqtt as well) and are working still working fine.
> 
> Just like @Anonymenet77 already wrote, with firmware version 26580700 all seems to be working fine from version 26650700 and up my HUE bulbs with firmware version 67.91.1 become unresponsive after some time.

For my part my concern is resolved since I was in version of CONZ_ConBeeII_0x26580700.bin.GCF 10-Mai-2020 13:16 👍 

do you still have the problem even after changing firmware?

MGdRomph, 2022-01-04:
> For my part my concern is resolved since I was in version of CONZ_ConBeeII_0x26580700.bin.GCF 10-Mai-2020 13:16 👍
>
> do you still have the problem even after changing firmware?

I am using 26580700 now as well, that works fine.

mmakaay, 2022-01-18:
Just upgraded four of my first generation HUE lighs (3 bulbs, 1 LED strip) to firmware version 67.91.1.
Ever since, these four lights have become really sluggish to react at times. When turning on or off the lights, at times they don't seem to react at all. After waiting a bit, they eventually change state. Since this happens only with these first gen lights since the OTA update, my suspicion is on the firmware.
I got quite a few other HUE lights running, also ones that are right next to the first gen lights, and those work just fine.

BTW: I don't have a ConbeeII stick active.
My current setup uses the Slaesh's CC2652RB stick.

mwthrane, 2022-02-04:
> I have 3 philips lights: bakgården1, 2 and 3, and a zigbee group "bakgården". They are quite close to each other. Lights 1 and 3 are controllable like they should be. Light 2, located in the middle, [...]

I have the same problem. I have 50 devices with none with less that 240 LQI. All routers are HUE.
I use deconz conbee II with firmware 26720700. Zigbee2MQTT 1.22.2 commit 1a0a9a6e

mwthrane, 2022-02-05:
> I have the same problem. I have 50 devices with none with less that 240 LQI. All routers are HUE. I use deconz conbee II with firmware 26720700. Zigbee2MQTT 1.22.2 commit [...]

Upgraded to Z2M 1.23.0 and tried FW 26580700 and 26680700. Does not make any diff.

funkstille, 2022-02-05:
Did you powercycle the bulbs and restart z2mqtt?

did you check if flashing the the stick turned out ok. Maybe downgrade did not work…?

```
sudo GCFFlasher_internal -t 60 -d /dev/ttyACM0 -f deCONZ_ConBeeII_0x264a0700.bin.GCF
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2214230-if00 (ConBee II)
deCONZ firmware version 26680700
```

mwthrane, 2022-02-05:
Yeah i did a complete new installation of z2m and powercycled all bulbs. 
I did check the firmware after all flashes and also unplugged the device before and after the flash.

Anonymenet77, 2022-02-05:
> > > J'ai 3 lampes philips: bakgården1, 2 et 3, et un groupe zigbee "bakgården". Ils sont assez proches l'un de l'autre. Les lumières 1 et 3 sont contrôlables comme elles devraient l'être. La lumière 2, située au milieu, ne l'est pas. Lorsque je l'allume/l'éteins, il n'y a aucune réaction à l'exception du message : `Publish 'set' 'state' to 'Bakgården2' failed: 'Error: Command 0x0017880102b354a7/11 genOnOff.on({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)' `MAIS : lorsque j'allume/éteins/dim en utilisant le groupe, les 3 lumières se comportent très bien ! Et aucun message d'erreur. Cela me paraît très étrange. Bogue dans zigbee2mqtt ? ou dans ConBee II ?
> > 
> > 
> > J'ai le même problème. J'ai 50 appareils dont aucun avec moins de 240 LQI. Tous les routeurs sont HUE. J'utilise deconz conbee II avec le firmware 26720700. Zigbee2MQTT 1.22.2 commit [1a0a9a6](https://github.com/Koenkk/zigbee2mqtt/commit/1a0a9a6eabf58855d3a6feda60bb79a1c34d17f8)
> 
> Mise à niveau vers Z2M 1.23.0 et essai des FW 26580700 et 26680700. Ne fait aucune différence.

Hello,

try the following firmwares for this solved the problem since no problem.

OK -> deCONZ_ConBeeII_0x264a0700.bin.GCF 27-May-2019 10:03
Non testé -> deCONZ_ConBeeII_0x26570700.bin.GCF 27-Apr-2020 09:59
OK -> deCONZ_ConBeeII_0x26580700.bin.GCF

> Upgraded to Z2M 1.23.0 and tried FW 26580700 and 26680700. Does not make any diff.

> > > J'ai 3 lampes philips: bakgården1, 2 et 3, et un groupe zigbee "bakgården". Ils sont assez proches l'un de l'autre. Les lumières 1 et 3 sont contrôlables comme elles devraient l'être. La lumière 2, située au milieu, ne l'est pas. Lorsque je l'allume/l'éteins, il n'y a aucune réaction à l'exception du message : `Publish 'set' 'state' to 'Bakgården2' failed: 'Error: Command 0x0017880102b354a7/11 genOnOff.on({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)' `MAIS : lorsque j'allume/éteins/dim en utilisant le groupe, les 3 lumières se comportent très bien ! Et aucun message d'erreur. Cela me paraît très étrange. Bogue dans zigbee2mqtt ? ou dans ConBee II ?
> > 
> > 
> > J'ai le même problème. J'ai 50 appareils dont aucun avec moins de 240 LQI. Tous les routeurs sont HUE. J'utilise deconz conbee II avec le firmware 26720700. Zigbee2MQTT 1.22.2 commit [1a0a9a6](https://github.com/Koenkk/zigbee2mqtt/commit/1a0a9a6eabf58855d3a6feda60bb79a1c34d17f8)
> 
> Mise à niveau vers Z2M 1.23.0 et essai des FW 26580700 et 26680700. Ne fait aucune différence.

Hello,

try the following firmwares for this solved the problem since no problem.

OK -> deCONZ_ConBeeII_0x264a0700.bin.GCF 27-May-2019 10:03
Not tested -> deCONZ_ConBeeII_0x26570700.bin.GCF 27-Apr-2020 09:59
OK -> deCONZ_ConBeeII_0x26580700.bin.GCF

Anonymenet77, 2022-02-05:
Hello,

try the following firmwares for me it solved the problem.
I haven't had any issues since.

OK -> deCONZ_ConBeeII_0x264a0700.bin.GCF 27-May-2019 10:03
Not tested -> deCONZ_ConBeeII_0x26570700.bin.GCF 27-Apr-2020 09:59
OK -> deCONZ_ConBeeII_0x26580700.bin.GCF 10-May-2020 13:16

mwthrane, 2022-02-05:
Already tried 0x26580700.bin.GCF, did not work.
Also tried 0x264a0700.bin.GC now. 
Did not solve the issue.

mvosz, 2022-03-13:
Same problem here with conbeeII as Home Assistant Addon in Firmware 26660700 and only two philips hue light bulbs. Usb extension in use. The problem was always there

hscholing, 2022-03-29:
Running my environment on a Homeassistant Supervised installation based on Linux. Can I take my Conbee stick out of this system and flash it on a different machine using the update instructions? Will I lose my configuration on the stick? Some lights are also unstable in my situation. Running combination of Ikea and Hue bulbs

funkstille, 2022-03-30:
@hscholing did it the same way as you are planing to do it and it worked for me

deosrc, 2022-04-23:
Unsure if this is the same issue but I am also having devices go offline in zigbee2mqtt. I have definitely seen Hue and Gledopto devices go offline but not sure about the others.

Interestingly, the "offline" devices are still controllable through groups. I also find that restarting zigbee2mqtt and waiting ~15 minutes brings them all back online again.

```
Zigbee2MQTT version: 1.25.0
Coordinator type: ConBee2/RaspBee2
Coordinator revision: 0x26580700

Total 63

By device type
Router: 42
End devices: 21

By power source
Mains (single phase): 42
Battery: 21

By vendor
Signify Netherlands B.V.: 19
IKEA of Sweden: 19
Philips: 9
LUMI: 6
LDS: 5
GLEDOPTO: 4
KE: 1
```

deosrc, 2022-04-23:
Sorry just found #11650 which looks more relevant to my issue above. Will check that and post there if still having issues.

closed 2022-06-05 by github-actions[bot]

Papnaas, 2022-06-28:
> Just upgraded four of my first generation HUE lighs (3 bulbs, 1 LED strip) to firmware version 67.91.1. Ever since, these four lights have become really sluggish to react at times. When turning on [...]

I have the same issue since updating my Hue Iris 7199960PH to FW 67.91.1. The lights are available and responsive for some seconds, then I get 
`Zigbee2MQTT:error 2022-06-29 00:43:54: Publish 'set' 'state' to 'Hue IRIS WoZi Ecke Sofa' failed: 'Error: Command 0x0017880101226bf0/11 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 64128 - 11 - 22 - 6 - 11 after 10000ms)'
Zigbee2MQTT:info  2022-06-29 00:43:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Hue IRIS WoZi Ecke Sofa' failed: 'Error: Command 0x0017880101226bf0/11 genOnOff.on({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 64128 - 11 - 22 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"Hue IRIS WoZi Ecke Sofa"},"type":"zigbee_publish_error"}'
`
So I'm pretty sure it is a FW issue of the hue bulbs. Is there any chance to downgrade the FW of the bulbs?

My configuration uses an Electrolama zig-a-zig-ah! (zzh!) CC2652R adapter.

Papnaas, 2022-06-29:
After some more tests, I tend to say, that with the update the Hue lights lost massively in signal strenght.
Everything seems to be ok as log as the devices are very close to the coordinator. But as soon as they move away, link quality is massively reduced. Even close to the coordinator the values are lower than before the update at very distant positions in the network. 
I even have the assumption that somehow the devices lost their router capability...

Papnaas, 2022-06-29:
Made a networkmap and yes, the Hue Iris are completely unconnected within each other and also with the coordinator. Signal quality is always 0. I really wonder, what they did at Philips with this FW update...

funkstille, 2022-06-29:
can not confirm. At least in my case with the downgraded ConBeeII Stick they are working as coordinator / sibling.

Papnaas, 2022-06-29:
@funkstille: Did you make a network map to check the link quality within your Hue equipment before downgrading? Was it different from after the downgrade? I still wonder why the IRIS do not connect to each other although they are right beside each other. There is no such problem with other devices like e.g. OSRAM plug which is at the same spot than one of my IRIS...

The release notes for the release 67.91.1 says something about "security issues". I wonder if it has to do something with that. That could at least explain why it has something to do with the Stick and its FW, but I'm a complete moron when it comes to Zigbee itself...

Maybe I should completely setup the network from scratch...

funkstille, 2022-06-29:
No I did not check before and after.
It was a kind of trial and error process.
At the point everything was working again, I stopped and left it that way.

But during that time I also tried a reset and rebuild from scratch with no luck either.
The only thing that worked for me was downgrading the conbeeII stick.

TheJulianJES, 2022-07-01:
There's a new update available for Hue lights (1.93.11 or 67.93.11). It's also available for some older non-Bluetooth Hue lights.
I don't think it'll help with your issues but just wanted to let you know.

CedricRBR, 2022-08-12:
@Papnaas Same here, all bulbs are on FW version 1.93.11. They pair fine with the ConBee2 but I can't control them whatsoever. 
On a side note: same happens with SML003 motion sensor on FW version 2.53.6.

Papnaas, 2022-08-13:
#

> Today I flashed deCONZ firmware version 26680700
> 
> ```
> sudo GCFFlasher_internal -t 60 -d /dev/ttyACM0 -f deCONZ_ConBeeII_0x264a0700.bin.GCF
> GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
> Reboot device /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2214230-if00 (ConBee II)
> deCONZ firmware version 26680700
> ```
> 
> and restarted zigbee2mqtt - everything is up and running again with latest zigbee2mqtt version. No problems with unresponsive hue bulbs anymore. I will test for a few days and try to update to latest conbee 2 firmware to double check if the problem occours again.

@CedricRBR Did you try to downgrade your conbee as suggested by @funkstille? It seemed to work with the conbee…
With my Electrolama zzh! I‘m still experiencing very low connectivity but to some extend the bulbs work again, sometimes I have to wait a little. So I actually thought about changing to the conbee.
@CedricRBR Would be interested to know if downgrading works for you as well!

TheJulianJES, 2022-08-16:
For those with unresponsive Hue lights from time to time, are you also running mains-powered IKEA devices on your network?

Papnaas, 2022-08-16:
> For those with unresponsive Hue lights from time to time, are you also running mains-powered IKEA devices on your network?

Not for me, no IKEA devices in my network…

MGdRomph, 2022-08-16:
> For those with unresponsive Hue lights from time to time, are you also running mains-powered IKEA devices on your network?

Nope

accelle17, 2022-08-22:
Two hue bulbs updated to 67.93.11, after that, I've experienced this issue and it won't go away unless I restart z2m.

mwthrane, 2022-09-15:
> For those with unresponsive Hue lights from time to time, are you also running mains-powered IKEA devices on your network?

Nope.

radionurd, 2022-11-08:
> For those with unresponsive Hue lights from time to time, are you also running mains-powered IKEA devices on your network?

yep

OrbitingOcelot, 2022-11-19:
I recently wasted a couple of hours while trying to switch my conbee2 based NW to z2m and ran into the issues mentioned. Downgrading the FW to 0x26580700 finally gave me a stable network. I made a PR to mention this in the z2m docs (supported adapters section), hopefully this will save others some trouble.

mgabor3141, 2022-12-12:
Raspbee II user here with similar issues until today. I downgraded to version 26680700 as @funkstille suggested and it works nicely so far.

I've also been using the deconz addon before and switched to zigbee2mqtt now, very happy with it so far. The phoscon interface was very hit or miss, it gave conflicting info about device states many times. Re-adding devices was also unreliable, as the frontend was saying inclusion is enabled while the logs had errors. Zigbee2mqtt is so much more reliable and intuitive.

[58 more replies]