Koenkk/zigbee2mqtt #9558: HG06467: Setting brightness results lost of color

PetriAsi, 2021-11-08:
## What happened
Setting brightness for device results lost of color that have been set. Device color is always white after setting brightness.

## What did you expect to happen
Only brightness of light  should change and selected hue and saturation should be kept.

## How to reproduce it (minimal and precise)
First noticed this from home assistant. Tried to set hs_color and brightness at same time, results were alway white.

I have reproduced  this also  from  Zigbee2MQTT  web gui,
Open devices exposes tab , set color and the set brightness. Result is always white.

If I set brigthness  first and then hue, final result is ok, but it flash to white first.

## Debug info
Zigbee2MQTT version: (docker image  koenkk/zigbee2mqtt version 1.22.0 )
Adapter hardware: Slaesh's CC2652RB stick
Adapter firmware version: 

Info message look always like same when setting only color or brightness or both:
```
 MQTT publish: topic 'zigbee2mqtt/JouluValonauha', payload '{"brightness":229,"color":{"b":229,"h":325,"s":43},"effect":null,"linkquality":75,"state":"ON"}'
```

PetriAsi, 2021-12-14:
How should I try to debug this?

nielsfaber, 2022-12-09:
@Koenkk Can we re-open this?
I can confirm this is still actual, changing brightness results in white color (also in the Z2M GUI).
I suppose the actual problem is caused by the light itself (it has some quirks, also within the Lidl ecosystem).
Would it be possible to change the driver to first set a brightness command, followed by the color? I think this will help a lot.

Koenkk, 2022-12-09:
@nielsfaber to debug this further, someone has to sniff the traffic when controlling this device via the original TuYa gateway: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

nielsfaber, 2022-12-09:
@Koenkk Thanks for the re-opening 🙂 
I'm willing to dive into it, but I don't have a CC2531. Would the CC2652P (SONOFF) work as well for the sniffing? It is not mentioned in the guide you linked.

Koenkk, 2022-12-10:
I've never tested it, but it might also work.

closed 2023-01-18 by github-actions[bot]