Koenkk/zigbee2mqtt #9561: PLUG EDP RE:DY (manufactured by EDP) missing power and energy info

nbranquinho, 2021-11-08:
## What happened
After pair the  PLUG EDP RE:DY (manufactured by EDP), I don't have the energy/power information (shows "null") and the log shows an error

## What did you expect to happen
Have the energy and power information

## How to reproduce it (minimal and precise)
In my case it's only pair the device with coordinator

## Debug info
Zigbee2MQTT version:
Adapter hardware: Conbee II
Adapter firmware version:  0x26720700
Frontend version 0.6.30
(HA Addon)

Zigbee2MQTT:error 2021-11-08 18:16:57: Failed to configure '0x00124b0003cd80a6', attempt 1 (Error: Read 0x00124b0003cd80a6/85 seMetering(["multiplier","divisor"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:293:23)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:413:22)
    at Object.readMeteringMultiplierDivisor (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:28:5)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/edp.js:20:13)
    at Configure.configure (/app/lib/extension/configure.ts:115:13))
Zigbee2MQTT:error 2021-11-08 18:17:07: Failed to configure '0x00124b0003cd80a6', attempt 2 (Error: Read 0x00124b0003cd80a6/85 seMetering(["multiplier","divisor"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:293:23)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:413:22)
    at Object.readMeteringMultiplierDivisor (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:28:5)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/edp.js:20:13)
    at Configure.configure (/app/lib/extension/configure.ts:115:13))
Zigbee2MQTT:error 2021-11-08 18:17:10: Failed to configure '0x00124b0003cd80a6', attempt 3 (Error: Read 0x00124b0003cd80a6/85 seMetering(["multiplier","divisor"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:293:23)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:413:22)
    at Object.readMeteringMultiplierDivisor (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:28:5)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/edp.js:20:13)
    at Configure.configure (/app/lib/extension/configure.ts:115:13))
Zigbee2MQTT:error 2021-11-08 18:19:44: Failed to configure 'Máquina Lavar Roupa', attempt 4 (Error: Read 0x00124b0003cd80a6/85 seMetering(["multiplier","divisor"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:293:23)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:413:22)
    at Object.readMeteringMultiplierDivisor (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:28:5)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/edp.js:20:13)
    at Configure.configure (/app/lib/extension/configure.ts:115:13)
    at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21))

sjorge, 2021-11-08:
`UNSUPPORTED_ATTRIBUTE` would seem to indicate the device does not support the cluster/attributes needed for the energy metering.

I wonder if there are different revisions of the device.

nbranquinho, 2021-11-09:
> `UNSUPPORTED_ATTRIBUTE` would seem to indicate the device does not support the cluster/attributes needed for the energy metering.
> 
> I wonder if there are different revisions of the device.

I don't know if this information helps or not but in ZHA, I get the information directly as a sensor. In Deconz the information is available as an attribute of a sensor.

closed 2021-12-25 by github-actions[bot]

r3pek, 2022-01-21:
Problem still occurs as of today

r3pek, 2022-01-21:
Ok... after some time, the "power" setting updates. Only the "energy" one isn't available (and it was available on ZHA)
[image]

r3pek, 2022-01-25:
@Koenkk can you please take a look here? From what I could gather, this device behaves the same as the "Innr SP 120" for example which just reports the multiplier and divisor as `UNSUPPORTED_ATTRIBUTE` (it actually is an optional attribute on ZigBee spec). Question is what are the sane defaults to this? 1/100, 1/1000, 1/1? Also, how can I test this on a running instance of HomeAssistant OS (since I don't have access to the running container)?

r3pek, 2022-01-25:
OK, confirmed ;)

```
module.exports = [
    {
        zigbeeModel: ['ZB-SmartPlug-1.0.0'],
        model: 'PLUG EDP RE:DY',
        vendor: 'EDP',
        description: 're:dy plug',
        fromZigbee: [fz.on_off, fz.metering],
        toZigbee: [tz.on_off],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(85);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
            endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 1000, multiplier: 1});
            await reporting.onOff(endpoint);
            //await reporting.readMeteringMultiplierDivisor(endpoint);
            await reporting.instantaneousDemand(endpoint);
        },
        exposes: [e.switch(), e.power(), e.energy()],
    },
];
```

This one works:

[image]