Koenkk/zigbee2mqtt #9564: SilverCrest PowerStrip 3 AC (16A) with 4 USB Model HG06338

nbranquinho, 2021-11-08:
## What happened
After pair , can't control each socket of device.
When I toggle the first socket, it's change all 3.
the  other two, return error.
**Zigbee2MQTT:error** 2021-11-08 20:39:07: Device '0x50325ffffe5308b5' has no endpoint 'l2'
**Zigbee2MQTT:error** 2021-11-08 20:39:07: Device '0x50325ffffe5308b5' has no endpoint 'l3'

## What did you expect to happen
control the 3 switch individual

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.22.0
Adapter hardware: Conbee II
Adapter firmware version: 0x26720700
Frontend version 0.6.30
(HA Addon)

https://zigbee.blakadder.com/Lidl_HG06338.html

2021-11-08T20:25:08.295Z zigbee-herdsman:controller:device:log Interview - got 'hardwareVersion' for device '0x50325ffffe5308b5'
2021-11-08T20:25:08.295Z zigbee-herdsman:controller:endpoint Read 0x50325ffffe5308b5/1 genBasic(["dateCode"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-11-08T20:25:08.296Z zigbee-herdsman:deconz:adapter zclFrame.payload:
2021-11-08T20:25:08.296Z zigbee-herdsman:deconz:adapter [ { attrId: 6 } ]
2021-11-08T20:25:08.296Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x2017 EP:1 SeqNr. 100 request id: 20
2021-11-08T20:25:08.307Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 20 status: 0
2021-11-08T20:25:08.308Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-11-08T20:25:08.308Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2021-11-08T20:25:08.308Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-11-08T20:25:08.312Z zigbee-herdsman:deconz:driver query aps data confirm
2021-11-08T20:25:08.317Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 101
2021-11-08T20:25:08.320Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x2017 request id: 20 confirm status: 0
2021-11-08T20:25:08.320Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-11-08T20:25:08.320Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 12
2021-11-08T20:25:08.358Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2021-11-08T20:25:08.358Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2021-11-08T20:25:08.360Z zigbee-herdsman:deconz:driver query aps data indication
2021-11-08T20:25:08.366Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 102
2021-11-08T20:25:08.370Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 102 srcAddr: 0x2017 destAddr: 0x0 profile id: 0x104 cluster id: 0x0 lqi: 255
2021-11-08T20:25:08.370Z zigbee-herdsman:deconz:frameParser response payload: 24,12,1,6,0,0,66,0
2021-11-08T20:25:08.371Z zigbee-herdsman:deconz:adapter resolve data request with transSeq Nr.: 12
2021-11-08T20:25:08.372Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":12,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":6,"status":0,"dataType":66,"attrData":""}],"Command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}},"address":8215,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
2021-11-08T20:25:08.373Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-11-08T20:25:08.373Z zigbee-herdsman:deconz:adapter response received
2021-11-08T20:25:08.374Z zigbee-herdsman:controller:device:log Interview - got 'dateCode' for device '0x50325ffffe5308b5'
2021-11-08T20:25:08.374Z zigbee-herdsman:controller:endpoint Read 0x50325ffffe5308b5/1 genBasic(["swBuildId"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-11-08T20:25:08.375Z zigbee-herdsman:deconz:adapter zclFrame.payload:
2021-11-08T20:25:08.375Z zigbee-herdsman:deconz:adapter [ { attrId: 16384 } ]
2021-11-08T20:25:08.383Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x2017 EP:1 SeqNr. 103 request id: 21
2021-11-08T20:25:08.399Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 21 status: 0
2021-11-08T20:25:08.399Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-11-08T20:25:08.399Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2021-11-08T20:25:08.399Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-11-08T20:25:08.405Z zigbee-herdsman:deconz:driver query aps data confirm
2021-11-08T20:25:08.410Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 104
2021-11-08T20:25:08.413Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x2017 request id: 21 confirm status: 0
2021-11-08T20:25:08.413Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-11-08T20:25:08.413Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 13
2021-11-08T20:25:08.450Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2021-11-08T20:25:08.450Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2021-11-08T20:25:08.452Z zigbee-herdsman:deconz:driver query aps data indication
2021-11-08T20:25:08.458Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 105
2021-11-08T20:25:08.462Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 105 srcAddr: 0x2017 destAddr: 0x0 profile id: 0x104 cluster id: 0x0 lqi: 255
2021-11-08T20:25:08.462Z zigbee-herdsman:deconz:frameParser response payload: 24,13,1,0,64,134
2021-11-08T20:25:08.462Z zigbee-herdsman:deconz:adapter resolve data request with transSeq Nr.: 13
2021-11-08T20:25:08.463Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":13,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":16384,"status":134}],"Command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}},"address":8215,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
2021-11-08T20:25:08.464Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-11-08T20:25:08.464Z zigbee-herdsman:deconz:adapter response received
2021-11-08T20:25:08.465Z zigbee-herdsman:controller:endpoint Read 0x50325ffffe5308b5/1 genBasic(["swBuildId"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
[... 26 lines cut ...]
2021-11-08T20:25:08.662Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101110
2021-11-08T20:25:08.662Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[... cut ...]
2021-11-08T20:25:16.754Z zigbee-herdsman:deconz:frameParser device state: 10100010
2021-11-08T20:25:16.755Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-11-08T20:25:26.742Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 114
2021-11-08T20:25:26.748Z zigbee-herdsman:deconz:frameParser device state: 10100010
2021-11-08T20:25:26.749Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1

simaoafonso, 2021-11-09:
I have the same problem if anyone knows how to solve this problem tell us pls

BFFonseca, 2021-11-30:
This is happening to me as well. When I toggle the 1st switch, they all toggle. When I try to toggle switches 2 or 3, I get errors.
Here's Z2M's log:
[image]

From what I can tell, the endpoints aren't being exposed properly.
Only the 1st endpoint is available.
This is what I have on Z2M:
[image]

And the database entry is:
```
{"id":17,"type":"Router","ieeeAddr":"0x50325ffffe4dfea7","nwkAddr":21599,"manufId":4098,"manufName":"_TZ3000_1obwwnmq","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_1obwwnmq","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1638287345937}
```

I'm using Zigbee2MQTT on Home Assistant with the official add-on (v1.21.0-4)

MattWestb, 2021-12-01:
I having one that is around one year old and its have 4 end points and working OK switching etch plug.

Its have App version 69 then reading it from basic cluster and some user is having version 70 that looks having problem in the firmware.

Do some user have one LIDL or tuya ZBGW for testing if its working there with some magic ?

hcastanho, 2021-12-12:
I have the exact same problem

`Zigbee2MQTT:error 2021-12-12 18:22:20: Device 'MQTT Extensao Escritorio' has no endpoint 'l2'
Zigbee2MQTT:debug 2021-12-12 18:22:21: Received MQTT message on 'zigbee2mqtt/MQTT Extensao Escritorio/l3/set' with data '{"state_l3":"ON"}'
Zigbee2MQTT:error 2021-12-12 18:22:21: Device 'MQTT Extensao Escritorio' has no endpoint 'l3'
Zigbee2MQTT:debug 2021-12-12 18:22:26: Received Zigbee message from 'MQTT Cama Hugo', type 'attributeReport', cluster 'genBasic', data '{"65506":30,"appVersion":66}' from endpoint 1 with groupID 0`

nunocm, 2021-12-13:
I also have the same problem, only the first endpoint works and it toggles all 3 switches.

I noticed we seem to be all Portuguese, could it be a problem specific to the version sold in LIDL Portugal? :thinking: 

Using Zigbee2MQTT version 1.22.1 and the Slaesh's CC2652RB stick as coordinator.

Theelepel88, 2021-12-19:
I have the same issue with a 07/2021 strip, only 1 switch that works, but controls all three. Other switched give the error message no endpoint available
Device 'ZigbeeStekker-Blok' has no endpoint 'l3'
ZIgbee2mqtt version 1.22.1 commit: c2b5229

nbranquinho, 2021-12-19:
I'm trying to read the Lidl Home application code, but so far I haven't found anything useful for this problem. I don't know if it's possible or not, to sniff the zigbee network to try to capture the communication between the APP (mobile phone) and the Zigbee network.
Has anyone done this?

MattWestb, 2021-12-19:
If you is having one EM35X or EFR32MGX coordinator or one TI CC-2531 with sniffing firmware you can sniffing all Zigbee traffic you like
For CC-2531 look on the Z2M guide and for EZSP devices https://github.com/zigpy/zigpy/wiki/Sniffing-Zigbee-traffic.

If you is paring the power strip and sniffing how the tuya ZBGW is "talking" with it i can looking in it and see if its doing some magic.

One thing is then you is posing one Zigbee sniffing then pairing one device its also possible see you current network key !!

BFFonseca, 2021-12-29:
> I also have the same problem, only the first endpoint works and it toggles all 3 switches.
> 
> I noticed we seem to be all Portuguese, could it be a problem specific to the version sold in LIDL Portugal? 🤔
> 
> Using Zigbee2MQTT version 1.22.1 and the Slaesh's CC2652RB stick as coordinator.

I agree this seems to be a Portuguese LIDL problem. The reports I've seen of this problem here and some other places are all from Portugal.
Could this be a firmware issue? I sure hope it's not a hardware revision to use less parts...

Can anyone connect their power strip to the official LIDL gateway and check if there are any firmware updates pending (if it even has that function). I can try this later this week.

Theelepel88, 2021-12-29:
I’m actually Dutch and have the same problem, with the cc2531 and the
Sonoff Zigbee 3 stick. Sold by lidl Netherlands during the Black Friday
sale
IAN 369230_2010 spsz3a1 version 07/2021 (25?)

nbranquinho, 2021-12-29:
> Can anyone connect their power strip to the official LIDL gateway and check if there are any firmware updates pending (if it even has that function). I can try this later this week.

I can confirm that: with the Lidl GW works normally.

> If you is having one EM35X or EFR32MGX coordinator or one TI CC-2531 with sniffing firmware you can sniffing all Zigbee traffic you like For CC-2531 look on the Z2M guide and for EZSP devices [...]

I don't have that hardware :-( I'm only using the CONBEE II and also have a SONOFF Zigbee with tasmota.

MattWestb, 2021-12-29:
Sonff ZBB and stick both can being used for sniffing Zigbee traffic so if having one tuya TBGW pleas sniff then paring the device with tuya ZBGW and posing the sniff so we can taking on look how they is doing there magic.

Zigpy Wiki how to sniffing with links to other guides and hardware https://github.com/zigpy/zigpy/wiki/Sniffing-Zigbee-traffic.

PS I have 2 early ones that is on the od firmware but LIDL is having them in this week for 25€ but i dont baying more them i dont need only the things that i dont have but i think the is one new batch with the new firmware and not only the PT version but we is going to knowing more in some days then more user is buying them.

JvD12, 2022-01-19:
Any update on this? I have multiple powerstrips but only the latest with app 70 has this problem.
But following the box and the website it should be possible: "Sockets can be controlled separately"
https://www.lidl.nl/p/silvercrest-usb-tafelcontactdoos-zigbee-smart-home/p100306994

svenahrensnl, 2022-01-19:
> Any update on this? I have multiple powerstrips but only the latest with app 70 has this problem. But following the box and the website it should be possible: "Sockets can be controlled separately" https://www.lidl.nl/p/silvercrest-usb-tafelcontactdoos-zigbee-smart-home/p100306994

Same issue over here (also in the Netherlands), bought the €25,- version in december and I can only control all the plugs at once. The older version (bought for €30,-) can control alle the plugs seperatly.

Theelepel88, 2022-02-19:
Still an issue.

coom, 2022-02-24:
Hi,
Same here with Zigbee2MQTT Edge. I've looked over Internet before buying this one and was pretty sure we can control each channel but it's not possible anymore.

Able to modify some files if you need someone to test.

Kind regards,

luminouw, 2022-02-24:
Hello,

I got a couple of this from LIDL (France) and I'm experiecing the same issue (3 switches reported, only first is functionnal but manages all channels simultaneously). I am using a usb coordinator, a Texas Instrument CC2531 CDC device (20180507).

I also gave it a shot within Home Assistant and its ZHA integration, using a Sonoff ZBBridge with less success (only reported one switch, turning on/off all channels at once).

I'm willing to help debug this with some guidance (only got so far as pairing devices and small debugging so far) if needed.

Cheers : )

MattWestb, 2022-02-24:
Sniffing then pairing it with one tuya ZBGW is the only thing that can reviling the "tuya magic" but very likely its the same as being found on the TS004F dimmer switch.

chiboule, 2022-02-25:
Someone has sniffed pairing with lidl gateway:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5493?fbclid=IwAR1mDixyEPEHQvFfktXE__vbpdYmQs5D_iPaqXVuUo35D25Iw8eX9sTiUts#issuecomment-1049679367

32 77.217605 0x0000 0x33e8 ZigBee 73 Transport Key
37 77.304642 0x0000 0x33e8 ZigBee HA 60 ZCL: Read Attributes, Seq: 26
41 77.340406 0x0000 0x33e8 ZigBee ZDP 54 Match Descriptor Request, Nwk Addr: 0x33e8, Profile: 0x0104
42 77.344490 0x33e8 0x0000 ZigBee 55 Route Record, Dst: 0x0000
44 77.348903 0x33e8 0x0000 ZigBee 45 APS: Ack, Dst Endpt: 1, Src Endpt: 1
46 77.355714 0x0000 0x33e8 ZigBee ZDP 54 Match Descriptor Request, Nwk Addr: 0x33e8, Profile: 0x0104
48 77.360761 0x33e8 0x0000 ZigBee ZDP 50 Match Descriptor Response, Nwk Addr: 0x33e8, Status: Success
50 77.365032 0x33e8 0x0000 ZigBee 45 APS: Ack, Dst Endpt: 0, Src Endpt: 0
52 77.368819 0x0000 0x33e8 ZigBee 45 APS: Ack, Dst Endpt: 0, Src Endpt: 0
54 77.383861 0x33e8 0x0000 ZigBee HA 100 ZCL: Read Attributes Response, Seq: 26
56 77.496106 0x0000 0x33e8 ZigBee ZDP 48 Active Endpoint Request, Nwk Addr: 0x33e8
59 77.586436 0x33e8 0x0000 ZigBee ZDP 54 Active Endpoint Response, Nwk Addr: 0x33e8, Status: Success
61 77.615392 0x0000 0x33e8 ZigBee ZDP 49 Simple Descriptor Request, Nwk Addr: 0x33e8, Endpoint: 1
63 77.619720 0x0000 0x33e8 ZigBee ZDP 49 Simple Descriptor Request, Nwk Addr: 0x33e8, Endpoint: 1
65 77.624341 0x33e8 0x0000 ZigBee 55 Route Record, Dst: 0x0000
67 77.667663 0x33e8 0x0000 ZigBee ZDP 72 Simple Descriptor Response, Nwk Addr: 0x33e8, Status: Success
70 77.689568 0x0000 0x33e8 ZigBee ZDP 49 Simple Descriptor Request, Nwk Addr: 0x33e8, Endpoint: 2
72 77.693622 0x33e8 0x0000 ZigBee ZDP 66 Simple Descriptor Response, Nwk Addr: 0x33e8, Status: Success
76 77.791111 0x0000 0x33e8 ZigBee ZDP 49 Simple Descriptor Request, Nwk Addr: 0x33e8, Endpoint: 3
78 77.796743 0x33e8 0x0000 ZigBee ZDP 66 Simple Descriptor Response, Nwk Addr: 0x33e8, Status: Success
81 77.873918 0x0000 0x33e8 ZigBee ZDP 49 Simple Descriptor Request, Nwk Addr: 0x33e8, Endpoint: 242
83 77.879582 0x33e8 0x0000 ZigBee 55 Route Record, Dst: 0x0000
85 77.927217 0x33e8 0x0000 ZigBee ZDP 60 Simple Descriptor Response, Nwk Addr: 0x33e8, Status: Success
87 78.009727 0x0000 0x33e8 ZigBee HA 52 ZCL: Write Attributes, Seq: 27
89 78.017570 0x33e8 0x0000 ZigBee HA 49 ZCL: Write Attributes Response, Seq: 27
92 78.068066 0x0000 0x33e8 ZigBee HA 52 ZCL: Write Attributes, Seq: 28
94 78.074163 0x33e8 0x0000 ZigBee HA 49 ZCL: Write Attributes Response, Seq: 28
98 79.611039 0x0000 0x33e8 ZigBee HA 64 ZCL: Read Attributes, Seq: 29
100 79.618361 0x33e8 0x0000 ZigBee 55 Route Record, Dst: 0x0000
102 79.621892 0x33e8 0x0000 ZigBee HA 84 ZCL: Read Attributes Response, Seq: 29
104 79.626481 0x33e8 0x0000 ZigBee HA 84 ZCL: Read Attributes Response, Seq: 29
106 79.633220 0x33e8 0x0000 ZigBee 45 APS: Ack, Dst Endpt: 1, Src Endpt: 3
108 79.687059 0x0000 0x33e8 ZigBee HA 54 ZCL: Read Attributes, Seq: 30
110 79.700514 0x33e8 0x0000 ZigBee 45 APS: Ack, Dst Endpt: 1, Src Endpt: 255
112 79.728811 0x33e8 0x0000 ZigBee HA 84 ZCL: Read Attributes Response, Seq: 29
114 79.744596 0x0000 0x33e8 ZigBee HA 50 ZCL: Read Attributes, Seq: 31
116 79.749522 0x33e8 0x0000 ZigBee 45 APS: Ack, Dst Endpt: 1, Src Endpt: 255
118 79.960060 0x33e8 0x0000 ZigBee HA 70 ZCL: Report Attributes, Seq: 69
120 79.993828 0x0000 0x33e8 ZigBee HA 50 ZCL: Default Response, Seq: 69
122 80.008494 0x33e8 0x0000 ZigBee HA 62 ZCL: Report Attributes, Seq: 70
124 80.053049 0x0000 0x33e8 ZigBee HA 50 ZCL: Default Response, Seq: 70
126 80.068121 0x33e8 0x0000 ZigBee HA 62 ZCL: Report Attributes, Seq: 71
128 80.119168 0x0000 0x33e8 ZigBee HA 50 ZCL: Default Response, Seq: 71
130 80.465417 0x33e8 0x0000 ZigBee ZDP 48 Node Descriptor Request, Nwk Addr: 0x0000
132 80.471961 0x0000 0x33e8 ZigBee ZDP 62 Node Descriptor Response, Rev: 22, Nwk Addr: 0x0000, Status: Success
134 80.476674 0x33e8 0x0000 ZigBee 58 Request Key
136 80.482132 0x33e8 0x0000 ZigBee 45 APS: Ack, Dst Endpt: 0, Src Endpt: 0
138 80.486994 0x0000 0x33e8 ZigBee 45 APS: Ack, Dst Endpt: 0, Src Endpt: 0
140 80.490458 0x0000 0x33e8 ZigBee 90 Transport Key
142 80.503702 0x33e8 0x0000 ZigBee 65 Verify Key
144 80.512457 0x0000 0x33e8 ZigBee 39 APS: Ack
146 80.516137 0x0000 0x33e8 ZigBee 67 Confirm Key, SUCCESS
148 80.523076 0x33e8 0x0000 ZigBee 48 APS: Ack
151 83.245567 0x33e8 0x0000 ZigBee HA 50 ZCL: Read Attributes, Seq: 72
153 83.283065 0x0000 0x33e8 ZigBee HA 56 ZCL: Read Attributes Response, Seq: 72
156 89.063597 0x33e8 0x0000 ZigBee HA 60 ZCL: Report Attributes, Seq: 73
158 89.095896 0x0000 0x33e8 ZigBee HA 50 ZCL: Default Response, Seq: 73
160 89.312141 0x33e8 0x0000 ZigBee HA 62 ZCL: Report Attributes, Seq: 74
162 89.359406 0x0000 0x33e8 ZigBee HA 50 ZCL: Default Response, Seq: 74
224 179.793750 0x0000 0x33e8 ZigBee HA 50 ZCL: Read Attributes, Seq: 32
226 179.798086 0x33e8 0x0000 ZigBee 55 Route Record, Dst: 0x0000
228 179.801524 0x33e8 0x0000 ZigBee 45 APS: Ack, Dst Endpt: 1, Src Endpt: 1
230 179.846138 0x33e8 0x0000 ZigBee HA 53 ZCL: Read Attributes Response, Seq: 32
234 184.871786 0x0000 0x33e8 ZigBee HA 50 ZCL: Read Attributes, Seq: 33
236 184.875695 0x33e8 0x0000 ZigBee HA 53 ZCL: Read Attributes Response, Seq: 33
238 184.879473 0x33e8 0x0000 ZigBee 45 APS: Ack, Dst Endpt: 1, Src Endpt: 1
241 189.952626 0x0000 0x33e8 ZigBee HA 50 ZCL: Read Attributes, Seq: 34
243 189.957263 0x0000 0x33e8 ZigBee HA 50 ZCL: Read Attributes, Seq: 34
245 189.961917 0x33e8 0x0000 ZigBee HA 53 ZCL: Read Attributes Response, Seq: 34
247 189.966993 0x33e8 0x0000 ZigBee 45 APS: Ack, Dst Endpt: 1, Src Endpt: 1

If it can help..

MattWestb, 2022-02-25:
Yes i have seen that but its not showing the attributes and commands with all flags that is needed for see what being dome under the hood.

MattWestb, 2022-02-25:
Hey @Element2 thanks for posting the briefly sniffing (in deCONZ git) of the LIDL power strip but can you pleas posting the capture file from wireshark so its possible to see all the flags and  parameters the tuya ZBGW is using then only "read attribute" is not helping.

Thanks in advance !!

Element2, 2022-02-25:
I'm willing to post any interesting frames here, but not the whole capture since the network key is in all the frames.

MattWestb, 2022-02-25:
Thanks Element2 !!
I knowing the problem and its the same posting the pin to your bank count of you Zigbee devices.

Can you folding out all feeds of frame 37, 54, 87, 89, 92, 94, 98, 102, 108, 112, 114, 151, 153, 224, 230, 234, 236, 241 and 245 (all attribute reading and replay from the device) and posting them or making on text file of them and attaching it in one post ?

I think the rest is not important then normally tuya is reading and writing attributes for making there **magic** and the rest is normal pairing what i can see.

MattWestb, 2022-02-25:
Sorry dont folding out Zigbee network security, And copy all shown elements then the network key is not being exposed.

Element2, 2022-02-25:
Here are the requested frames.
[frames.txt](https://github.com/Koenkk/zigbee2mqtt/files/8142079/frames.txt)

MattWestb, 2022-02-25:
The first is little of standard tuya "magic spell":
```
Coordinator request attributes:

    Command: Read Attributes (0x00)
    Attribute: Manufacturer Name (0x0004)
    Attribute: ZCL Version (0x0000)
    Attribute: Application Version (0x0001)
    Attribute: Model Identifier (0x0005)
    Attribute: Power Source (0x0007)
    Attribute: Unknown (0xfffe)

Device response:

    Command: Read Attributes Response (0x01)
    Status Record, String: _TZ3000_1obwwnmq
        Attribute: Manufacturer Name (0x0004)
        Status: Success (0x00)
        Data Type: Character String (0x42)
        String: _TZ3000_1obwwnmq
    Status Record, Uint8: 3
        Attribute: ZCL Version (0x0000)
        Status: Success (0x00)
        Data Type: 8-Bit Unsigned Integer (0x20)
        Uint8: 3 (0x03)
    Status Record, Uint8: 70
        Attribute: Application Version (0x0001)
        Status: Success (0x00)
        Data Type: 8-Bit Unsigned Integer (0x20)
        Uint8: 70 (0x46)
    Status Record, String: TS011F
        Attribute: Model Identifier (0x0005)
        Status: Success (0x00)
        Data Type: Character String (0x42)
        String: TS011F
    Status Record
        Attribute: Power Source (0x0007)
        Status: Success (0x00)
        Data Type: 8-Bit Enumeration (0x30)
        Power Source: Mains (single phase) (0x01)
    Status Record, Enum8: 0
        Attribute: Unknown (0xfffe)
        Status: Success (0x00)
        Data Type: 8-Bit Enumeration (0x30)
        Uint8: 0 (0x00)
```
That can being enough for some devices.
Next is one writing one attribute:
```
    Command: Write Attributes (0x02)
    Attribute Field, Uint8: 13
        Attribute: Unknown (0xffde)
        Data Type: 8-Bit Unsigned Integer (0x20)
        Uint8: 13 (0x0d)

Device responce:
    Command: Write Attributes Response (0x04)
    Status Record
        Status: Success (0x00)
```
No manufacture code or flags is being used.

I think only reading the attributes can being enough because in the "long sniff" the device is reporting all endpoints after that and the system is reading simple discretion from them.
If not working then try adding the attribute write and in 95% sure its working.

I cant testing then my Powers trips is still on the original firmware (Zigbee certificated one) so need some user that can testing it then have putting in the commands in the converter.

kkossev, 2022-02-25:
@MattWestb the standard tuya "magic spell" may be enough to make these LIDL powerstrip sockets be controlled individually!

Last days I had a very similar case with [this ](https://www.aliexpress.com/item/1005003821797706.html) device  ( combined wall switch with a socket for Brazil ) :

```
inClusters: 0003,0004,0005,0006,E000,E001,0000
outClusters: 0019,000A
model: TS0003
manufacturer: _TZ3000_vjhcenzo
```
The two switches and the one socket can be controlled individually using the standard OnOff cluster 0x006, but different endpoints 1, 2, and 3.
The problem was that sending OnOff command to endpoint 3 for example resulted in switching on or off also the first and the second subdevices!

Sending the 
```
Coordinator request attributes:

    Command: Read Attributes (0x00)
    Attribute: Manufacturer Name (0x0004)
    Attribute: ZCL Version (0x0000)
    Attribute: Application Version (0x0001)
    Attribute: Model Identifier (0x0005)
    Attribute: Power Source (0x0007)
    Attribute: Unknown (0xfffe)
```
during the Zigbee pairing process solved the issue.

MattWestb, 2022-02-25:
@kkossev Its more or less the standard for tuya device both "normal Zigbee" ones and its looks like some MCU / DP devices is needing the same spells for working "normal".
I think the list is around 10 now of devices  that is needing spell casting for working more normal.

Have you starting putting it as standard for all tuya devices in Hubitat or is you waiting for problems before adding it ?

Element2, 2022-02-25:
Don't know if this is relevant, but this new powerstrip has also problems with pairing and discovering all three endpoints in Phoscon.
Most of the time it shows only 1 outlet, sometimes 2 or 3. 
Anyway, I can compile and test if you want to try something.

Albenss, 2022-03-04:
I also have the same problem, only the first endpoint works and it toggles all 3 switches.
Zigbee2Mqtt 1.24

realPy, 2022-03-08:
Same problem as reported here with a new fresh device buy in france.
I'm succesfull patch it and it work now. I'm post the patch today.

realPy, 2022-03-08:
For people want to test now:
edit the zigbee-herdsman-converters/devices/lidl.js

for the device "HG06338" change the configure section:
`configure: async (device, coordinatorEndpoint, logger) => { 
 
            endpoint = device.getEndpoint(1); 
 
            await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]); 
            for (const ID of [1, 2, 3]) { 
                await reporting.bind(device.getEndpoint(ID), coordinatorEndpoint, ['genOnOff']); 
            } 
        }`

As @MattWestb mentionned you need read some attributes on endpoint1

realPy, 2022-03-08:
correct submit here:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/3967

luminouw, 2022-03-08:
Tested here, the device now works as expected, good job ! 👍

simaoafonso, 2022-03-08:
@luminouw you just edited the devices.js right? I did it and didnt worked for me.
sorry if it is a noob question

realPy, 2022-03-08:
> @luminouw you just edited the devices.js right? I did it and didnt worked for me. sorry if it is a noob question

You must edit the lidl.js and restart zigbee2mqtt

caloli74, 2022-03-08:
Thanks a lot, it works perfectly

luminouw, 2022-03-08:
@simaoafonso As @realPy said, I just edited the `lidl.js` to add the two lines and restarted zigbee2mqtt to make the change available. Associated the device and it worked right away : )

realPy, 2022-03-08:
> @simaoafonso As @realPy said, I just edited the `lidl.js` to add the two lines and restarted zigbee2mqtt to make the change available. Associated the device and it worked right away : )
Yes you are right i'm forgot to mention to remove device and reassociated

simaoafonso, 2022-03-08:
@realPy I did all that but not on the lidl.js, where can I find it?

realPy, 2022-03-08:
> @realPy I did all that but not on the lidl.js, where can I find it? Im using the home assistant addon and on the zigbee-herdsman-converters folder there is only the devices.js can someone tell me what im doing wrong?

I dont known how you have launch zigbee2mqtt.It must be in the node-modules/zigbee-herdsman-converters/devices/lidl.js
You can also search it with a find / | grep lidl.js
But this modification is more for advanced users. So may be the best things for you is wait the next release. I think @Koenkk will merge quickly this little fix.

simaoafonso, 2022-03-08:
> I dont known how you have launch zigbee2mqtt.It must be in the node-modules/zigbee-herdsman-converters/devices/lidl.js You can also search it with a find / | grep lidl.js But this modification is [...]

I dont have that devices folder I just have the devices.js. I´ll wait for the next realese, tanks!

Koenkk, 2022-03-08:
@realPy could you make a PR? Just click the pencil/edit icon here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices%2Flidl.js

closed 2022-03-08 by Koenkk

danielsticlan, 2024-05-26:
Hello.
If any of you know what happened to this zigbee extender (see photo and video) from Lidl and how I can fix it.... I would be grateful 🙏🙏🙏

Thank you in advance...
https://drive.google.com/file/d/10WD1rZmzGQOTBPuPCVMpVlC8GT4QiC0n/view?usp=sharinghttps://drive.google.com/file/d/10WD1rZmzGQOTBPuPCVMpVlC8GT4QiC0n/view?usp=sharing

[image]

MattWestb, 2024-05-26:
The power splay have one capacitor that is being bad (its start expanding and can blowing) and need being replaced but its not recommended to do if you is having the skill doing it right.
I have repairing  one of 2 i having plus one very old IKEA outlet (of around 10) and the material is under 1€.