Koenkk/zigbee2mqtt #9565: Upgraded from CC2531 to CC2652 - cannot add devices to zigbee network

realalexandergeorgiev, 2021-11-08:
## What happened
I did upgrade from CC2531 to CC2652. I did not change the pan_id or other values. The old config worked mostly flawless

## What did you expect to happen
Adding new devices to the network should work. Adding old devices as well.

## How to reproduce it (minimal and precise)
I removed a Xiaomi wireless switch and tried to add it - without success.

## Debug info
Zigbee2MQTT version: 1.22.0 commit: unknown
Adapter hardware: CC26X2R1
Adapter firmware version: 20210708 coordinator zstack3x0

I had no luck pairing an IKEA E1734 as well. Both new batteries, 30cm next to adapter.

Martin004, 2021-11-09:
Hello, I have the same hardware and it worked without problem.

It could be an problem from your Xiaomi switch. 
Try this:
In this case, you should first start the pairing process of the device according to the instructions, i.e. press the reset button for 10 seconds. As soon as the Smart Switch is in pairing mode, continue to press the reset button every second with a pointed object.

As a result, the Smart Switch regularly sends a sign of life and is (re) recognized by the gateway within a short time.

realalexandergeorgiev, 2021-11-09:
Thank you for the quick reply. 

Very strange. I have tried the IKEA switch as well and it didn't work. I have borrowed 2 Xiaomi switches and none is able to pair.

So my issue isn't related to the switches I guess...

Martin004, 2021-11-09:
check your configuration.yaml
important parameter from my one as example:
`homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://mqtt
serial:
  port: /dev/ttyUSB0
advanced:
  rtscts: false
  baudrate: 115200
  log_level: debug
  channel: 25
  network_key:
    - xxx
    - xxx
    - xx
    - xxx
    - xxx
    - xxx
    - xx
    - xxx
    - x
    - xxx
    - xxx
    - xx
    - xx
    - xx
    - xx
    - xxx
devices:
  '0x00158d000448ae93':
    friendly_name: PaulmannRGB2`

the network key is masked. There are normaly numbers ;)

perhaps send a debug log.

It could be important to pair the device near by the cc2652 stick (5cm distance).

realalexandergeorgiev, 2021-11-09:
I use mosquito, but the configuration seems similar. Rtscts is false, 115200 etc. I wouldnt be able to control my network with errors here, right? 

The log doesn't show anything for the joining devices. Only the other devices exchanging data.

Martin004, 2021-11-09:
That could be your problem. 
To use the zigbee stick you will need a software component to communicate with the stick. This is the zigbee2mqtt software from Koenkk. This component is connecteted to the broker mosquito. And your Home automation Software is connected to mosquito.
If you only see communication from mosquito and nothing from zigbee2mqtt. You will have a problem on the zigbee2mqtt component . 
Normaly if you find the configuration.yml file in your directory. In the same directory is an sub folder called "log". There you will find the debug log from zigbee2mqtt. 

My installation based on docker. So I can inspect all running containers.
https://github.com/ct-Open-Source/ct-Smart-Home

If your installation based also of docker. Inspect all running containers.

Martin004, 2021-11-09:
If you restart your zigbee component you will get a log like this:

`
debug 2021-11-09 14:35:58: Loaded state from file /app/data/state.json
info  2021-11-09 14:35:58: Logging to console and directory: '/app/data/log/2021-11-09.14-35-55' filename: log.txt
debug 2021-11-09 14:35:58: Removing old log directory '/app/data/log/2021-11-07.12-49-14'
info  2021-11-09 14:35:58: Starting Zigbee2MQTT version 1.22.0-dev (commit #6ba743c
)
info  2021-11-09 14:35:58: Starting zigbee-herdsman (0.13.166)
debug 2021-11-09 14:35:58: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[25],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"baudRate":115200,"path":"/dev/ttyUSB0","rtscts":false}}'
info  2021-11-09 14:35:59: zigbee-herdsman started (resumed)
info  2021-11-09 14:35:59: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}'
debug 2021-11-09 14:35:59: Zigbee network parameters: {"channel":25,"extendedPanID":"0xdddddddddddddddd","panID":6754}
info  2021-11-09 14:35:59: Currently 57 devices are joined:
info  2021-11-09 14:35:59: PaulmannRGB2 (0x00158d000448ae93): 371000002 - Paulmann Amaris LED panels (Router)
info  2021-11-09 14:35:59: PaulmannRGB1 (0x00158d000448ab66): 371000002 - Paulmann Amaris LED panels (Router)
info  2021-11-09 14:35:59: DoorSensor_2 (0x00158d0004148762): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2021-11-09 14:35:59: DoorSensor_1 (0x00158d0003e81487): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2021-11-09 14:35:59: DoorSensor_3 (0x00158d0003e787dc): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2021-11-09 14:35:59: DoorSensor_4 (0x00158d0003e8139d): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
`

realalexandergeorgiev, 2021-11-09:
Ok, and now?

`
debug 2021-11-09 15:36:49: Loaded state from file /share/zigbee2mqtt/state.json
info  2021-11-09 15:36:49: Logging to console and directory: '/share/zigbee2mqtt/log/2021-11-09.15-36-47' filename: log.txt
debug 2021-11-09 15:36:49: Removing old log directory '/share/zigbee2mqtt/log/2021-10-19.16-40-23'
info  2021-11-09 15:36:49: Starting Zigbee2MQTT version 1.22.0 (commit #unknown)
info  2021-11-09 15:36:49: Starting zigbee-herdsman (0.13.164)
debug 2021-11-09 15:36:49: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/share/zigbee2mqtt/coordinator
_backup.json","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","databasePath":"/share/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[22
1,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyUSB0"}}'
info  2021-11-09 15:36:50: zigbee-herdsman started (resumed)
info  2021-11-09 15:36:50: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x
0"}'
debug 2021-11-09 15:36:50: Zigbee network parameters: {"channel":11,"extendedPanID":"0x00124b0019369f5c","panID":6754}
info  2021-11-09 15:36:50: Currently 60 devices are joined: [..]
`

Martin004, 2021-11-09:
look the difference 
`"serialPort":{"path":"/dev/ttyUSB0"}}` 
and my version: 
`"serialPort":{"baudRate":115200,"path":"/dev/ttyUSB0","rtscts":false}}`

I think your configuration.yaml settings could be the problem.
Check my second post.

realalexandergeorgiev, 2021-11-09:
How would I be communicating with the devices if the baudrate was wrong?

Martin004, 2021-11-09:
However, there is a difference and my version works.
Try to eleminate the difference.

realalexandergeorgiev, 2021-11-09:
I appreciate your efforts in helping me, but your logic isnt right here. Its like asking me to paint my car green, because yours is and yours works :)

Martin004, 2021-11-09:
Nice analogy :)
I am not an expert and only try to isolate the problem by eliminating differences.
But maybe others have a better idea.

Regards

hnykda, 2021-11-09:
@realalexandergeorgiev I just did exactly the same thing - today upgraded from cc2531 to CCC2652 (slaesh), it works flawlessly, but I cannot pair any new devices, even if I am close to the coordinator (like IKEA repeater 10 cm from slaesh stick). Adding worked fine when I was on cc2531.

hnykda, 2021-11-10:
@Koenkk  And here is the sniffed traffic when:

1. waiting for 5 seconds after starting the sniff
2. putting coordinator into the pair mode (Allow join)
3. resetting ikea repeater (being 50 cm from the coordinator)

https://dropfiles.org/J022SQ50

Koenkk, 2021-11-10:
This looks very similar to https://github.com/Koenkk/zigbee2mqtt/issues/8803 and seems to be a bug in the TI firmware. TI released a new SDK firmware and I'm currently upgrading to it. Will notify in https://github.com/Koenkk/zigbee2mqtt/issues/8803 when it is ready for testing

hnykda, 2021-11-10:
Thank you :heart: ! Let me know if I can test or help with anything.

realalexandergeorgiev, 2021-11-10:
> This looks very similar to #8803 and seems to be a bug in the TI firmware. TI released a new SDK firmware and I'm currently upgrading to it. Will notify in #8803 when it is ready for testing

You are right. It totally sounds like the other issue. Apologies for not seeing it and opening this!! I subscribed the other issue and wait.

I guess I won't add new devices soon :/

hnykda, 2021-11-15:
I had to start from scratch: https://github.com/Koenkk/zigbee2mqtt/issues/8803#issuecomment-969190346

closed 2021-12-23 by github-actions[bot]