Koenkk/zigbee2mqtt #9566: Ikea LED2005R5 - starts at brightness 1

tb205gti, 2021-11-08:
## What happened

When turning on (wall switch, 230v) the Ikea LED2005R5 (GU10 345lm) - they start at brightness 1, but only if they have been turned of via zigbee first.

## What did you expect to happen

The led's should turn on at the previous level.

## How to reproduce it (minimal and precise)

1) Turn the led's on from the wall switch Turn them off by setting state=off. Now cut the power to the led's. wait 3 seconds, then power them back up. They will not start at brightness 1, and not the previous brightness.

2) Turn the led's on from the wall switch. Leave them for some seconds, not cut the power to them again. Wait 3 seconds - power them back on, no they will turn on with the correct (previous) brightness.

Subscribing to the mqtt topic I can see that immediately after turning on the led's I get three messages after the announce message ('{"message":"announce","meta":{"friendly_name":"IkeaGU10_1"},"type":"device_announced"}')

1: '{"brightness":210,"color_mode":"color_temp","color_temp":340,"color_temp_startup":65535,"linkquality":23,"power_on_behavior":"on","state":"ON","update":{"state":"idle"},"update_available":false}'

2: '{"brightness":1,"color_mode":"color_temp","color_temp":340,"color_temp_startup":65535,"linkquality":21,"power_on_behavior":"on","state":"ON","update":{"state":"idle"},"update_available":false}'

3: '{"brightness":1,"color_mode":"color_temp","color_temp":340,"color_temp_startup":65535,"linkquality":21,"power_on_behavior":"on","state":"ON","update":{"state":"idle"},"update_available":false}'

Now why does it send out the last two messages? If only it was the first, things would be correct.

## Debug info
Zigbee2MQTT version: 1.22.0 commit: 0a1fccf6
Adapter hardware: CC2531
Adapter firmware version: 20201127
Ikea LED2005R5 FW: 1.0.012

Otah, 2021-11-21:
Same here. I wonder if that't the problem of the bulb, rather... 🤷‍♂️ On power restore, I doubt there is anything Zigbee2mqtt can do. It really rather seems to me as the issue of the bulbs at this is happening to me from the very beginning.

tb205gti, 2021-12-25:
Still an issue

closed 2022-02-02 by github-actions[bot]