Koenkk/zigbee2mqtt #9573: Ikea LED1949C5 new device recently added but not recognized

aliessum, 2021-11-09:
## What happened
Tried to add this new Ikea LED1949C5. Pairing is ok, but it is not recognized. See log

## What did you expect to happen
I checked the zigbee-herdsman-converters/devices/ikea.js file and I can see that the device is recently added (at the end of the file).
## How to reproduce it (minimal and precise)
Tried to add the device the same way like I did several times before with others.
To check if I did not make a mistake I paired a Sonoff zigbee motion sensor with succes.

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531
Adapter firmware version: CC2531_SOURCE_ROUTING_2019061
Zigbee2MQTT version   1.18.1 commit: 1.18.1
Coordinator type zStack12
Coordinator revision 20190619
Frontend version 0.3.76

[image]
9

sjorge, 2021-11-10:
1.18.1 is rather old, it's probably not in that release. The latest is 1.22.0 which is a lot newer.

aliessum, 2021-11-10:
Hi Sjorge,
Thanks for your reply!
It seems that Z2M is not updated automatically from within HA. I found an issue about this (#7478).
After removing the repositorie and reinstalling, problem solved!
Thanks!

closed 2021-11-10 by aliessum