Koenkk/zigbee2mqtt #9579: Device friendly names lost on configuration change and restart - Docker environment

adams-family, 2021-11-09:
## What happened

First of all, I'm new to here and it's more than likely that I'm doing something wrong - it's just not obvious :(

All devices' friendly names are lost when `configuration.yaml` is edited and zigbee2mqtt is restarted with `docker-compose restart`

## What did you expect to happen

Devices' friendly names should be preserved.

## How to reproduce it (minimal and precise)

  1. Joined a couple of Ikea devices to zigbee2mqtt
  2. Assigned them friendly names
  3. Edited configuration.yaml
  4. Restarted zigbee2mqtt with `docker-compose restart`
  5. All friendly names are gone and all devices are reset to their original hexa-names

My `docker-compose.yml` does contain a persistent volume for `data`:

```
version: '2.1'

services:
  zigbee2mqtt:
    image: koenkk/zigbee2mqtt
    restart: always
    network_mode: "host"
    volumes:
      - ./data:/app/data
      - /run/udev:/run/udev:ro
    ports:
      - 8080:8080
    environment:
      - TZ=Europe/Berlin
    devices:
      - /dev/ttyUSB1:/dev/ttyACM0
```

## Debug info
Zigbee2MQTT version: 1.22.0 (commit #0a1fccf)
Adapter hardware: CC2652RB
Adapter firmware version: (not sure but I believe it will not depend on it)

closed 2021-12-17 by github-actions[bot]

MrCustomizer, 2021-12-17:
I think this ticket should be reopened.

I have the same problem. Also new to zigbee2mqtt, so it's possible I'm doing something wrong, too. I'm also using Docker. Any advice? Do you need some specific information to debug this?

Lyr3x, 2022-02-08:
I agree that this issue should be reopened. 
It does not make sense to update the configuration.yaml (which should be static). We already have a database in place, so IMO things like names and groups and even the state itself, should be persisted in the database as those are not part of the service configuration which is red once on startup, but application data.

One must manually update the configuration.yaml if using automated provisioning of a server or the devices are gone after a deployment.

kirel, 2022-10-09:
I have also experienced this. Where are friendly names stored right now?

kirel, 2022-10-09:
I've looked into the config and the problem seems to be that everything is stored in `configuration.yml` - I was overwriting this file when re-provisioning. Adding
```
devices: devices.yml
groups: groups.yml
```
like https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html#extract-config-to-separate-files solves my problem.

For future googlers.