Koenkk/zigbee2mqtt #9582: Multiple MQTT servers - failover MQTT

adub08, 2021-11-10:
## Is your feature request related to a problem? Please describe
In order to try and maximize redundancy on my home automation system, i have been setup a failover system for node-red and my MQTT broker. This would be simplified if zigbee2mqtt included the option for multiple MQTT brokers

## Describe the solution you'd like
Ideally i would like to be able to configure multiple MQTT brokers which zigbee2mqtt attempts to connect to in order, if one fails reconnection attempts then move to the next. 

## Describe alternatives you've considered
I have considered having a second host start a zigbee2mqtt server at the same IP address as the primary in docker if the main one goes offline, however this is not ideal, and would be simplified considerably and be more reliable if a second server could be defined in the zigbee2mqtt config.

## Additional context
One problem of having all parts of a home automated (such as lights, switches, heaters, aircon, blinds, etc), is if there is a failure in home assistant/node-red. Trying to remedy this has led me to setting up a second identical node-red instance on a Raspberry Pi, which can take over if the main host running node-red and the MQTT broker fails. I personally feel part of trying to get wide spread take up of zigbee2mqtt should include some thought into how to keep basic system functionality (i.e. turning on lights in the house), when parts of the system fallover.

Thanks.

hell1337hound, 2021-11-10:
yes please!!

danieledwardgeorgehitchcock, 2021-11-10:
This could possibly be achievable by a distributed instance of MQTT on a container orchestration platform such as Kubernetes. The distributed Kubernetes platform has a load-balancer and can be set up to failover a dead node to another instance. This way, you would not need to maintain multiple applications with various 'hot' failover addresses, you can use the native functionality of your apps and Kubernetes

bartplessers, 2021-11-14:
What I have done:
- first server with hass+mqtt
- second server with another instance of hass+mqtt server
- on my own internal dns:
  - users are pointed to first server as hass-frontend when surfing to 'hass.mydomain.org'
  - hass is using 'mqtt.mydomain.org' as mqtt-server
  - 'mqtt.mydomain.org' is pointing to ip of first server

When I want to failover, I just have change the definition in DNS, and let it point to my second server

Recently I'm publishing things over nginx. This makes it even simpler as I only need to redefine the backend ip of the nginx publishing rule (which can be done on the nginx frontend addon in hass itself...)

Grtz
B

onedr0p, 2021-12-01:
I achieve this in Kubernetes by using [emqx](https://github.com/emqx/emqx) and scaling it across my nodes. For anymore else who's into Kubernetes feel free to take my helm values and use emqx too! 

https://github.com/onedr0p/home-cluster/blob/e523a5198db49e4714beb565930ff26fda763a88/cluster/apps/home/emqx/helm-release.yaml#L23-L60

adub08, 2021-12-01:
Thanks everyone.
@onedr0p That looks real interesting, i haven't come across emqx before, i'll check it out and have a go at setting up kubernetes (only run docker before) but that looks like a good solution

onedr0p, 2021-12-01:
@adub08 If you ever want to go the k8s route, come join the k8s at home Discord. https://discord.gg/k8s-at-home

closed 2022-01-09 by github-actions[bot]