Koenkk/zigbee2mqtt #9586: Web UI Crashes under 1.22.0

thetravellor, 2021-11-10:
## What happened

Upgrade from 1.21.2 -> 1.22.0
systemctl loads daemon, but Web UI is not functioning: 

home@black:/opt/zigbee2mqtt/data/log/2021-11-10.19-12-49$ sudo systemctl status zigbee2mqtt
● zigbee2mqtt.service - zigbee2mqtt
     Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2021-11-10 19:37:35 AEDT; 24s ago
   Main PID: 98556 (npm start)
      Tasks: 23 (limit: 38291)
     Memory: 116.2M
     CGroup: /system.slice/zigbee2mqtt.service
             ├─98556 npm start
             ├─98576 sh -c node index.js
             └─98577 node index.js

reflashed stick with latest firmware, restarted zigbee2mqtt no change.

deleted zigbee2mqtt directory reinstalled from github, copied in data directory, restarted daemon, same result
stopped daemon via systemctl

created a docker-compose directory, created docker compose file, copied in data, brought docker instance up, same issue

## What did you expect to happen

Everything loads normally, WebUI to be available

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.22.0
Adapter hardware: CC2652R ((Electrolama)
Adapter firmware version: 20210708

Log.txt
info  2021-11-10 19:12:49: Logging to directory: '/opt/zigbee2mqtt/data/log/2021-11-10.19-12-49' filename: log.txt
info  2021-11-10 19:12:49: Starting Zigbee2MQTT version 1.22.0 (commit #b3e9afaf)
info  2021-11-10 19:12:49: Starting zigbee-herdsman (0.13.164)
info  2021-11-10 19:12:50: zigbee-herdsman started (resumed)
info  2021-11-10 19:12:50: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}'
info  2021-11-10 19:12:50: Currently 21 devices are joined:
info  2021-11-10 19:12:50: Desk Switch (0xb4e3f9fffe30e9f2): E1743 - IKEA TRADFRI ON/OFF switch (EndDevice)
info  2021-11-10 19:12:50: Desk Strip (0x000d6ffffeb9fd4f): ICPSHC24-10EU-IL-1 - IKEA TRADFRI driver for wireless control (10 watt) (Router)
info  2021-11-10 19:12:50: Vertical Strip (0x90fd9ffffe8385e4): ICPSHC24-10EU-IL-1 - IKEA TRADFRI driver for wireless control (10 watt) (Router)
info  2021-11-10 19:12:50: Stephen Bedside1 (0x588e81fffe26a7f0): LED1624G9 - IKEA TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white (Router)
info  2021-11-10 19:12:50: Stephen Bedside2 (0x680ae2fffe77730b): LED1624G9 - IKEA TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white (Router)
info  2021-11-10 19:12:50: TVRoom Strip (0x000d6ffffeb1ddc9): ICPSHC24-10EU-IL-1 - IKEA TRADFRI driver for wireless control (10 watt) (Router)
info  2021-11-10 19:12:50: Stephen Bedside Switch (0x680ae2fffe3a888e): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
info  2021-11-10 19:12:50: TV Room Switch (0xb4e3f9fffeb33af3): E1743 - IKEA TRADFRI ON/OFF switch (EndDevice)
info  2021-11-10 19:12:50: Gabbys Bedside1 (0x842e14fffea3205a): LED1624G9 - IKEA TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white (Router)
info  2021-11-10 19:12:50: Gabbys Bedside2 (0x842e14fffea1754d): LED1624G9 - IKEA TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white (Router)
info  2021-11-10 19:12:50: Repeater - Patio (0x842e14fffe6431a1): E1746 - IKEA TRADFRI signal repeater (Router)
info  2021-11-10 19:12:50: Elena's Bed Light (0xec1bbdfffeb44da2): ICPSHC24-10EU-IL-1 - IKEA TRADFRI driver for wireless control (10 watt) (Router)
info  2021-11-10 19:12:50: Girls Room Light (0x680ae2fffe794584): LED1624G9 - IKEA TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white (Router)
info  2021-11-10 19:12:50: Garage Thermometer (0x00158d0006bec264): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2021-11-10 19:12:50: Centre House Thermometer (0x00158d0006bec208): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2021-11-10 19:12:50: CC2532RB Router (0x00124b0023933c53): ti.router - Custom devices (DiY) Texas Instruments router (Router)
info  2021-11-10 19:12:50: Fish Tank Light (0x0017880108fb3f0c): 9290022408 - Philips Hue smart plug - AU (Router)
info  2021-11-10 19:12:50: Fish Tank Pump (0x0017880108fd87d4): 9290022408 - Philips Hue smart plug - AU (Router)
info  2021-11-10 19:12:50: Couch Light (0x84fd27fffe3764f0): LED1624G9 - IKEA TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white (Router)
info  2021-11-10 19:12:50: Desk Light 1 (0x0017880106bc548e): 9290012573A - Philips Hue white and color ambiance E26/E27/E14 (Router)
info  2021-11-10 19:12:50: Desk Light 2 (0x0017880106bc540f): 9290012573A - Philips Hue white and color ambiance E26/E27/E14 (Router)
warn  2021-11-10 19:12:50: `permit_join` set to  `true` in configuration.yaml.
warn  2021-11-10 19:12:50: Allowing new devices to join.
warn  2021-11-10 19:12:50: Set `permit_join` to `false` once you joined all devices.
info  2021-11-10 19:12:50: Zigbee: allowing new devices to join.
info  2021-11-10 19:12:50: Connecting to MQTT server at mqtt://localhost:1883
info  2021-11-10 19:12:50: Connected to MQTT server
info  2021-11-10 19:12:50: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2021-11-10 19:12:51: Configuring 'Stephen Bedside2'
info  2021-11-10 19:13:12: Successfully configured 'Stephen Bedside2'
info  2021-11-10 19:13:12: Configuring 'Gabbys Bedside1'
info  2021-11-10 19:13:34: MQTT publish: topic 'zigbee2mqtt/Stephens Bedside Lights', payload '{"brightness":254,"color":{"x":0.368,"y":0.269},"color_mode":"xy","state":"OFF"}'
info  2021-11-10 19:13:34: Successfully configured 'Gabbys Bedside1'
info  2021-11-10 19:13:34: Configuring 'Gabbys Bedside2'
error 2021-11-10 19:13:55: Publish 'set' 'state' to 'Vertical Strip' failed: 'Error: Command 0x90fd9ffffe8385e4/1 genOnOff.off({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
info  2021-11-10 19:13:55: Successfully configured 'Gabbys Bedside2'
info  2021-11-10 19:13:55: Configuring 'Girls Room Light'
error 2021-11-10 19:13:56: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of null (reading 'clients')
    at Frontend.stop (/opt/zigbee2mqtt/lib/extension/frontend.ts:57:39)
    at Controller.callExtensions (/opt/zigbee2mqtt/lib/controller.ts:312:40)
    at Controller.stop (/opt/zigbee2mqtt/lib/controller.ts:192:9)
    at stop (/opt/zigbee2mqtt/index.js:101:5)
    at process.handleQuit (/opt/zigbee2mqtt/index.js:107:9))
info  2021-11-10 19:13:56: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
info  2021-11-10 19:13:56: Disconnecting from MQTT server
info  2021-11-10 19:13:56: Stopping zigbee-herdsman...
info  2021-11-10 19:14:04: Stopped zigbee-herdsman

toto9262, 2021-11-11:
I have the same problem, I removed my external converters in config it works, but I reinstalled the old version

mixday, 2021-11-17:
I confirm also this issue. The frontend is not available.

closed 2021-12-21 by thetravellor