Koenkk/zigbee2mqtt #9592: Aqara Opple WXCJKG12LM and WXCJKG13LM (possibly WXCJKG11LM but I don't have those) aren't registering double click, hold and release, triple click

dnedeltchev, 2021-11-10:
## What happened
I had WXCJKG12LM and WXCJKG13LM configured with various routines for double, triple and hold/release actions. This is all in the kids room so I am not sure when it stopped working exactly. So it could have been a while. I just started to configure another WXCJKG12LM and noticed that the top 2 buttons only register a single click. I re-paired the device but that didn't help. Then I picked up the WXCJKG13LM knowing that this was configured with double, triple and hold/release routines but the same thing happened with it as well. So it looks like this stopped working some time back and re-pairing the devices does not fix the issue.

## What did you expect to happen
I had these working with single, double, triple and hold/release. Now it seems that single click is available on all buttons but double, triple and hold/release is available on the bottom buttons but not all where it was available on all buttons.

## How to reproduce it (minimal and precise)
1. Pair the device
2. change mode to command
3. test buttons

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC26X2R1 (zzh from Electrolama)
Adapter firmware version: 20210120 (it's not the latest version, yet)

dnedeltchev, 2021-12-11:
Test

closed 2021-12-19 by github-actions[bot]

thehijacker, 2021-12-24:
I have same issue. Today I have received WXCJKG11LM and it only registeres single click event.

```
Info MQTT publish: topic 'zigbee2mqtt/Aqara Server Room', payload '{"action":"button_1_single","battery":100,"linkquality":132,"operation_mode":null,"voltage":3000}'
Info MQTT publish: topic 'zigbee2mqtt/Aqara Server Room', payload '{"action":"","battery":100,"linkquality":132,"operation_mode":null,"voltage":3000}'
Info MQTT publish: topic 'zigbee2mqtt/Aqara Server Room/action', payload 'button_1_single'
Info MQTT publish: topic 'zigbee2mqtt/Aqara Server Room', payload '{"action":"button_2_single","battery":100,"linkquality":132,"operation_mode":null,"voltage":3000}'
Info MQTT publish: topic 'zigbee2mqtt/Aqara Server Room', payload '{"action":"","battery":100,"linkquality":132,"operation_mode":null,"voltage":3000}'
Info MQTT publish: topic 'zigbee2mqtt/Aqara Server Room/action', payload 'button_2_single'
```

How to debug this?

dnedeltchev, 2021-12-24:
> I have same issue. Today I have received WXCJKG11LM and it only registeres single click event.
> 
> ```
> Info MQTT publish: topic 'zigbee2mqtt/Aqara Server Room', payload '{"action":"button_1_single","battery":100,"linkquality":132,"operation_mode":null,"voltage":3000}'
> Info MQTT publish: topic 'zigbee2mqtt/Aqara Server Room', payload '{"action":"","battery":100,"linkquality":132,"operation_mode":null,"voltage":3000}'
> Info MQTT publish: topic 'zigbee2mqtt/Aqara Server Room/action', payload 'button_1_single'
> Info MQTT publish: topic 'zigbee2mqtt/Aqara Server Room', payload '{"action":"button_2_single","battery":100,"linkquality":132,"operation_mode":null,"voltage":3000}'
> Info MQTT publish: topic 'zigbee2mqtt/Aqara Server Room', payload '{"action":"","battery":100,"linkquality":132,"operation_mode":null,"voltage":3000}'
> Info MQTT publish: topic 'zigbee2mqtt/Aqara Server Room/action', payload 'button_2_single'
> ```
> 
> Enabled debug log and I can see following error when I double click:
> 
> ```
> Debug No converter available for 'WXCJKG11LM' with cluster 'genLevelCtrl' and type 'commandStep' and data '{"stepmode":1,"stepsize":85,"transtime":7}'
> Debug Received Zigbee message from 'Aqara Server Room', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":1,"stepsize":85,"transtime":7}' from endpoint 1 with groupID 0
> Debug No converter available for 'WXCJKG11LM' with cluster 'genLevelCtrl' and type 'commandStep' and data '{"stepmode":1,"stepsize":85,"transtime":7}'
> Debug Received Zigbee message from 'Aqara Server Room', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":1,"stepsize":85,"transtime":7}' from endpoint 1 with groupID 0
> Debug No converter available for 'WXCJKG11LM' with cluster 'genLevelCtrl' and type 'commandStep' and data '{"stepmode":1,"stepsize":85,"transtime":7}'
> Debug Received Zigbee message from 'Aqara Server Room', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":1,"stepsize":85,"transtime":7}' from endpoint 1 with groupID 0
> Debug No converter available for 'WXCJKG11LM' with cluster 'genLevelCtrl' and type 'commandStep' and data '{"stepmode":1,"stepsize":85,"transtime":7}'
> ```

I am not sure how to troubleshoot but have a look at this other thread. The bot keeps closing these issues and I am unable to reopen them hence I started a new one...
https://github.com/Koenkk/zigbee2mqtt/issues/8709#event-5593686183
The only thing with this approach is that there is a downside that I discovered. The battery reporting has gone haywire. It goes down to 0 then it's back to 100% then back down to 0... but at least the double and triple click, hold an release are back. It's just annoying and no one is looking into this. I take it this isn't high priority for enough people.

thehijacker, 2021-12-24:
@dnedeltchev: I also looked at that topic.

Tried everything. But sending "command" also always fails. And device is awake as I can press button just before and after sending mqtt message.

Tried to reconfigure. Delete. Pair again.... Nothing works. Lots of errors in debug log if I press double or hold any of the two buttons. It works fine with one of my older same type devices. All 6 button combination works with that one. With the new one only single left and right click.

dnedeltchev, 2021-12-24:
> @dnedeltchev: I also looked at that topic.
>
> Tried everything. But sending "command" also always fails. And device is awake as I can press button just before and after sending mqtt message.
>
> Tried to [...]

Have you tried allowing devices to join before you try to reconfigure? For me it worked with join allowed and disabled for all my devices old (purchased last year) and new (purchased this year in July).

thehijacker, 2021-12-24:
Yes. Tried that also. Pairing enabled or disabled.

```
Debug Publishing 'set' 'operation_mode' to 'Aqara Server Room'
Error Publish 'set' 'operation_mode' to 'Aqara Server Room' failed: 'Error: Write 0x04cf8cdf3c7c6ae0/1 aqaraOpple({"mode":0}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 34450 - 1 - 71 - 64704 - 4 after 10000ms)'
Debug Error: Write 0x04cf8cdf3c7c6ae0/1 aqaraOpple({"mode":0}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 34450 - 1 - 71 - 64704 - 4 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (internal/timers.js:557:17) at processTimers (internal/timers.js:500:7)
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'operation_mode' to 'Aqara Server Room' failed: 'Error: Write 0x04cf8cdf3c7c6ae0/1 aqaraOpple({\"mode\":0}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 34450 - 1 - 71 - 64704 - 4 after 10000ms)'","meta":{"friendly_name":"Aqara Server Room"},"type":"zigbee_publish_error"}'
```

```
Debug Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"Aqara Server Room","transaction":"n0x1t-9"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Aqara Server Room"},"status":"ok","transaction":"n0x1t-9"}'
```

I also see this in single click. Lots of messages repeat itself:

```
Info MQTT publish: topic 'zigbee2mqtt/Aqara Server Room/action', payload 'button_1_single'
Debug Received Zigbee message from 'Aqara Server Room', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Aqara Server Room', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Aqara Server Room', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Aqara Server Room', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Aqara Server Room', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Aqara Server Room', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Aqara Server Room', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
```

thehijacker, 2021-12-24:
Maybe @Koenkk can comment on this? And reopen this issue?

thehijacker, 2021-12-24:
OK. Fixed it. I unbind it from coordinator. Then issue event command. Now I get all events correctly.

Also before when I pressed single button and no other was working, it was turning on/off all the Ikea lights in the house. Maybe related.