Koenkk/zigbee2mqtt #9594: Xiaomi DJT11LM doesn't export "action" to Home Assistant

RubenKelevra, 2021-11-10:
## What happened

The documentation states, that there's an "action" field, containing keywords like "vibration", "drop", or "tilt".

This field is shown in Zigbee2MQTT, but not in HA.

## What did you expect to happen

This field should be exported to HA as well.

## Debug info
Zigbee2MQTT version: 1.21.2-dev commit: 7fed9e7
Coordinator type: zStack3x0
Coordinator revision: 20201026
Frontend version: 0.6.22
Adapter hardware: Slaesh's CC2652RB stick

Koenkk, 2021-11-10:
Make sure you don't have the `homeassistant_legacy_triggers` in your `configuration.yaml` (https://www.zigbee2mqtt.io/guide/configuration/homeassistant.html). Note that the `action` sensor will always appear empty: https://www.zigbee2mqtt.io/guide/faq/#why-is-the-action-property-always-empty

navycrow, 2021-11-11:
I have the same problem with this device. I understand the current operation but I do not find it consistent. According to me, there must be 3 actions: "tilt", "vibration", "drop" and a timeout which returns a value "none"

McGiverGim, 2021-11-11:
I've several of this sensors. And I think the problem is in Home Assistant.

For example, in this log for a "tilt", it seems zigbee2mqtt changes between null, tilt and blank. A little strange... but it writes tilt to the action topic, so it seems ok:

```log
Zigbee2MQTT:info  2021-11-11 12:06:48: MQTT publish: topic 'zigbee2mqtt/vibracion_silla_miguel', payload '{"action":null,"angle":6,"angle_x":-2,"angle_x_absolute":92,"angle_y":7,"angle_y_absolute":83,"angle_z":-82,"battery":91,"last_seen":"2021-11-11T12:06:48+01:00","linkquality":93,"sensitivity":"medium","strength":107,"voltage":2985}'
Zigbee2MQTT:info  2021-11-11 12:06:48: MQTT publish: topic 'zigbee2mqtt/vibracion_silla_miguel', payload '{"action":"tilt","angle":5,"angle_x":-2,"angle_x_absolute":92,"angle_y":7,"angle_y_absolute":83,"angle_z":-82,"battery":91,"last_seen":"2021-11-11T12:06:48+01:00","linkquality":96,"sensitivity":"medium","strength":107,"voltage":2985}'
Zigbee2MQTT:info  2021-11-11 12:06:48: MQTT publish: topic 'zigbee2mqtt/vibracion_silla_miguel', payload '{"action":"","angle":5,"angle_x":-2,"angle_x_absolute":92,"angle_y":7,"angle_y_absolute":83,"angle_z":-82,"battery":91,"last_seen":"2021-11-11T12:06:48+01:00","linkquality":96,"sensitivity":"medium","strength":107,"voltage":2985}'
Zigbee2MQTT:info  2021-11-11 12:06:48: MQTT publish: topic 'zigbee2mqtt/vibracion_silla_miguel/action', payload 'tilt'
Zigbee2MQTT:info  2021-11-11 12:06:49: MQTT publish: topic 'zigbee2mqtt/vibracion_silla_miguel', payload '{"action":null,"angle":5,"angle_x":-2,"angle_x_absolute":92,"angle_y":7,"angle_y_absolute":83,"angle_z":-82,"battery":91,"last_seen":"2021-11-11T12:06:49+01:00","linkquality":96,"sensitivity":"medium","strength":107,"voltage":2985}'
Zigbee2MQTT:info  2021-11-11 12:06:49: MQTT publish: topic 'zigbee2mqtt/vibracion_silla_miguel', payload '{"action":null,"angle":5,"angle_x":-2,"angle_x_absolute":92,"angle_y":3,"angle_y_absolute":87,"angle_z":-87,"battery":91,"last_seen":"2021-11-11T12:06:49+01:00","linkquality":93,"sensitivity":"medium","strength":107,"voltage":2985}'
```

But looking at Home Assistant, it seems it changes between "None" and empty:
[image]

I've several automations in Node Red, and they read the state without problem:
```json
		"new_state": {
			"entity_id": "sensor.vibracion_silla_miguel_accion",
			"state": "tilt",
			"attributes": {
				"action": "tilt",
				"angle": 5,
				"angle_x": -2,
				"angle_x_absolute": 92,
				"angle_y": 2,
				"angle_y_absolute": 88,
				"angle_z": -88,
				"battery": 91,
				"last_seen": "2021-11-11T12:19:03+01:00",
				"linkquality": 81,
				"sensitivity": "medium",
				"strength": 91,
				"voltage": 2985,
				"friendly_name": "Vibración silla Miguel",
				"icon": "mdi:gesture-double-tap"
			},
```

So it seems more a representation problem of Home Assistant. The state is in the object, but it last too few to see it.

jlaska, 2021-11-27:
Same issue here.  I do *not* have `homeassistant_legacy_triggers` defined in `configuration.yaml`

I see the following from zigvee2mqtt when triggering this sensor with vibration...

```
zigbee2mqtt             | Zigbee2MQTT:info  2021-11-27 18:17:36: MQTT publish: topic 'zigbee2mqtt/Laundry Room Vibration Sensor', payload '{"action":null,"angle":169,"angle_x":1,"angle_x_absolute":89,"angle_y":0,"angle_y_absolute":90,"angle_z":89,"battery":100,"linkquality":90,"sensitivity":"high","strength":8,"voltage":3035}'
zigbee2mqtt             | Zigbee2MQTT:info  2021-11-27 18:17:48: MQTT publish: topic 'zigbee2mqtt/Laundry Room Vibration Sensor', payload '{"action":"vibration","angle":169,"angle_x":1,"angle_x_absolute":89,"angle_y":0,"angle_y_absolute":90,"angle_z":89,"battery":100,"linkquality":87,"sensitivity":"high","strength":8,"voltage":3035}'
zigbee2mqtt             | Zigbee2MQTT:info  2021-11-27 18:17:48: MQTT publish: topic 'zigbee2mqtt/Laundry Room Vibration Sensor', payload '{"action":"","angle":169,"angle_x":1,"angle_x_absolute":89,"angle_y":0,"angle_y_absolute":90,"angle_z":89,"battery":100,"linkquality":87,"sensitivity":"high","strength":8,"voltage":3035}'
zigbee2mqtt             | Zigbee2MQTT:info  2021-11-27 18:17:48: MQTT publish: topic 'zigbee2mqtt/Laundry Room Vibration Sensor/action', payload 'vibration'
zigbee2mqtt             | Zigbee2MQTT:info  2021-11-27 18:17:52: MQTT publish: topic 'zigbee2mqtt/Laundry Room Vibration Sensor', payload '{"action":null,"angle":169,"angle_x":1,"angle_x_absolute":89,"angle_y":0,"angle_y_absolute":90,"angle_z":89,"battery":100,"linkquality":60,"sensitivity":"high","strength":8,"voltage":3035}'
```

It shows `action` values of `[null, "vibration", ""]`

RubenKelevra, 2021-12-30:
This is not stale, just haven't come around to check the notification yet.

closed 2022-02-07 by github-actions[bot]