Koenkk/zigbee2mqtt #9597: SONOFF BASICZBR3 doesn't responding to a change state from OFF to ON

padziu, 2021-11-10:
## What happened
I connected SONOFF BASICZBR3 (https://www.zigbee2mqtt.io/devices/BASICZBR3.html) without problems and is is visible on the network map but device doesn't respond to change the state.
Logs: https://www.toptal.com/developers/hastebin/pepoquboci.rust
The interesting fact is that physical button in device changes state but with zigbee this is not possible.
[image]
[image]

## What did you expect to happen
The switch should change status to on.
## How to reproduce it (minimal and precise)
Just change the state from off to on.
## Debug info
Zigbee2MQTT version: 1.22.0-2
Adapter hardware: zStack12 (CC2531)
Adapter firmware version: 20201127

ipeacocks, 2021-11-11:
I have absolutely the same device and it works well. Have you tried to repair it?

padziu, 2021-11-11:
I changed the channel from 11 to 25 and the device works fine.
[image]

closed 2021-11-11 by padziu