Koenkk/zigbee2mqtt #9599: Leedarson ZM350STW1TCF error setting color_temp_startup and brightness

drbios, 2021-11-10:
## What happened
Leedarson ZM350STW1TCF error setting color_temp_startup and brightness
## What did you expect to happen
set start up attrinutes after turn on the swith light
## How to reproduce it (minimal and precise)
setting the attribute in the zigbee2mqtt web page reports the error
## Debug info

Zigbee2MQTT version: 1.22.0-dev commit: cd3df10a
Adapter hardware: egony CC2652P RF-STAR zStack3x0 
Adapter firmware version: 20210708

Nov 10 21:15:01 HAGYC CRON[4025]: (pi) CMD (~/duckdns/duck.sh >/dev/null 2>&1)
Nov 10 21:16:59 HAGYC npm[3905]: #033[34mZigbee2MQTT:debug#033[39m 2021-11-10 21:16:59: Received MQTT message on 'zigbee2mqtt/0x00158d000225d479/set' with data '{"color_temp_startup":370}'
Nov 10 21:16:59 HAGYC npm[3905]: #033[34mZigbee2MQTT:debug#033[39m 2021-11-10 21:16:59: Publishing 'set' 'color_temp_startup' to '0x00158d000225d479'
Nov 10 21:16:59 HAGYC npm[3905]: #033[31mZigbee2MQTT:error#033[39m 2021-11-10 21:16:59: Publish 'set' 'color_temp_startup' to '0x00158d000225d479' failed: 'Error: Write 0x00158d000225d479/1 lightingColorCtrl({"startUpColorTemperature":370}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Nov 10 21:16:59 HAGYC npm[3905]: #033[34mZigbee2MQTT:debug#033[39m 2021-11-10 21:16:59: Error: Write 0x00158d000225d479/1 lightingColorCtrl({"startUpColorTemperature":370}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
Nov 10 21:16:59 HAGYC npm[3905]:     at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:314:23)
Nov 10 21:16:59 HAGYC npm[3905]:     at Endpoint.write (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:396:22)
Nov 10 21:16:59 HAGYC npm[3905]:     at Object.convertSet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:944:13)
Nov 10 21:16:59 HAGYC npm[3905]:     at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:247:36)
Nov 10 21:17:01 HAGYC CRON[4038]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Nov 10 21:17:26 HAGYC npm[3905]: #033[34mZigbee2MQTT:debug#033[39m 2021-11-10 21:17:26: Received MQTT message on 'zigbee2mqtt/0x00158d000225d479/set' with data '{"color_temp_startup":243}'
Nov 10 21:17:26 HAGYC npm[3905]: #033[34mZigbee2MQTT:debug#033[39m 2021-11-10 21:17:26: Publishing 'set' 'color_temp_startup' to '0x00158d000225d479'
Nov 10 21:17:26 HAGYC npm[3905]: #033[31mZigbee2MQTT:error#033[39m 2021-11-10 21:17:26: Publish 'set' 'color_temp_startup' to '0x00158d000225d479' failed: 'Error: Write 0x00158d000225d479/1 lightingColorCtrl({"startUpColorTemperature":243}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Nov 10 21:17:26 HAGYC npm[3905]: #033[34mZigbee2MQTT:debug#033[39m 2021-11-10 21:17:26: Error: Write 0x00158d000225d479/1 lightingColorCtrl({"startUpColorTemperature":243}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
Nov 10 21:17:26 HAGYC npm[3905]:     at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:314:23)
Nov 10 21:17:26 HAGYC npm[3905]:     at Endpoint.write (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:396:22)
Nov 10 21:17:26 HAGYC npm[3905]:     at Object.convertSet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:944:13)
Nov 10 21:17:26 HAGYC npm[3905]:     at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:247:36)

Koenkk, 2021-11-11:
`UNSUPPORTED_ATTRIBUTE` attribute indicates the device does not support it. I've disabled it now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

drbios, 2021-11-11:
> `UNSUPPORTED_ATTRIBUTE` attribute indicates the device does not support it. I've disabled it now.
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

This is sad, it is a necessary function that the light bulb remembers its state before the loss of energy.  Is there any other way to know if there is this ability to remember the state before the power outage?

Koenkk, 2021-11-11:
No, the bulb doesn't support this (at least not via the standard way, you could check if there is an option to enable this from the original bridge)

leroyloren, 2021-11-19:
Same issue for device https://github.com/Koenkk/zigbee2mqtt/issues/9599

` Publish 'set' 'color_temp_startup' to 'LED rack' failed: 'Error: Write xxx/3 lightingColorCtrl({"startUpColorTemperature":370}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

closed 2021-12-28 by github-actions[bot]

bgewehr, 2022-01-02:
Same for 

```
Zigbee Modell FL 120 C
Zigbee Hersteller innr
Beschreibung Color Flex LED strip 
Unterstützungsstatus unterstützt
Firmware Datum 20200822
Firmware Version 0x28002162
Hersteller Innr
Modell FL 120 C
```

```
Error Publish 'set' 'color_temp_startup' to '0x00158d00046206b9' failed: 'Error: Write 0x00158d00046206b9/1 lightingColorCtrl({"startUpColorTemperature":65535}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
```