Koenkk/zigbee2mqtt #9603: Availability feature is not working

Gooman-rus, 2021-11-11:
## What happened
I have tried to use device-availability feature (https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#availability-advanced-configuration) but it is not working. Nothing happens. No messages in zigbee2mqtt/[FRIENDLY_NAME]/availability topic.

## What did you expect to happen
Availability feature working as it described in documentation.

## How to reproduce it (minimal and precise)
Add `availability: true` option to the configuration and check mqtt topics.

## Debug info
Zigbee2MQTT version: 1.22.0-2 (homeassistant addon)
Adapter hardware: CC2538
Adapter firmware version: 20200211

hnykda, 2021-11-11:
I actually experience the same behavior.

bartplessers, 2021-11-17:
see also #9699
I have same behaviour, but if you rename a device, the topic appears...

Koenkk, 2021-11-17:
Could you provide the debug logging when starting z2m with this feature enabled?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Gooman-rus, 2021-11-18:
Here is my debug start log when I put `availability: true` to the configuration:

`debug 2021-11-18 08:42:22: Loaded state from file /config/zigbee2mqtt/state.json
info  2021-11-18 08:42:22: Logging to console and directory: '/config/zigbee2mqtt/log/2021-11-18.08-42-21' filename: log.txt
debug 2021-11-18 08:42:22: Removing old log directory '/config/zigbee2mqtt/log/2021-11-10.18-12-52'
info  2021-11-18 08:42:22: Starting Zigbee2MQTT version 1.22.0 (commit #unknown)
info  2021-11-18 08:42:22: Starting zigbee-herdsman (0.13.164)
debug 2021-11-18 08:42:22: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":true},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
info  2021-11-18 08:42:23: zigbee-herdsman started (resumed)
info  2021-11-18 08:42:23: Coordinator firmware version: '{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20200211,"transportrev":2},"type":"zStack30x"}'
debug 2021-11-18 08:42:23: Zigbee network parameters: {"channel":11,"extendedPanID":"0x00124b001940be43","panID":6754}
info  2021-11-18 08:42:23: Currently 36 devices are joined:
info  2021-11-18 08:42:23: Aqara_main_temp_hum_pres (0x00158d000349c2f8): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2021-11-18 08:42:23: Aqara_movement_room (0x00158d00042209f8): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
info  2021-11-18 08:42:23: Aqara_double_key (0x00158d00036d3dc9): WXKG02LM_rev2 - Xiaomi Aqara double key wireless wall switch (2018 model) (EndDevice)
info  2021-11-18 08:42:23: Socket_tablet (0x00158d00039fa6af): ZNCZ02LM - Xiaomi Mi power plug ZigBee (Router)
info  2021-11-18 08:42:23: Balcony_temp_hum (0x00158d00039ea56c): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)
info  2021-11-18 08:42:23: Aqara_movement_hallway (0x00158d00034cf394): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
info  2021-11-18 08:42:23: Socket_Olya (0x00158d0003531cff): ZNCZ02LM - Xiaomi Mi power plug ZigBee (Router)
info  2021-11-18 08:42:23: Socket_fridge (0x00158d00039c0362): ZNCZ02LM - Xiaomi Mi power plug ZigBee (Router)
info  2021-11-18 08:42:23: Socket_kettle (0x00158d00035f627f): ZNCZ02LM - Xiaomi Mi power plug ZigBee (Router)
info  2021-11-18 08:42:23: Socket_EU_boiler (0x04cf8cdf3c759c89): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2021-11-18 08:42:23: Geiger_dosimeter (0x00124b001ec82ed2): DIYRuZ_Geiger - DIYRuZ [DiY Geiger counter](https://modkam.ru/?p=1591) (Router)
info  2021-11-18 08:42:23: IKEA_bulb_bath_1_new (0x086bd7fffe06054e): LED1649C5 - IKEA TRADFRI LED bulb E12/E14/E17 400 lumen, dimmable warm white, chandelier opal (Router)
info  2021-11-18 08:42:23: Aqara_LED_room (0x00158d0002d755c6): ZNLDP12LM - Xiaomi Aqara smart LED bulb (Router)
info  2021-11-18 08:42:23: HEIMAN_1 (0x00158d0001a8db49): HS1RC-N - HEIMAN Smart remote controller (EndDevice)
info  2021-11-18 08:42:23: Aqara_lamp2_kitchen (0x00158d0003e74e71): ZNLDP12LM - Xiaomi Aqara smart LED bulb (Router)
info  2021-11-18 08:42:23: Aqara_lamp1_kitchen (0x00158d0003e74ee9): ZNLDP12LM - Xiaomi Aqara smart LED bulb (Router)
info  2021-11-18 08:42:23: Aqara_bath_temp_hum_pres (0x00158d00022fd187): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2021-11-18 08:42:23: Socket_EU_wash (0x04cf8cdf3c763ca0): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (EndDevice)
info  2021-11-18 08:42:23: Aqara_water_leakage (0x00158d000242c4ce): SJCGQ11LM - Xiaomi Aqara water leak sensor (EndDevice)
info  2021-11-18 08:42:23: Aqara_contact (0x00158d000312e746): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2021-11-18 08:42:23: One_key_btn_kitchen (0x00158d00026db4f3): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
info  2021-11-18 08:42:23: Movement_kitchen (0x00158d0003cc36b2): RTCGQ01LM - Xiaomi MiJia human body movement sensor (EndDevice)
info  2021-11-18 08:42:23: One_key_btn_bath (0x00158d000363969a): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
info  2021-11-18 08:42:23: Kitchen_temp_hum (0x00158d00033e7c4f): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)
info  2021-11-18 08:42:23: Socket_EU_room_main (0x04cf8cdf3c867ea8): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2021-11-18 08:42:23: Outside_temp_hum (0x00158d0003cb8522): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)
info  2021-11-18 08:42:23: IKEA_bulb_bath_3 (0x086bd7fffe2c634e): LED1649C5 - IKEA TRADFRI LED bulb E12/E14/E17 400 lumen, dimmable warm white, chandelier opal (Router)
info  2021-11-18 08:42:23: IKEA_bulb_bath_2 (0x000d6ffffefd8e25): LED1649C5 - IKEA TRADFRI LED bulb E12/E14/E17 400 lumen, dimmable warm white, chandelier opal (Router)
info  2021-11-18 08:42:23: Contact_leakage_bath (0x00158d000423f442): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
info  2021-11-18 08:42:23: Flower_1 (0x00124b002260a15c): DIYRuZ_Flower - DIYRuZ [Flower sensor](http://modkam.ru/?p=1700) (EndDevice)
[... cut ...]
debug 2021-11-18 08:42:24: Received MQTT message on 'zigbee2mqtt/Aqara_bath_temp_hum_pres' with data '{"battery":38,"humidity":57.68,"linkquality":84,"pressure":1017.4,"temperature":21.9,"voltage":2875}'
debug 2021-11-18 08:42:24: Received MQTT message on 'zigbee2mqtt/Flower_1' with data '{"battery":11,"humidity":0,"illuminance":5413,"illuminance_lux":3,"linkquality":39,"pressure":0,"soil_moisture":89,"temperature_bme":0,"temperature_ds":0,"voltage":2600}'
warn  2021-11-18 08:42:24: Using experimental new availability feature
info  2021-11-18 08:42:24: Started frontend on port 0.0.0.0:8099`

Koenkk, 2021-11-18:
@Gooman-rus could you check if it works in the edge variant of the hass container? I see that I didn't add those in the normal variant. If I know it works in edge I can add it to the normal variant.

Gooman-rus, 2021-11-19:
@Koenkk can installing of edge version somehow damage or change my config of the original version of z2m?

Koenkk, 2021-11-19:
@Gooman-rus no, you can freely switch between those versions

Gooman-rus, 2021-11-19:
@Koenkk I can confirm that `availability: true` feature works with z2m-edge.

Koenkk, 2021-11-19:
great, this should be fixed now in the 1.22.0-3 addon.

juliend20, 2021-12-21:
At first. @Koenkk Thank you so much for your work. I found a similar issue some of my sensor (battery powered) have no information of availability. But I received a state update from it. The last communication of device is below of availability timeout.

closed 2022-02-10 by github-actions[bot]