Koenkk/zigbee2mqtt #9604: Hybrid E-Wand CP180335E getting error "Failed to interview" when paired

gregjwinter, 2021-11-11:
# Bug Report
Hybrid E-Wand CP180335E getting error "Failed to interview" when paired
## What happened
E-Wand joined OK but does not complete interview process.
Failed to interview '0x588e81fffe9710d6', device has not successfully been paired
## What did you expect to happen
All my other Zigbee devices paired OK from HUE and Leviton but not E-wand from Current Products Corp.
## How to reproduce it (minimal and precise)
Happens when I try to pair the E-Wand device. Tried 5X Throws the following error:
Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Received Zigbee message from '0x588e81fffe9710d6', type 'raw', cluster '64528', data '{"data":[13,99,18,62,128,192,0,0,0,0,2,0,0,0,136,225,64,0,0,183,65,0,255,0,0,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Received Zigbee message from '0x588e81fffe9710d6', type 'attributeReport', cluster 'genPowerCfg', data '{"11":10,"12":11,"2":1200,"3":3,"4":"00.00.18","5":255,"6":22,"7":"cpc:ewand","mainsFrequency":300,"mainsVoltage":4}' from endpoint 196 with groupID 0
## Debug Info
Zigbee2MQTT version: 1.22.0-2
Adapter hardware: CircuitSetup's CC2652P2 USB Coordinator
Adapter firmware version: 20210708

Koenkk, 2021-11-17:
Could you provide the herdsman debug logging when trying to pair this device?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

gregjwinter, 2021-11-17:
Thank you for looking at this.  I am attaching file to this message.   Just in case it does not come through here is the main part:

2021-11-17T20:28:34.006Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1,"srcaddr":27654,"srcendpoint":196,"dstendpoint":1,"wasbroadcast":1,"linkquality":18,"securityuse":0,"timestamp":4667442,"transseqnumber":0,"len":64,"data":{"type":"Buffer","data":[0,77,10,7,0,66,9,99,112,99,58,101,119,97,110,100,4,0,66,8,48,48,46,48,48,46,49,56,5,0,32,255,6,0,33,22,0,0,0,32,4,1,0,33,44,1,2,0,33,176,4,3,0,32,3,11,0,33,10,0,12,0,32,11]}}
2021-11-17T20:28:34.008Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":77,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":7,"dataType":66,"attrData":"cpc:ewand"},{"attrId":4,"dataType":66,"attrData":"00.00.18"},{"attrId":5,"dataType":32,"attrData":255},{"attrId":6,"dataType":33,"attrData":22},{"attrId":0,"dataType":32,"attrData":4},{"attrId":1,"dataType":33,"attrData":300},{"attrId":2,"dataType":33,"attrData":1200},{"attrId":3,"dataType":32,"attrData":3},{"attrId":11,"dataType":33,"attrData":10},{"attrId":12,"dataType":32,"attrData":11}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":27654,"endpoint":196,"linkquality":18,"groupID":0,"wasBroadcast":true,"destinationEndpoint":1}'
2021-11-17T20:28:34.012Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-11-17T20:28:35.876Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,32,68,129,0,0,25,0,87,238,11,13,0,54,0,193,5,73,0,0,12,1,85,1,0,11,16,21,1,0,11,0]
2021-11-17T20:28:35.876Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,32,68,129,0,0,25,0,87,238,11,13,0,54,0,193,5,73,0,0,12,1,85,1,0,11,16,21,1,0,11,0]
2021-11-17T20:28:35.878Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [1,87,238,29,0]
2021-11-17T20:28:35.878Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,32,68,129,0,0,25,0,87,238,11,13,0,54,0,193,5,73,0,0,12,1,85,1,0,11,16,21,1,0,11,0,1,87,238,29,0]
2021-11-17T20:28:35.879Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 32 - 2 - 4 - 129 - [0,0,25,0,87,238,11,13,0,54,0,193,5,73,0,0,12,1,85,1,0,11,16,21,1,0,11,0,1,87,238,29] - 0
2021-11-17T20:28:35.879Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":61015,"srcendpoint":11,"dstendpoint":13,"wasbroadcast":0,"linkquality":54,"securityuse":0,"timestamp":4785601,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,85,1,0,11,16,21,1,0,11,0,1]}}
2021-11-17T20:28:35.883Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":85,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"fieldControl":0,"manufacturerCode":4107,"imageType":277,"fileVersion":16780032},"Command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}},"address":61015,"endpoint":11,"linkquality":54,"groupID":0,"wasBroadcast":false,"destinationEndpoint":13}'
2021-11-17T20:28:35.885Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-11-17T20:28:43.638Z zigbee-herdsman:controller:device:log Interview - quirks check for 'undefined'-'undefined'-'EndDevice'
2021-11-17T20:28:43.638Z zigbee-herdsman:controller:device:log Interview - quirks did not match
2021-11-17T20:28:43.643Z zigbee-herdsman:controller:device:log Interview - failed for device '0x588e81fffe9710d6' with error 'Error: AREQ - ZDO - simpleDescRsp after 10000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)'
2021-11-17T20:28:43.650Z zigbee-herdsman:controller:error Interview failed for '0x588e81fffe9710d6 with error 'Error: AREQ - ZDO - simpleDescRsp after 10000ms'
Zigbee2MQTT:error 2021-11-17 15:28:43: Failed to interview '0x588e81fffe9710d6', device has not successfully been paired

[PairError_2021-11-17.txt](https://github.com/Koenkk/zigbee2mqtt/files/7557844/PairError_2021-11-17.txt)

Koenkk, 2021-11-17:
Do you by any chance have the original gateway? If yes, could you make a sniff when pairing this device to it? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

gregjwinter, 2021-11-18:
I don't - I have the Circuit Setup's CC2652P2 USB Coordinator.  I also have a ZZH! coming in someday...  If you can point me to the product you are talking about I can buy one if they are still available?  Thanks.

Koenkk, 2021-11-18:
- @gregjwinter can you try pairing the device very close to the coordinator?
- @scottpk @Leah-Anderson  is there any special pairing procedure needed for this device?

gregjwinter, 2021-11-18:
I did try 5x with each change.  First I did try pairing right up next to coordinator in my basement.  Then I tried the USB extension cable, tried pairing again.  Finally tried to pair next to close by repeater plugged into wall.  The E-wand makes the ode to joy tune to indicate successful pair each time too.  Also replaced batteries with same result.  Keep getting the interview failed message. Thank you.

scottpk, 2021-11-18:
No, I didn't follow any special procedure except what was in the manual. That said I'm pretty sure all mine are Gen 2 (they show as CP180335E-01, not sure if that is different than CP180335E or what they've changed between Gen 1 and Gen 2).

@gregjwinter are these new? I had to replace one that kept dropping off the network, and had a very positive experience with Current Products Corp's support. Soon as I explained my issue and gave my order number a fedex package was on the way. They may be able to help you if there's something wrong with your E-Wand.

https://www.currentproductscorp.com/e-wand-support

gregjwinter, 2021-11-18:
Yes I purchased one unit from Amazon. There is a tape seal that I had to cut through to open the box.  It has "A ZigBee Certified Product on the box.  First thing I did when I had this issue is call support at Current Products Corp to verify that I did indeed have a Gen 2 unit.  He asked me to unscrew the battery cover and see if the grommet is blue vs black.  I have blue grommet which is Gen 2 unit.  My part number printed in small print on unit is: CP180355E-01 serial 2408210007.

scottpk, 2021-11-18:
I think I bought mine off Amazon as well, they gave me no trouble. If you
haven't already I would try the factory reset procedure from the manual,
otherwise you may have a faulty unit.

gregjwinter, 2021-11-18:
I just tried 2x factory reset and tried to pair.  I will contact manufacturer now.  Thank you. -Greg.l.

gregjwinter, 2021-11-19:
Good news.  I have an advanced replacement on the way and arriving on Tueaday!  Will report back.

gregjwinter, 2021-11-26:
@Koenkk
@scottpk 
Received a brand new (direct from Current Products Corp) E-Wand with blue O-ring ZigBee unit and just tried to pair it.  It fails and get the same error "Failed to interview" when I try to pair it.  The LQI shows up as 153.  Tried close to repeater and next to coordinator 3X.  Here is the main error here:

2021-11-26T01:58:21.942Z zigbee-herdsman:controller:error Interview failed for '0x588e81fffe9711ab with error 'Error: AREQ - ZDO - simpleDescRsp after 10000ms'

Is there anything else I can do?
Here is a complete dump from the logs:

2021-11-26T01:58:11.880Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 7 - 2 - 5 - 133 - [156,173,0,156,173,1,1] - 199
2021-11-26T01:58:11.881Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - activeEpRsp - {"srcaddr":44444,"status":0,"nwkaddr":44444,"activeepcount":1,"activeeplist":[1]}
2021-11-26T01:58:11.881Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-11-26T01:58:11.885Z zigbee-herdsman:controller:device:log Interview - got active endpoints for device '0x588e81fffe9711ab'
2021-11-26T01:58:11.886Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - simpleDescReq - {"dstaddr":44444,"nwkaddrofinterest":44444,"endpoint":196}
2021-11-26T01:58:11.886Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,4,156,173,156,173,196,224]
2021-11-26T01:58:11.901Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,84,68,129,0,0,1,0,156,173,196,1,1,150,0,142,44,135,0,0,64,0,17,10,7,0,66,9,99,112,99,58]
2021-11-26T01:58:11.902Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,84,68,129,0,0,1,0,156,173,196,1,1,150,0,142,44,135,0,0,64,0,17,10,7,0,66,9,99,112,99,58]
2021-11-26T01:58:11.904Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [101,119,97,110,100,4,0,66,8,48,48,46,48,48,46,49,56,5,0,32,255,6,0,33,13,0,0,0,32,4,1,0]
2021-11-26T01:58:11.904Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,84,68,129,0,0,1,0,156,173,196,1,1,150,0,142,44,135,0,0,64,0,17,10,7,0,66,9,99,112,99,58,101,119,97,110,100,4,0,66,8,48,48,46,48,48,46,49,56,5,0,32,255,6,0,33,13,0,0,0,32,4,1,0]
2021-11-26T01:58:11.908Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [33,44,1,2,0,33,176,4,3,0,32,3,11,0,33,10,0,12,0,32,11,188,108,10,6]
2021-11-26T01:58:11.908Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,84,68,129,0,0,1,0,156,173,196,1,1,150,0,142,44,135,0,0,64,0,17,10,7,0,66,9,99,112,99,58,101,119,97,110,100,4,0,66,8,48,48,46,48,48,46,49,56,5,0,32,255,6,0,33,13,0,0,0,32,4,1,0,33,44,1,2,0,33,176,4,3,0,32,3,11,0,33,10,0,12,0,32,11,188,108,10,6]
2021-11-26T01:58:11.908Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 84 - 2 - 4 - 129 - [0,0,1,0,156,173,196,1,1,150,0,142,44,135,0,0,64,0,17,10,7,0,66,9,99,112,99,58,101,119,97,110,100,4,0,66,8,48,48,46,48,48,46,49,56,5,0,32,255,6,0,33,13,0,0,0,32,4,1,0,33,44,1,2,0,33,176,4,3,0,32,3,11,0,33,10,0,12,0,32,11,188,108,10] - 6
2021-11-26T01:58:11.909Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1,"srcaddr":44444,"srcendpoint":196,"dstendpoint":1,"wasbroadcast":1,"linkquality":150,"securityuse":0,"timestamp":8858766,"transseqnumber":0,"len":64,"data":{"type":"Buffer","data":[0,17,10,7,0,66,9,99,112,99,58,101,119,97,110,100,4,0,66,8,48,48,46,48,48,46,49,56,5,0,32,255,6,0,33,13,0,0,0,32,4,1,0,33,44,1,2,0,33,176,4,3,0,32,3,11,0,33,10,0,12,0,32,11]}}
2021-11-26T01:58:11.911Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":17,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":7,"dataType":66,"attrData":"cpc:ewand"},{"attrId":4,"dataType":66,"attrData":"00.00.18"},{"attrId":5,"dataType":32,"attrData":255},{"attrId":6,"dataType":33,"attrData":13},{"attrId":0,"dataType":32,"attrData":4},{"attrId":1,"dataType":33,"attrData":300},{"attrId":2,"dataType":33,"attrData":1200},{"attrId":3,"dataType":32,"attrData":3},{"attrId":11,"dataType":33,"attrData":10},{"attrId":12,"dataType":32,"attrData":11}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":44444,"endpoint":196,"linkquality":150,"groupID":0,"wasBroadcast":true,"destinationEndpoint":1}'
2021-11-26T01:58:11.914Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-11-26T01:58:11.916Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,4,0,96]
2021-11-26T01:58:11.916Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,4,0,96]
2021-11-26T01:58:11.916Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 4 - [0] - 96
2021-11-26T01:58:11.916Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - simpleDescReq - {"status":0}
2021-11-26T01:58:11.917Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-11-26T01:58:12.423Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,192,156,173,131,52]
2021-11-26T01:58:12.424Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,192,156,173,131,52]
2021-11-26T01:58:12.424Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 192 - [156,173,131] - 52
2021-11-26T01:58:12.424Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - stateChangeInd - {"state":156}
2021-11-26T01:58:12.424Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
[... cut ...]
2021-11-26T01:59:51.870Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":16,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":0}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":50984,"endpoint":11,"linkquality":33,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2021-11-26T01:59:51.872Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

Koenkk, 2021-11-26:
Could you make sniff when trying to pair this device?https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

gregjwinter, 2021-11-27:
@Koenkk - do I need to buy a CC2531 adapter for this purpose (sniffing)?  I do have 2 different CC2652R1 units: zzh! and CircuitSetup's CC2652P2 USB Coordinator.  If these do not work it would be easier to get a HUSBZB-1 stick on Amazon.  Let me know and I'll order one.  Thanks!

Koenkk, 2021-11-27:
@gregjwinter sniffing using a C2652 seems to be possible with packet sniffer 2 (haven't tried this myself yet). fws are bundled with packet sniffer 2: https://software-dl.ti.com/simplelink/esd/simplelink_cc13x2_26x2_sdk/3.20.00.68/exports/docs/zigbee/html/zigbee/packet_sniffer.html#hardware-setup

scottpk, 2021-11-28:
Sorry, I'm still relatively new to all this ZigBee and custom home automation stuff so I'm not sure how much help I can be, other than to say, I had no trouble going through the initial setup with these, no trouble at all aside from the one they replaced for me through warranty, and my setup is this:

- unRAID server 6.9.2
  - Ryzen 5 2600
  - 64GB DDR4
- HomeAssistant OS running as a VM
  - 2GB RAM
  - 32GB storage
- USB devices passed through:
  - Dresden Electronic ConBee II
  - Broadcom Corp BCM2046B1 (that's bluetooth so probably irrelevant to this)

Mosquitto Broker and Zigbee2MQTT are both running on the HAOS VM

I think these are the only flags in my Zigbee2MQTT config which aren't default:
```
advanced:
  last_seen: ISO_8601
  legacy_api: false
experimental:
  new_api: true
```

gregjwinter, 2021-12-02:
@scottpk @Koenkk - some news that is positive this time!
First I purchased a new coordinator - Electrolama zig-a-zig-ah! (zzh!) and replaced the CircuitSetup's CC2652P2 USB Coordinator I was using.  Paired my replacement E-wand - and it paired, sort of.  It seems to work, however there is a small red triangle with a bang on it. (s
[image]
ee
When I hover over the icon it says "Interview Failed".   I also set-up loopback traffic capture in Wireshark and can sniff Zigbee traffic.  What data do you need from this?  Happy to help resolve this last hurdle.  Thanks.

gregjwinter, 2021-12-03:
@Koenkk - how do I capture the pairing of the E-wand and send the data to you?  I have this in my config:
zigbee_herdsman_debug: true

There is a "log" tab in Home Assistant: Supervisor --> MQTT2Zigbee but it only captures 100 events and it missed the pairing event.  I also used Wireshark to capture a pcapng type file with all the events.  How do I send this to you?  Thanks. -Greg.

scottpk, 2021-12-03:
I think these would be the instructions to follow for capturing the logging:
https://www.zigbee2mqtt.io/guide/usage/debug.html#enabling-logging
It says zigbee-herdsman outputs straight to system IO but it may be possible to pipe it to an output file.

gregjwinter, 2021-12-03:
@scottpk I can see zigbee-herdman events that show up in the log tab of the zigbee2mqqt addon but it seems to miss the pairing event.  I don't know how to grab the past events.  I am running Home Assistant OS/Supervised addon and when I go into Terminal I try to enter "docker ps" under the [core-ssh ~]$ it replies "bash: docker: command not found"

Any ideas of what I can do?  I am stuck.
@Koenkk - I did get the sniff events when I tried to pair the E-Wand.  They are here:
[e-wandtext.txt](https://github.com/Koenkk/zigbee2mqtt/files/7651652/e-wandtext.txt)
 
Feel like we are getting close, at least it recognizes the manufacture/device.  I just fails the interview process.  Thanks guys. -Greg.

gregjwinter, 2021-12-03:
I think I captured the herdman log of the device pairing finally.

**This is the most interesting part:**
2021-12-03T19:39:19.032Z zigbee-herdsman:controller:device:log Interview - quirks check for 'E-Wand'-'Current Products Corp'-'EndDevice'
2021-12-03T19:39:19.034Z zigbee-herdsman:controller:device:log Interview - quirks did not match
2021-12-03T19:39:19.036Z zigbee-herdsman:controller:device:log Interview - failed for device '0x588e81fffe9710d6' with error 'Error: AREQ - ZDO - simpleDescRsp after 10000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)'
2021-12-03T19:39:19.041Z zigbee-herdsman:controller:error Interview failed for '0x588e81fffe9710d6 with error 'Error: AREQ - ZDO - simpleDescRsp after 10000ms'
Zigbee2MQTT:error 2021-12-03 14:39:19: Failed to interview '0x588e81fffe9710d6', device has not successfully been paired

**Attaching the whole thing:**
[e-wandtext.txt](https://github.com/Koenkk/zigbee2mqtt/files/7651981/e-wandtext.txt)

Koenkk, 2021-12-04:
@gregjwinter could you share me that pcapng file that wireshark creates on save? To keep the sniff as clean as possible:
- start sniffing
- permit joining
- factory reset device
- wait +- 2 mins
- save pcapng file

gregjwinter, 2021-12-05:
@Koenkk the file is here: https://www.dropbox.com/s/xvpsdkpaesc43ud/E-Wand_20121205.pcapng?dl=0
Let me know if I did this wrong or you need anything else.  Thank you. -Greg.

Koenkk, 2021-12-06:
@gregjwinter thanks for the sniff, I don't see the e-wand initiating any pairing, it should send a "Beacon" request (under the info column) after you factory reset it, can you manage to produce this?

gregjwinter, 2021-12-06:
@Koenkk  OK I have re-created the sniff with the pairing taking place.  So I started sniffing, permit joining, factory reset e-wand, wait 30 sec., initiated e-wand paring, waited 2 minutes, then saved the pcapng file.  I now see the "Beacon Request" under Info column.  If I missed any steps let me know and I'll try it again.
File is here: https://www.dropbox.com/s/8m500s48mtq7nzy/E-Wand_20121206.pcapng?dl=0

Koenkk, 2021-12-06:
@gregjwinter could you do the same and also provide the herdsman debug log of this pairing attempt? With this sniff it shouldn't fail with a `error 'Error: AREQ - ZDO - simpleDescRsp after 10000ms` before (because I see the simple descriptor response in the sniff)

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

gregjwinter, 2021-12-06:
@Koenkk Did you see this? https://www.dropbox.com/s/mrc1z4e7taedn84/E-Wand_Herdsman.txt?dl=0
This was some of the data I quickly copied from the text window within Home Assistance interface.

If this is not enough and need the log:  I  already have herdsman debug logging turned on.  I am running Home Assistant OS/Supervised addon and when I go into Terminal addon to HA I try to enter "docker ps" under the [core-ssh ~]$ it replies "bash: docker: command not found"  Sorry for my lack of Linux knowledge.  I also tried using putty to log into the Raspberry Pi that my HA is installed and it will not allow me to connect.  Thanks. -Greg.

Koenkk, 2021-12-07:
Did an attempt to fix it, can you check?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

scottpk, 2021-12-07:
Just a pointer since you mentioned you too are using HomeAssistant OS - scroll to the bottom of that page. Below is a screenshot of what it's talking about, you would uninstall the one called "Zigbee2mqtt" and install "Zigbee2mqtt Edge"

[image]

gregjwinter, 2021-12-07:
OK @Koenkk will do. Thank you. @scottpk Will I have to re-pair all my devices if I uninstall "Zigbee2mqtt"?  Is there a way to save this before I do switch over to the Edge version?

Koenkk, 2021-12-07:
@gregjwinter no repairing needed, just uninstall normal and install edge.

gregjwinter, 2021-12-07:
@Koenkk Progress!  I can pair the device using Edge version just fine.  Things appear normal, however I cannot control the tilt at all.  Previously I could pair the device (and have that ! indicator) but failing the interview.  But I could tilt/control the blind motor.  Now the ! indicator is gone and I don't have any control.

Here is my most current log here: https://www.dropbox.com/s/0sfgro509a65a6i/E-Wand_Herdsman_20211207.txt?dl=0
Thanks. -Greg.

gregjwinter, 2021-12-07:
@Koenkk Got back to this after work.  I realized I did not have Herdsman debug turned on.  For some odd reason my control from Home Assistant is working now!  But don't close this yet as I have more testing to do :-)  I did see this in the log:

Zigbee2MQTT:debug 2021-12-07 17:33:30: No converter available for 'CP180335E-01' with cluster '64528' and type 'raw' and data '{"data":[13,99,18,111,128,168,0,0,0,0,2,0,0,0,64,96,62,0,0,156,65,1,0,0,0,180,0],"type":"Buffer"}'

What does this mean?  It does send this when I closed the shades:
Zigbee2MQTT:debug 2021-12-07 17:32:25: Publishing 'set' 'tilt' to 'Hallway Piano Blind Center'
Zigbee2MQTT:info  2021-12-07 17:32:25: MQTT publish: topic 'zigbee2mqtt/Hallway Piano Blind Center', payload '{"battery":84,"linkquality":57,"tilt":100}'
Zigbee2MQTT:debug 2021-12-07 17:32:25: Received Zigbee message from 'Hallway Piano Blind Center', type 'raw', cluster '64528', data '{"data":[13,99,18,104,128,168,0,0,0,0,2,0,0,0,64,32,62,0,0,156,65,1,0,0,0,180,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-12-07 17:32:25: No converter available for 'CP180335E-01' with cluster '64528' and type 'raw' and data '{"data":[13,99,18,104,128,168,0,0,0,0,2,0,0,0,64,32,62,0,0,156,65,1,0,0,0,180,0],"type":"Buffer"}'

Koenkk, 2021-12-08:
Could you provide the data/database.db entry of this device for the following 2 cases:
- z2m 1.22.1 (NON edge), so pairing fails but controlling is fine
- z2m edge, pairing is OK but controlling fails.

gregjwinter, 2021-12-08:
Koen - I can control the device in Home Assistant now.  Just started to work later in the day.  However as of now I do not have any feedback of position (shows up as unknown) and does not sync up to Google Assistant like my IKEA shades do, also controlled by Zigbee2MQQT.  Here is a copy of my edge database.db file: https://www.dropbox.com/s/hlkns1s0h8qpzdd/DatabaseDb_20211208.txt?dl=0

Here is a link to a ZIP file containing some screen shots showing the unknown, comparison of data from IDEA "vs" Entity returned for my test E-Wand: https://www.dropbox.com/s/k2b3h7ynov011g8/E-Wand_20211208.zip?dl=0

Again, thank you for your help with this.  Really appreciate it. -Greg.

scottpk, 2021-12-08:
I'm surprised, I did not have so many problems...

The Google Assistant issue I may have an idea for, because they didn't show up automatically in Alexa either. I had to go into the Home Assistant Cloud settings, and click the button that says "Sync Entities to Amazon" - then after telling the Alexa app to search again they were there. So it might be the same thing for Google Assistant.

Koenkk, 2021-12-08:
@gregjwinter the position/state (open/close) should be reported once you set the position once via HA. If not, please provide the debug log when setting it.

gregjwinter, 2021-12-09:
I just tried it today, works but no position feed back in HA.  Here is my Herdsman log: https://www.dropbox.com/s/fn9qfbbmjnsd10q/E-Wand_Herdsman_20211209.txt?dl=0

Note the line at bottom: Zigbee2MQTT:debug 2021-12-09 12:42:56: No converter available for 'CP180335E-01' with cluster '64528' and type 'raw' and data '{"data":[13,99......  What does this mean?

@scottpk Yes I do push the button "Sync Entities to Google" after making changes in HA.  Does your E-Wand give you data like this picture?
[image]
I get unknown in this field even after operating E-Wand a few times.  Thanks. -Greg.

scottpk, 2021-12-09:
I'm on the stable version not the dev version, but mine looks like this:
[image]
I'd never looked at this before but they do show unknown. But it doesn't seem to matter, the HomeAssistant UI still seems to know they're open or closed (probably based on position).

gregjwinter, 2021-12-09:
Interesting @scottpk - thanks for sharing!  The only major difference is that you have a "last_seen" attribute that shows up on your E-Wands.  I wonder @Koenkk if this is what the "no converter message" (see my last post) that comes up in the Herdsman log I sent you?

Koenkk, 2021-12-09:
@gregjwinter 
- last seen is send only when the last seen feature is enabled: https://www.zigbee2mqtt.io/guide/configuration/mqtt.html#mqtt-behaviour
- no converter message: type `raw` indicates the message could not be parsed by zigbee-herdsman, that means it is either a non-valid message according to the ZCL standards OR that it is a manufacturer specific cluster (most likely). So we don't know what the device wants to say with it :)

gregjwinter, 2021-12-15:
The only way I can control the E-Wand is within the Home Assistant itself.  For some reason the entity comes into Google Assistant  and shows up as something to control, but cannot be controlled by saying OK Google, close the shades like I can do with my Ikea shades.  I suspect the reason is that Google expects the entity parameter to be "current_position" and not "current_tilt_position."  

@Koenkk is there a way to change this parameter to "current_position"?
Thanks. -Greg.

Koenkk, 2021-12-16:
@gregjwinter `current_position` and `current_tilt_position` are not things that come from z2m, this seems to be a HA thingy. I suggest to request help @ HA for this case.

Leah-Anderson, 2021-12-19:
@gregjwinter I haven't fully tested this but you could try manually configuring the device in homeassistant (configuration.yaml) to imitate the position attribute. You'll obviously have to adjust for whatever your device name is but here's what I was able to get working (asking google to open,close, open to a specific percent, etc.) :

```
  - platform: mqtt
    name: "Test Blinds"
    json_attributes_topic: "zigbee2mqtt/Living Room Back Blinds"
    position_topic: "zigbee2mqtt/Living Room Back Window Blinds"
    position_template: "{{ value_json.tilt }}"
    set_position_topic: "zigbee2mqtt/Living Room Back Window Blinds/set/tilt"
    command_topic: "zigbee2mqtt/Living Room Back Window Blinds/set/tilt"
    position_open: 100
    position_closed: 0
    payload_open: "100"
    payload_close: "0"
    payload_stop: "STOP"
```

If I had to guess Alexa can handle tilt devices where as Google can't from the looks of it. Originally when I added this I wanted to imitate the position at the converter level as that also fixed the state coloring in the HA dashboard. In any case if this works for you then you can prevent zigbee2mqtt from publishing discovery information (https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#home-assistant-device-registry) and just manually configure them. 

With regards to the manufacturer specific cluster someone did provide more information about that (https://github.com/Koenkk/zigbee2mqtt/issues/7648#issuecomment-902237381) but honestly I'm not sure exactly how to implement or address that. Hopefully that helps!

gregjwinter, 2021-12-22:
@Koenkk I purchased 2 more E-Wands directly from manufacturer.  They passed interview and show up in my Zigbee2Mqtt Edge controls.  When I try to control them I get this error message in the log:

Error Publish 'set' 'tilt' to 'Hallway Shade Right' failed: 'Error: Command 0x84fd27fffe80d5ce/196 closuresWindowCovering.goToTiltPercentage({"percentagetiltvalue":0}, {"sendWhenActive":true,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 1789 - 196 - 48 - 258 - 11 after 10000ms)'

I think we saw this message before you fixed things before.  Anything I can do?  I thought I was good but I was hoping to control 3 shades in my Hallway windows.  Thank you. -Greg.

I also get the same message with the other shade:
Error Publish 'set' 'tilt' to 'Hallway Shade Left' failed: 'Error: Command 0x847127fffe4cfa5a/196 closuresWindowCovering.goToTiltPercentage({"percentagetiltvalue":0}, {"sendWhenActive":true,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 59713 - 196 - 41 - 258 - 11 after 10000ms)'

Koenkk, 2021-12-23:
This means the device cannot be reached, do they work when moving them closer to the coordinator?

gregjwinter, 2021-12-23:
@Koenkk **Thanks for the response.**  I just tried to re-paired one of them close to the coordinator (within a foot).  Same thing, it shows up in m2q just fine but no response when I try to control it.  I did see this error in the log after I paired it:

 Failed to configure '0x84fd27fffe80d5ce', attempt 1 (Error: Bind 0x84fd27fffe80d5ce/1 genPowerCfg from '0x00124b00237e084c/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (internal/timers.js:557:17) at processTimers (internal/timers.js:500:7))

**Also here is the debug when I try to move the new E-Wand:**
Debug Received Zigbee message from 'Hallway Shade Right', type 'raw', cluster '64528', data '{"data":[13,99,18,27,128,200,0,0,0,0,2,0,0,0,249,86,66,0,0,166,65,1,255,172,0,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Debug No converter available for 'CP180335E-01' with cluster '64528' and type 'raw' and data '{"data":[13,99,18,27,128,200,0,0,0,0,2,0,0,0,249,86,66,0,0,166,65,1,255,172,0,0,0],"type":"Buffer"}'
Debug Received Zigbee message from 'Hallway Shade Left', type 'raw', cluster '64528', data '{"data":[13,99,18,120,128,200,0,0,0,0,2,0,0,192,227,146,67,0,0,162,65,1,0,172,0,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Debug No converter available for 'CP180335E-01' with cluster '64528' and type 'raw' and data '{"data":[13,99,18,120,128,200,0,0,0,0,2,0,0,192,227,146,67,0,0,162,65,1,0,172,0,0,0],"type":"Buffer"}'

Debug Received Zigbee message from 'Hallway Shade Center', type 'raw', cluster '64528', data '{"data":[13,99,18,19,128,154,0,0,0,0,2,0,0,176,202,250,67,0,0,161,65,1,0,135,0,80,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Debug No converter available for 'CP180335E-01' with cluster '64528' and type 'raw' and data '{"data":[13,99,18,19,128,154,0,0,0,0,2,0,0,176,202,250,67,0,0,161,65,1,0,135,0,80,0],"type":"Buffer"}'
Debug Received Zigbee message from 'Hallway Shade Right', type 'raw', cluster '64528', data '{"data":[13,99,18,28,128,200,0,0,0,0,2,0,0,192,125,85,66,0,0,163,65,1,255,171,0,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0

**Finally here is a screen shot of device seen by m2q:**
[image]

Koenkk, 2021-12-23:
Interesting, could you make a sniff of:
- Pairing the device to z2m
- after that trying to control it

gregjwinter, 2021-12-27:
@Koenkk Here is a sniff with a pair of new left shade and then I try to open and then close it:
https://www.dropbox.com/s/2gw5z8mah5pkha4/E-Wand_20121227_LeftShadePairControl.pcapng?dl=0

Here is a sniff with the working center shade with a open and then close command.  This one works:
https://www.dropbox.com/s/c1f3me9bvxrooya/E-Wand_20121227_CenterShade.pcapng?dl=0

Finally a sniff of just the non-working left shade open and close command.  Does not work.
https://www.dropbox.com/s/ioo5h466uryps0l/E-Wand_20121227_LeftShade.pcapng?dl=0

The command to open and close comes directly from Zigbee2MQQT Edge controls in Home Assistant and I am using the controls on the tab "Dashboard".

Thank you. -Greg.

Koenkk, 2021-12-29:
@gregjwinter thanks, I've checked your sniff and it seems that the coordinator doesn't send the command. What should happen is that the coordinator send the command to the parent of the e-wand, then the e-wand should do a data request (which it does) and send the command.

Could you make a sniff again and also provide the herdsman debug log when sending the command?  A sniff + log with just the failing command is enough (like `E-Wand_20121227_LeftShade.pcapng`)

gregjwinter, 2021-12-30:
@Koenkk So first thing I did was change the firmware on my Electrolama zig-a-zig-ah! (zzh!) moving it to CC2652R_coordinator_20211217.  Not sure if that is the version is the one you wanted me to flash.  Once I did this it took a restart of the service and unplugging the zzh! and plugging it back in to get stuff working again.

I went to start testing and the left and right shades work now!  I am not sure if the removal of the zzh to my laptop and then putting it back in made it restart the service or if it is the firmware upgrade.

The only thing that does not work is Google Assistant but I can control them in Home Assistant.  I just got a work around working using Node Red to trigger the open_cover_tilt command and it seems to work.
[image]
s

Koenkk, 2021-12-30:
Good to hear, assuming this issue can be closed now. If the issue re-appear feel free to reopen this issue with the sniff + logging as mentioned in https://github.com/Koenkk/zigbee2mqtt/issues/9604#issuecomment-1002676848

> The only thing that does not work is Google Assistant but I can control them in Home Assistant. I just got a work around working using Node Red to trigger the open_cover_tilt command and it seems to work.

This seems to be related to Home Assistant, I suggest creating a post on the HA forum

closed 2021-12-30 by Koenkk

gregjwinter, 2021-12-30:
Thank you for all the help with this. -Greg.

shanelord01, 2022-07-09:
Hi team. Sorry (not sorry) to reopen this - but is there any way to get the State field in HA to reflect the state of this product?

Seem Z2M can see the state (open stop close is shown) - but it's not translating that into the state field of HA.

Thanks,
Shane.