Koenkk/zigbee2mqtt #9627: Issue with ZZH on HASSOS with Zigbee2MQTT addon

fservida, 2021-11-12:
## What happened
Hello, since upgrading to a zzh stick I'm having timeout issues after ~24h of running.
Usually replugging the stick is sufficient to resume correct operation.
Especially strange is that sometime the zigbee command pass through if sent from HA (can toggle a light with a lot of delay, sometimes), but it loses all messages coming from the zigbee network (no sensors, nor state acknowledgments from the light).

I'm running a very basic network with one Ikea light and switch, and one aqara temperature sensor and a xiaomi switch.

Though it might be related to https://github.com/Koenkk/zigbee2mqtt/issues/8663, but I'm running Hassos 6.6 so it should be fixed if I understood correctly.

Debug logs from zigbee2mqtt:
- extract: https://pastebin.com/xLRN9iRs
- full: https://gist.github.com/fservida/04c5525e383200d0c1a64910ff6edec3

I do see this message in the kernel log around the time it stops working:
```
[229842.233725] ch341-uart ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[233683.223719] ch341-uart ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
```

I've tried running the script here to test the zzh once the network fails and it gives
```
Got 0 bytes in response to PING command: b''
FAIL|Expected length mismatch
```
https://gist.githubusercontent.com/omerk/0ee0e447a9e36786b4ff71d8f8126a23/raw/2dd88e739fe601d7c1e246b9bc117d513d47309e/znp-uart-test.py

## What did you expect to happen
Network should not experience timeout issues after some time running

## How to reproduce it (minimal and precise)
RPI 3B+
Hassos
- Home Assistant OS 6.6
- core-2021.11.1
- kernel 5.10.17-v8

## Debug info
Zigbee2MQTT version: 1.22.0-2
Adapter hardware: ZZH REV A5 - CC2652
Adapter firmware version: 
  - Coordinator type
    - zStack3x0
  - Coordinator revision
    - 20210708

robertalexa, 2021-11-13:
Have a look at this pinned issue
https://github.com/Koenkk/zigbee2mqtt/issues/8663

There is currently a kernel bug and 5.10.17-v8 is affected. `5.10.*: 5.10.58 till 5.10.61 are affected, fixed in 5.10.62`

fservida, 2021-12-14:
Anyone else experiencing this issue on HA OS?
I can provide additional logs if needed.

closed 2022-01-22 by github-actions[bot]

AdmiralStipe, 2022-04-11:
Same issue here, present since mid January 2022, although not on HASSOS, but on Supervised installation (RPI4B, currently with latest kernel 5.15.32-v7l+, everything updated to latest version).

fservida, 2022-04-15:
I've still not resolved this, would love feedback from the developers. Just reverted to a CC2531 as it was more stable :/

Are you by any chance using xiaomi or Ikea devices in your network?

AdmiralStipe, 2022-04-15:
Yes, I'm using mostly Xiaomi (temperature sensors, window sensors, cube, plug) and Ikea devices (controllers, buttons, presence sensors).
But I doubt the issue is related to devices used as it worked perfectly until last update (Z2MQTT version and coordinator firmware) with the same setup.
In the meantime I tried a small bunch of possible solutions regarding USB autosuspend, but none of them worked.

AdmiralStipe, 2022-04-20:
@fservida , can you re-open the issue again, as it's now closed? If it stays closed, most likely noone will deal with it. I would really like to boost it again, as it's very annoying.

fservida, 2022-04-22:
Hi, I’d love to but I can’t given it’s the bot that closed it…
@Koenkk or @omerk pinging for help on this issue