Koenkk/zigbee2mqtt #9635: Take filtered_attributes into account in exposes information

itavero, 2021-11-13:
## Is your feature request related to a problem? Please describe
Because of a feature request someone suggested for homebridge-z2m (see itavero/homebridge-z2m#295), I was looking through the code to see how the `filtered_attributes` config option is used and I noticed that it does not seem to influence the `exposes` information being published.

However, I think it makes sense to update the `access` attribute of `exposes` entries if the `property` is filtered via `filtered_attributes`.
Otherwise, software depending on this information, might incorrectly assume that information is being published, while it isn't.

## Describe the solution you'd like
If `filtered_attributes` is configured for a device, filter the `exposes` information so that the `access` level takes this filter into account. If this would result in "no access", the entry should be removed from the `exposes` information. 

## Describe alternatives you've considered
The depending software could request the Zigbee2MQTT configuration and figure it out by itself.
However, I think it makes more sense to fix the information source.

Koenkk, 2021-11-14:
This would be easy to do for binary, numeric sensors but gets a bit more complicated for lights. e.g. what should happen when you put `filtered_attributes: ["brightness"]`?

sjorge, 2021-11-14:
> This would be easy to do for binary, numeric sensors but gets a bit more complicated for lights. e.g. what should happen when you put `filtered_attributes: ["brightness"]`?

Bulb with brightness control -> bulb (or switch)
Bulb with brightness/color_temp -> bulb with color_temp but not brightness
Bulb with brightness/color -> bulb with color but not brightness

Not sure HA support the later, but based on https://nrchkb.github.io/wiki/service/lightbulb/ I'd expect that to be valid for HomeKit.

So I think it should be a valid expose data payload to just advertise a state + color(_temp) without brightness.

Worst case I guess for HA if it detects state, color, color_temp but not brightness and that config is not support it just downgrade all the way to just a bulb with on/off and nothing else.

Koenkk, 2021-11-15:
@sjorge thanks for your input, we can do something like this: 
- Generic: don't expose it when `property` is one of the `filtered_attributes`
- Specific: remove feature from `features` when `property` is one of the `filtered_attributes`, if this results in no features being left, don't expose it at all.

(docs on generic/specific expose types: https://www.zigbee2mqtt.io/guide/usage/exposes.html)

itavero, 2021-12-18:
I'll see if I have some time in the upcoming weeks to figure out where this can be implemented.

I also have a follow up feature request, also related to itavero/homebridge-z2m#295, to think about a way to set the `filtered_attributes` (and perhaps other configuration) based on the device model.
The reasoning behind this, is that if you have multiple of the same devices you probably want them to behave the same.
One of the things I would filter out would be the internal temperature sensors of some devices (as they deviate too much to be useful).

sjorge, 2021-12-18:
There is already a way to set device options like this for all devices.

I guess it would make sense to have a similar one that can set it for all devices of a certain modelid! The main question is how to go about it… in theory it’s easy to find this, but there are some devices (TuYa) that share model ideas and use fingerprinting to match then, so simply filtering on a key based on modelid won’t work… I like the idea though! But simple filter probably won’t be enough 🤔

itavero, 2021-12-21:
I reckon you mean that the Zigbee model ID and manufacturer are equal in those cases?
It's been awhile since I worked on the Zigbee2MQTT code base, but we do eventually also determine what kind of device it actually is, right?

sjorge, 2021-12-21:
Yes, I hope the ‘model’ we set is unique but I am not 100% sure on that.

Unfortunately the modelId coming from the device itself is not :(, a lot of tuya devices for example have the same one and we need fancy fingerprint tricks to identify them see https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js

~ sjorge

closed 2022-01-30 by github-actions[bot]