Koenkk/zigbee2mqtt #9638: Gledopto GL-S-007Z state not changing on first click

RadianceNL, 2021-11-13:
## What happened
I have serval Gledopto GL-S-007Z  spots and when I want to turn them on their state doesn't change.
On first click the state in HA is still off. After the second click the state changes to on but the brightness percentage stays always on 0%. The same happens when turning off the spots.

The spot _does_ turn on and off on the first clicks to change the state, it's just the state doesn't change in the HA interface.
In the Z2M dashboard from the Z2M HA addon the state does change as expected. When I turn the spot on through the Z2M dashboard the same behavior takes place as in the Lovelace UI.

It did work as expected in the past but I can't remember which Z2M update broke this device.
Other devices like Ikea Tradfri spots and bulbs work as expected, it's just specific to the GLED spots.
 
## What did you expect to happen
Previously the spots where working like any other light entities, on click state changes to on with the correct brightness percentage.

## How to reproduce it (minimal and precise)
Click on the entity in dashboard, state doesn't change
Second click on the entity, state changes to on but brightness always stay on 0%
Click again to turn off, state doesn't change on first click
Second click to turn off, state changes to off.

## Debug info
Zigbee2MQTT version: 1.22.0 commit: unknown (HA addon)
Adapter hardware: Texas Instruments LAUNCHXL-CC26X2R1
Adapter firmware version: CC2652R_coordinator_20210708.hex

DJTerentjev, 2021-11-13:
@RadianceNL sorry to intervene. what is firmware version of your spots? I have the same spots and turn on/off working on the first click. I feel like Gledopto keep changing firmware  and spots can react differently. I got 3.0.1. Previously I had GL-S-007Z with older firmware version and couldn't make them work at all.

RadianceNL, 2021-11-13:
> @RadianceNL sorry to intervene. what is firmware version of your spots? I have the same spots and turn on/off working on the first click. I feel like Gledopto keep changing firmware and spots can [...]

I'm on firmware version 2.0.0

artist67, 2021-11-16:
I have the same, in version 2 as well.
Currently, the only way to turn them on is sending two separate commands to them, first brightness, then "on".
Turning off is simply possible by sending "off".

RoboMagus, 2021-12-17:
I've also got issues with these Gledopto bulbs. 
They worked quite fine when they were in my Philips hue network, however since migrating them to Z2M last week I noticed issues with them flickering a few times when turning the bulbs on using a HA automation. 
Similarly, their state every now and then becomes different from the one in HA.
Also when using the remote that was included with the set of 2 of these bulbs, the status of the bulbs is not propagated.

If I had to guess, this al seems to be like the bulbs not publishing their state or something...

closed 2022-01-25 by github-actions[bot]