Koenkk/zigbee2mqtt #9644: z2m still pings device while herdsman has entered stopping phase

sjorge, 2021-11-13:
## What happened
While working on the STARKVIND I noticed that z2m still pings devices while it has entered the stopping state:
```
Zigbee2MQTT:info  2021-11-13 19:36:12: Stopping zigbee-herdsman...
Zigbee2MQTT:debug 2021-11-13 19:36:22: Succesfully pinged 'switch/entrance/light' (attempt 1/1)
Zigbee2MQTT:debug 2021-11-13 19:36:25: Succesfully pinged 'light/livingroom/main/bulb1' (attempt 1/1)
Zigbee2MQTT:debug 2021-11-13 19:36:28: Succesfully pinged 'light/livingroom/main/bulb2' (attempt 1/1)
Zigbee2MQTT:debug 2021-11-13 19:36:31: Succesfully pinged 'light/livingroom/main/bulb3' (attempt 1/1)
Zigbee2MQTT:debug 2021-11-13 19:36:34: Succesfully pinged 'light/livingroom/main/bulb4' (attempt 1/1)
Zigbee2MQTT:debug 2021-11-13 19:36:37: Succesfully pinged 'light/balcony/main/bulb' (attempt 1/1)
Zigbee2MQTT:debug 2021-11-13 19:36:40: Succesfully pinged 'light/hallway/main/bulb' (attempt 1/1)
```

## What did you expect to happen
That it would stop pinging while herdsman is trying to stop.

## How to reproduce it (minimal and precise)
Stop z2m while it is pinging a bunch of devices.

## Debug info
Zigbee2MQTT version:  1.22.0-dev commit: 3bfc7f31
Adapter hardware: CC1352P-2
Adapter firmware version: 20210708

Koenkk, 2021-11-14:
I expect that https://github.com/Koenkk/zigbee2mqtt/commit/3141a14fec1f9aca738c0e788d0315453e1e6fe5 fixes it, in the reverse order there is a potential race condition when an event comes in between these 2 lines, so timers are stopped, events comes in (since `super.stop()` has not been called yet) which creates the timers again.

sjorge, 2021-11-14:
Yes, that would make sense. I'll keep an eye out if i see it again or not. (But I'm usually not watching the log unless actively working on a device to support)

closed 2021-11-14 by sjorge