Koenkk/zigbee2mqtt #9645: MS-105B Dimmer - "No converter available" when using comands other then zigbee2mqtt/FRIENDLY_NAME/set "brightness": XXX and "state": ON/OFF

Pitdelux, 2021-11-13:
Hi everyone,

I'm having issues using my Moes MS-105B Smart dimmer module (2 gang).

When I publish the following commands in MQTT, everything works as expected:
- zigbee2mqtt/FRIENDLY_NAME/set with payload {"state_l1": "ON"}, {"state_l1": "OFF"}
- zigbee2mqtt/FRIENDLY_NAME/set with payload {"brightness_l1": VALUE} where VALUE is a number between 0 and 254

The following command doesn't work:
- zigbee2mqtt/FRIENDLY_NAME/set with payload {"state_l1": "TOGGLE"}

Command zigbee2mqtt/FRIENDLY_NAME/set with payload {"brightness_l1":156,"transition":3} brings the brightness to 156 but without any transition.  I get a message mentionning that "No converter available for 'transition' " 

And when using the following commands, nothing happens and I get a message mentionning that "No converter available" for the called command in Z2M
- zigbee2mqtt/FRIENDLY_NAME/set {"brightness_move_l1": -40}
- zigbee2mqtt/FRIENDLY_NAME/set {"brightness_step_l1": 40} 

It's the same for both gangs (_l1 and _l2).  What works (or doesn't) for one works for the other.

Can anyone help with this?

[image]

Many thanks in advance.
David

knallhare, 2021-12-05:
Hi David,

I'm struggeling with the same issues. Have you had any luck on this yet?

sniicker, 2021-12-07:
Hi Guys,

i have the same problem. Very frustrating. Did not found a way till now.

knallhare, 2021-12-07:
Yeah, it's very frustrating. I have both MS-105Z and MS-105B. None of them work.

closed 2022-01-16 by github-actions[bot]

somik321, 2022-05-23:
the problem is not solved