Koenkk/zigbee2mqtt #9646: Incorrect state message for Schneider Electric 545D6102

kenni, 2021-11-13:
Using a Schneider Electric 545D6102 (LK FUGA wiser wireless dimmer).

## What happened
Clicking on the physical on or off buttons on the device doesn't update state_l1 to the correct on/off state. Instead the change can be seen in the variable "state", which isn't part of the l1 endpoint and hence isn't used or passed on.

## What did you expect to happen
Clicking on the physical on or off buttons on the device should update state_l1 to the correct on/off state.

## How to reproduce it (minimal and precise)
Toggle the on/off physical buttons on the device and check the State page in Zigbee2MQTT and see how "state_l1" isn't updated, while "state" is:
    "state": "ON",
    "state_l1": "OFF"
->
    "state": "OFF",
    "state_l1": "OFF"
->
    "state": "ON",
    "state_l1": "OFF"
-> etc.

## Debug info
Zigbee2MQTT version: 1.22.0-2
Adapter hardware: zzh! (Rev A5)
Adapter firmware version: Z-Stack 3.x.0 20210708

Debug logs of one click on the physical power-on-button is attached.
[One_click_on_physical_on_button.txt](https://github.com/Koenkk/zigbee2mqtt/files/7532886/One_click_on_physical_on_button.txt)

gbjmkm, 2021-11-24:
I see the same here. This also goes for "CCT5011-0001/CCT5011-0002/MEG5011-0001" and "545D6514"

tomato20, 2021-11-24:
Also experiencing the same problem...

Koenkk, 2021-11-24:
Should be fixed now, after updating I recommend to stop z2m, clear `data/state.json` and start z2m (otherwise the `state` property will stay in the state).

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

kenni, 2021-12-28:
> Should be fixed now, after updating I recommend to stop z2m, clear `data/state.json` and start z2m (otherwise the `state` property will stay in the state).

Thanks a lot for the quick fix - I can confirm that it seems to work as expected :) Any chance that you can apply a similar fix to 545D6514? I'm experiencing exactly the same issue here, this device just has two independent relays instead of one. I'll stay on the dev branch for now, so I'm able to test quickly, in case you push something. Thanks again.

Koenkk, 2021-12-29:
@kenni 545D6514 should already work, can you provide the debug log from starting z2m till the wrong state update is send?

kenni, 2021-12-29:
@Koenkk Sure, I've uploaded the debug log for 545D6514.

I start z2m and toggle between pressing the on and the off buttons for relay 1 and relay 2. Four times per relay, eight presses in total. None of the state changes are registered, when looking at the State in the z2m web interface, the states stay at:

{
    "linkquality": 36,
    "state_l1": "ON",
    "state_l2": "ON",
    "action": ""
}

[545D6514_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7790235/545D6514_log.txt)

Thanks

kenni, 2021-12-29:
Ohh, and I'm doing the testing on 0xb4e3f9fffed5a047 if that isn't obvious.

Koenkk, 2021-12-29:
I don't see any messages being send to `zigbee2mqtt/Køkken Afbryder/l1/set` or  `zigbee2mqtt/Køkken Afbryder/l2/set`, make sure to control the switch while controlling it. If you did control it, does it work when controlling via the z2m UI? If yes then this looks to be a HA issue.

kenni, 2021-12-29:
Please bare with me, I'm not sure I fully understand the concept, but I do not have any issues controlling the relays by using the z2m UI. Pressing "Off" turns off the relay and pressing "On" turns on the relay. It's when I press the physical contacts that the state doesn't get updated in z2m or HA.

Koenkk, 2021-12-29:
Ah sorry, understood it wrong, could you:
- update to z2m 1.22.1
- go to the frontend and press the configure button, provide the logging when pressing this
- provide the data/database.db entry of this device

kenni, 2021-12-29:
I'm on z2m 1.22.1. Do you mean the "Reconfigure"-button for the device? I've pressed it, but it doesn't provide a lot of information in the log.

Just let me know if other information is needed.

[database.db.txt](https://github.com/Koenkk/zigbee2mqtt/files/7790817/database.db.txt)
[reconfigure.txt](https://github.com/Koenkk/zigbee2mqtt/files/7790818/reconfigure.txt)

Koenkk, 2021-12-30:
Implemented a possible fix.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

kenni, 2022-01-01:
@Koenkk It works! Thanks a lot for all of your help :)

FYI, another user, @gbjmkm , wrote in a comment above that the same issue exists for three other models in the same series from Schneider Electric: "CCT5011-0001/CCT5011-0002/MEG5011-0001". I don't have any of these, so I can't confirm it nor test it, but perhaps it's the same one-line change needed?

Koenkk, 2022-01-01:
@kenni thanks, assuming this can be closed now.

@gbjmkm make sure to check with the new 1.22.2 release, it should at least fix the issue for `545D6514`

closed 2022-01-01 by Koenkk