Koenkk/zigbee2mqtt #9648: Ikea's STARKVIND Aire Purifier (E2007) not working properly

kikettas, 2021-11-14:
## What happened
After adding and configuring the STARKVIND Ikea's air purifier to Zigbee2mqtt, the exposed action do no work as expected. 
It is not possible to change the mode, or go from OFF to ON state (it works the other way around, you can turn it off),

## What did you expect to happen
That both the ON/OFF and mode switches work.

## How to reproduce it (minimal and precise)
Try to turn it ON while is OFF. Try to change the mode with is ON.

## Debug info
Zigbee2MQTT version: 1.22.0
Adapter hardware: Electrolama CC2652R.
Adapter firmware version: 20210708.

sjorge, 2021-11-14:
Yes, there are currently a few new issues. Very basic support is broken in the latest z2m release but fixed in dev.

Better support is pending https://github.com/Koenkk/zigbee-herdsman-converters/pull/3345

sjorge, 2021-11-14:
PR got merged, so should all be good when you run on the dev branch

kikettas, 2021-11-15:
Thabk you @sjorge 👌 I’ll wait for the stable version release.

smartmatic, 2021-11-25:
Is it know when the next stable version will be released?

Koenkk, 2021-11-25:
Next stable release will be on 1 december, assuming this can be closed now.

closed 2021-11-25 by Koenkk

createcandle, 2022-01-14:
I just received this device, but I am seeing issues when trying to actuate it. This is using the latest stable Z2M (1.22.2).

- Trying to change state:  `No converter available for 'state' ("ON")`
- trying to change mode: `No converter available for 'mode' ("1")`

But getting the state also seems to fail:
debug 2022-01-14 21:23:45: Received MQTT message on 'zigbee2mqtt/0xcc86ecfffXXXXX/get' with data '{"state":""}'
- using '/get' on the state: `No converter available for 'state' ("")`

I _am_ able to change the `led_state`, so at least some `to_zigbee` functionality it working.

This is all based on Exposes functionality.

I noticed that the [Starkvind documentation](https://www.zigbee2mqtt.io/devices/E2007.html) mentions `fan_mode` as the property to use, while the exposes data uses `mode`.

Similarly, it mentions `fan_state`, while the exposes data actually uses `state`.

Other sensors are working fine:
[image]

Debug logs show this issue:

```
Configuring '0xcc86ecfffe54396d'
Received Zigbee message from '0xcc86ecfffeXXXXXX', type 'raw', cluster 'manuSpecificIkeaAirPurifier', data '{"data":[24,124,18,1],"type":"Buffer"}' from endpoint 1 with groupID null
No converter available for 'E2007' with cluster 'manuSpecificIkeaAirPurifier' and type 'raw' and data '{"data":[24,124,18,1],"type":"Buffer"}'
Received Zigbee message from '0xcc86ecfffXXXXXX', type 'raw', cluster 'manuSpecificIkeaAirPurifier', data '{"data":[24,124,18,1],"type":"Buffer"}' from endpoint 1 with groupID null
No converter available for 'E2007' with cluster 'manuSpecificIkeaAirPurifier' and type 'raw' and data '{"data":[24,124,18,1],"type":"Buffer"}'
Failed to configure '0xcc86ecfffXXXXXX', attempt 3 (Error: ConfigureReporting 0xcc86ecfffXXXXXX/1 manuSpecificIkeaAirPurifier([{"attribute":"filterRunTime","minimumReportInterval":3600,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4476,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)
 at DeconzAdapter.sendZclFrameToEndpoint (/home/pi/.webthings/data/zigbee2mqtt-adapter/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:641:23))
```

Database entry:
`{"id":30,"type":"Router","ieeeAddr":"0xcc86ecfffXXXXXX","nwkAddr":3805,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"STARKVIND Air purifier","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":7,"inClusterList":[0,3,4,5,514,64599,64637],"outClusterList":[25,1024,1066],"clusters":{"genBasic":{"attributes":{"modelId":"STARKVIND Air purifier","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":3,"appVersion":16,"stackVersion":103,"hwVersion":1,"dateCode":"20210428","swBuildId":"1.0.033"}},"manuSpecificIkeaAirPurifier":{"attributes":{"particulateMatter25Measurement":15,"filterRunTime":12}}},"binds":[{"cluster":64637,"type":"endpoint","deviceIeeeAddress":"0x00212effff0539e5","endpointID":1}],"configuredReportings":[{"cluster":64637,"attrId":4,"minRepIntval":60,"maxRepIntval":3600,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":16,"stackVersion":103,"hwVersion":1,"dateCode":"20210428","swBuildId":"1.0.033","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1642192753273,"useImplicitCheckin":true}

```
Starting Zigbee2MQTT version 1.22.2 (commit #1a0a9a6)
Starting zigbee-herdsman (0.13.188)
```
And using the Conbee2 stick with latest firmware

sjorge, 2022-01-14:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/ed0690c74a50b262909eb4d26b56c6c270431aa2/lib/exposes.js#L346-L351

The names are`mode` and `state` but the properties are `fan_mode` and `fan_state` respectively. I just checked and the frontend is sending the properties correctly. And the device is responding correctly.

I assume this `webthings` platform you are using is also taking the name instead of the property value of the expose data, which results in an incorrect payload being send. The frontend also had this bug but it got fixed a while ago here: 
https://github.com/nurikk/zigbee2mqtt-frontend/commit/fff9f5d70414b9f6ede5d4b16f82d8ea6c4a6aa3

I'm pretty sure this is a webthings bug but I can't find much info about it.

createcandle, 2022-01-14:
Ah, I see, thanks. Do you recommend that I use `property` instead of `name` everywhere?

sjorge, 2022-01-14:
Well for creating the payload, yes `property` should be used everywhere. For a lot of expose data property==name but not always.

`property` => use in payload
`name` => use for displaying somewhere, say a webui, application ui, ...

createcandle, 2022-01-14:
It's working now. Thanks for your really fast response.

The only odd thing is that I never receive `fan_speed` data, but that's probably another bug at my end.

sjorge, 2022-01-14:
There should hè reporting for fan_speed configured, if it still doesn’t show up maybe try reconfiguring it, it should add the reporting again

createcandle, 2022-01-16:
It showed up OK on my second test Pi. But my main system still refuses to show the fan speed. I suspect I have to delete a file somewhere? I already tried reinstalling Z2M, but to my surprise that didn't work.

sjorge, 2022-01-16:
> It showed up OK on my second test Pi. But my main system still refuses to show the fan speed. I suspect I have to delete a file somewhere? I already tried reinstalling Z2M, but to my surprise that didn't work.

Double check if the fanSpeed reporting is configure,
[image]

createcandle, 2022-01-24:
@sjorge I don't have that interface :-) I connect to Z2M via MQTT only. Is there a way to check without that UI? What files does it operate on?

sjorge, 2022-01-24:
> @sjorge I don't have that interface :-) I connect to Z2M via MQTT only. Is there a way to check without that UI? What files does it operate on?

It’s probably easier to just trigger a reconfigure

createcandle, 2022-01-24:
How would I do that? I tried un-pairing and re-pairing, but that didn't seem to work.

sjorge, 2022-01-24:
> How would I do that? I tried un-pairing and re-pairing, but that didn't seem to work.

That should also have done a (re)configure step, did you ever add an external_convertor? Maybe one of those is causing issues.

createcandle, 2022-01-25:
Yes I have had an external converter. For a different device though.

Could I delete a line from the database file to fix this?

sjorge, 2022-01-25:
> Yes I have had an external converter. For a different device though.

Probably not going to be an issue then

> Could I delete a line from the database file to fix this?

Unlikely, after removing the device there should be no entries left.