Koenkk/zigbee2mqtt #9653: Zigbee2MQTT does not detect new device

jos149, 2021-11-14:
## Information about the device + link
I have a motion detection sensor which is not a supported device yet. I obtained it via my previous work. It is a zigbee enabled sensor which used to connect without any problems with the gateway used for our customers. The problem is: I cannot find any brand/type of the device.

## data/database.db entry of the device
What I want to do is to add the device properly to the existing database. For that reason, I tried to add the device in Zigbee2MQTT. However, without any luck! The sensor is blinking when it is in pair mode, and the Zigbee2MQTT is on permitjoin: true. In the mean time, while trying to add the device, I connected a Zigbee antenna to my pc running Wireshark. As I am not very experienced I am not sure what to do with it yet, however, when having the sensor in pair-mode Wireshark floods with 'channel request' messages. Hence, some activity is occuring! Now the question for you is: how can we make sure the device will be able to connect?

I attached the summary of the broadcast message down below. Not sure what to do with it however. To me, it appears that the sensor might be working on a different Zigbee channel than the sensor in the Zigbee2MQTT instance?
----
## Packet 1090
- 1091
- 2871.601023
- 
- Broadcast
- ZigBee Green Power
- 12
- Channel Request

Burki24, 2021-11-15:
If you don't have any details regarding this device, it will be helpfull to send a photo. So we can have a lokk on it.

jos149, 2021-11-18:
> If you don't have any details regarding this device, it will be helpfull to send a photo. So we can have a lokk on it.

Here they are. Sorry for my late reply.[image][image]

jos149, 2021-11-24:
Based on the photos; does anyone have a clue?

closed 2022-01-01 by github-actions[bot]