Koenkk/zigbee2mqtt #9658: Missed publishes with optimistic publishing

sjorge, 2021-11-14:
## What happened
As discovered while working on https://github.com/Koenkk/zigbee-herdsman-converters/pull/3345, it seems we miss some data when optimistically setting a state in the tz converter and then receiving a attReport that also sets the value (possibly to something else)

### Examples
Setting fan_mode to auto, the [tz.ikea_air_purifier_fan_mode](https://github.com/Koenkk/zigbee-herdsman-converters/blob/5ecd483a0c8cd7e7076adb8688126d5eec2b603d/converters/toZigbee.js#L1915-L1940) returns the state + a report is received once the device has executed the change, it contains additional data like fan_speed.

```
Zigbee2MQTT:debug 2021-11-13 21:21:21: Received MQTT message on 'zigbee2mqtt/airpurifier/bedroom/set' with data '{"fan_mode":"auto"}'
Zigbee2MQTT:debug 2021-11-13 21:21:21: Publishing 'set' 'fan_mode' to 'airpurifier/bedroom'
Zigbee2MQTT:debug 2021-11-13 21:21:21: Received Zigbee message from 'airpurifier/bedroom', type 'attributeReport', cluster 'manuSpecificIkeaAirPurifier', data '{"fanMode":1,"fanSpeed":30}' from endpoint 1 with groupID 0
XXX: detected fanSpeed='30'
XXX: detected fanMode='1'
YYY: result state={"fan_speed":5,"fan_mode":"auto","fan_state":"ON"}, data={"fanSpeed":30,"fanMode":1}
Zigbee2MQTT:info  2021-11-13 21:21:21: MQTT publish: topic 'zigbee2mqtt/airpurifier/bedroom', payload '{"air_quality":"good","child_lock":"UNLOCK","fan_mode":"auto","fan_speed":1,"fan_state":"ON","led_enable":true,"linkquality":78,"pm25":0,"replace_filter":false}'
Zigbee2MQTT:info  2021-11-13 21:21:22: MQTT publish: topic 'zigbee2mqtt/airpurifier/bedroom', payload '{"air_quality":"good","child_lock":"UNLOCK","fan_mode":"auto","fan_speed":1,"fan_state":"ON","led_enable":true,"linkquality":78,"pm25":0,"replace_filter":false}'
```

This was with some extra debugging in [fz.ikea_air_purifier](https://github.com/Koenkk/zigbee-herdsman-converters/blob/5ecd483a0c8cd7e7076adb8688126d5eec2b603d/converters/fromZigbee.js#L4298-L4370) that printed the msg.data and state result and also triggered inside the fanMode and fanSpeed if blocks to make sure they were being correctly reached.

2nd example of setting fan_mode to off
```
Zigbee2MQTT:debug 2021-11-13 21:22:11: Received MQTT message on 'zigbee2mqtt/airpurifier/bedroom/set' with data '{"fan_mode":"off"}'
Zigbee2MQTT:debug 2021-11-13 21:22:11: Publishing 'set' 'fan_mode' to 'airpurifier/bedroom'
Zigbee2MQTT:debug 2021-11-13 21:22:11: Received Zigbee message from 'airpurifier/bedroom', type 'attributeReport', cluster 'manuSpecificIkeaAirPurifier', data '{"fanMode":0,"fanSpeed":0}' from endpoint 1 with groupID 0
XXX: detected fanSpeed='0'
XXX: detected fanMode='0'
YYY: result state={"fan_speed":0,"fan_mode":"off","fan_state":"OFF"}, data={"fanSpeed":0,"fanMode":0}
Zigbee2MQTT:info  2021-11-13 21:22:11: MQTT publish: topic 'zigbee2mqtt/airpurifier/bedroom', payload '{"air_quality":"good","child_lock":"UNLOCK","fan_mode":"off","fan_speed":4,"fan_state":"OFF","led_enable":true,"linkquality":93,"pm25":0,"replace_filter":false}'
Zigbee2MQTT:info  2021-11-13 21:22:12: MQTT publish: topic 'zigbee2mqtt/airpurifier/bedroom', payload '{"air_quality":"good","child_lock":"UNLOCK","fan_mode":"off","fan_speed":4,"fan_state":"OFF","led_enable":true,"linkquality":93,"pm25":0,"replace_filter":false}'
```

3rd example setting fan_mode to a fixed speed
```
Zigbee2MQTT:debug 2021-11-13 21:22:52: Received MQTT message on 'zigbee2mqtt/airpurifier/bedroom/set' with data '{"fan_mode":"9"}'
Zigbee2MQTT:debug 2021-11-13 21:22:52: Publishing 'set' 'fan_mode' to 'airpurifier/bedroom'
Zigbee2MQTT:debug 2021-11-13 21:22:52: Received Zigbee message from 'airpurifier/bedroom', type 'attributeReport', cluster 'manuSpecificIkeaAirPurifier', data '{"fanMode":50,"fanSpeed":50}' from endpoint 1 with groupID 0
XXX: detected fanSpeed='50'
XXX: detected fanMode='50'
YYY: result state={"fan_speed":9,"fan_mode":"9","fan_state":"ON"}, data={"fanSpeed":50,"fanMode":50}
Zigbee2MQTT:info  2021-11-13 21:22:52: MQTT publish: topic 'zigbee2mqtt/airpurifier/bedroom', payload '{"air_quality":"unknown","child_lock":"UNLOCK","fan_mode":"9","fan_speed":4,"fan_state":"ON","led_enable":true,"linkquality":63,"pm25":-1,"replace_filter":false}'
Zigbee2MQTT:info  2021-11-13 21:22:53: MQTT publish: topic 'zigbee2mqtt/airpurifier/bedroom', payload '{"air_quality":"unknown","child_lock":"UNLOCK","fan_mode":"9","fan_speed":4,"fan_state":"ON","led_enable":true,"linkquality":63,"pm25":-1,"replace_filter":false}'
```

There is more, but I think not really relevant info in the PR, when the change comes in via attReport only (aka physically changing it on the device) or with `filtered_optimistic` including `fan_mode` and `fan_state` the issue is not present. Hence this issue as it only happens when the optimistic state publish is in the mix.

Alsop notice that here are 2 publish in the example further indicating something doesn't seem to be correct.

## What did you expect to happen
Both states to be correctly merged and debounced or at the very lease the 2nd publish to have the correct information returned from the fz function.

## How to reproduce it (minimal and precise)
Have optimistic enabled and setting fan_mode on a STARKVIND Air Purifier (probably on other devices too, but just noticed while working on this one)

## Debug info
Zigbee2MQTT version: 1.22.0-dev commit: d192bbd2
Adapter hardware: CC1352P-2
Adapter firmware version: 20210708

Koenkk, 2021-11-15:
Can you provide the logging adding the changes from https://github.com/Koenkk/zigbee2mqtt/pull/9682/files (reminder, when editing `.ts` files don't forgot to run `npm run build` afterwards)

sjorge, 2021-11-15:
```Zigbee2MQTT:debug 2021-11-15 18:21:35: Received MQTT message on 'zigbee2mqtt/airpurifier/bedroom/set' with data '{"fan_state":"OFF"}'
Zigbee2MQTT:debug 2021-11-15 18:21:35: Publishing 'set' 'fan_state' to 'airpurifier/bedroom'
Zigbee2MQTT:debug 2021-11-15 18:21:36: Received Zigbee message from 'airpurifier/bedroom', type 'attributeReport', cluster 'manuSpecificIkeaAirPurifier', data '{"fanMode":0,"fanSpeed":0,"particulateMatter25Measurement":65535}' from endpoint 1 with groupID 0
PUBLISH PUBLISH { fan_mode: 'off', fan_state: 'OFF' }
payload { fan_mode: 'off', fan_state: 'OFF' }
newstate {
  fan_state: 'OFF',
  fan_speed: 1,
  pm25: 0,
  air_quality: 'good',
  led_enable: true,
  child_lock: 'UNLOCK',
  replace_filter: false,
  fan_mode: 'off'
}
Zigbee2MQTT:info  2021-11-15 18:21:36: MQTT publish: topic 'zigbee2mqtt/airpurifier/bedroom', payload '{"air_quality":"good","child_lock":"UNLOCK","fan_mode":"off","fan_speed":1,"fan_state":"OFF","led_enable":true,"linkquality":96,"pm25":0,"replace_filter":false}'
publish {
  fan_state: 'OFF',
  fan_speed: 1,
  pm25: 0,
  air_quality: 'good',
  led_enable: true,
  child_lock: 'UNLOCK',
  replace_filter: false,
  fan_mode: 'off',
  linkquality: 96
}
payload {}
newstate {
  fan_state: 'OFF',
  fan_speed: 1,
  pm25: 0,
  air_quality: 'good',
  led_enable: true,
  child_lock: 'UNLOCK',
  replace_filter: false,
  fan_mode: 'off'
}
Zigbee2MQTT:info  2021-11-15 18:21:36: MQTT publish: topic 'zigbee2mqtt/airpurifier/bedroom', payload '{"air_quality":"good","child_lock":"UNLOCK","fan_mode":"off","fan_speed":1,"fan_state":"OFF","led_enable":true,"linkquality":96,"pm25":0,"replace_filter":false}'
publish {
  fan_state: 'OFF',
  fan_speed: 1,
  pm25: 0,
  air_quality: 'good',
  led_enable: true,
  child_lock: 'UNLOCK',
  replace_filter: false,
  fan_mode: 'off',
  linkquality: 96
}```

sjorge, 2021-11-15:
Thats so weird as it already seems lost before it gets there?

sjorge, 2021-11-15:
I added some of the debug code back into fromZigbee:
```patch
root@amethyst:/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters# diff -u converters/fromZigbee.js.bak converters/fromZigbee.js
--- converters/fromZigbee.js.bak        Mon Nov 15 18:24:03 2021
+++ converters/fromZigbee.js    Mon Nov 15 18:24:54 2021
@@ -4365,6 +4365,7 @@
                 state['fan_state'] = (fanMode === 'off' ? 'OFF' : 'ON');
             }

+            console.log(`fromZigbee: state='${JSON.stringify(state)}' data='${JSON.stringify(msg.data)}'`);
             return state;
         },
     },
```

```
Zigbee2MQTT:debug 2021-11-15 18:28:17: Received MQTT message on 'zigbee2mqtt/airpurifier/bedroom/set' with data '{"fan_mode":"8"}'
Zigbee2MQTT:debug 2021-11-15 18:28:17: Publishing 'set' 'fan_mode' to 'airpurifier/bedroom'
Zigbee2MQTT:debug 2021-11-15 18:28:20: Received Zigbee message from 'airpurifier/bedroom', type 'attributeReport', cluster 'manuSpecificIkeaAirPurifier', data '{"fanMode":45,"fanSpeed":45}' from endpoint 1 with groupID 0
fromZigbee: state='{"fan_speed":8,"fan_mode":"8","fan_state":"ON"}' data='{"fanSpeed":45,"fanMode":45}'
PUBLISH PUBLISH { fan_mode: '8', fan_state: 'ON' }
payload { fan_mode: '8', fan_state: 'ON' }
newstate {
  fan_state: 'ON',
  fan_speed: 1,
  pm25: 1,
  air_quality: 'good',
  led_enable: true,
  child_lock: 'UNLOCK',
  replace_filter: false,
  fan_mode: '8'
}
Zigbee2MQTT:info  2021-11-15 18:28:20: MQTT publish: topic 'zigbee2mqtt/airpurifier/bedroom', payload '{"air_quality":"good","child_lock":"UNLOCK","fan_mode":"8","fan_speed":1,"fan_state":"ON","led_enable":true,"linkquality":90,"pm25":1,"replace_filter":false}'
publish {
  fan_state: 'ON',
  fan_speed: 1,
  pm25: 1,
  air_quality: 'good',
  led_enable: true,
  child_lock: 'UNLOCK',
  replace_filter: false,
  fan_mode: '8',
  linkquality: 90
}
payload {}
newstate {
  fan_state: 'ON',
  fan_speed: 1,
  pm25: 1,
  air_quality: 'good',
  led_enable: true,
  child_lock: 'UNLOCK',
  replace_filter: false,
  fan_mode: '8'
}
Zigbee2MQTT:info  2021-11-15 18:28:20: MQTT publish: topic 'zigbee2mqtt/airpurifier/bedroom', payload '{"air_quality":"good","child_lock":"UNLOCK","fan_mode":"8","fan_speed":1,"fan_state":"ON","led_enable":true,"linkquality":90,"pm25":1,"replace_filter":false}'
publish {
  fan_state: 'ON',
  fan_speed: 1,
  pm25: 1,
  air_quality: 'good',
  led_enable: true,
  child_lock: 'UNLOCK',
  replace_filter: false,
  fan_mode: '8',
  linkquality: 90
}
```

sjorge, 2021-11-15:
As you can see from the fromZigbee call triggered by the attReport it does return a `state` variable that contains fan_speed update. But it never makes it.

Koenkk, 2021-11-15:
Looks like the issue already happens in the fromZigbee converter, can you add the following logging to it?

```js
if (msg.data.hasOwnProperty('fanSpeed')) {
    console.log('setting fan speed', msg.data['fanSpeed'])
    let fanSpeed = msg.data['fanSpeed'];
    console.log('fanspeed', fanSpeed);
    if (fanSpeed >= 10) {
        fanSpeed = (((fanSpeed - 5) * 2) / 10);
    } else {
        fanSpeed = 0;
    }

    state['fan_speed'] = fanSpeed;
    console.log('SET IN STATE', state['fan_speed'])
}
```

sjorge, 2021-11-15:
I'll add those as wel when I am back at the PC, but I had those (same locations) in the original PR for debugging and it all looked good. If it was messed up there I'd expect it not to work with optimistic turned off either but it does.

Also see https://github.com/Koenkk/zigbee2mqtt/issues/9658#issuecomment-969196522, where I added an extra console.log just before the return and it logs the data as expected.

```
Zigbee2MQTT:debug 2021-11-15 18:28:20: Received Zigbee message from 'airpurifier/bedroom', type 'attributeReport', cluster 'manuSpecificIkeaAirPurifier', data '{"fanMode":45,"fanSpeed":45}' from endpoint 1 with groupID 0
fromZigbee: state='{"fan_speed":8,"fan_mode":"8","fan_state":"ON"}' data='{"fanSpeed":45,"fanMode":45}'
```

Given `fan_speed` made it into the `state` var just before it is returned.

sjorge, 2021-11-15:
@Koenkk with the ones you asked, and mine at the end still in place:
```
Zigbee2MQTT:debug 2021-11-15 19:26:27: Received MQTT message on 'zigbee2mqtt/airpurifier/bedroom/set' with data '{"fan_mode":"7"}'
Zigbee2MQTT:debug 2021-11-15 19:26:27: Publishing 'set' 'fan_mode' to 'airpurifier/bedroom'
Zigbee2MQTT:debug 2021-11-15 19:26:27: Received Zigbee message from 'airpurifier/bedroom', type 'attributeReport', cluster 'manuSpecificIkeaAirPurifier', data '{"fanMode":40,"fanSpeed":40}' from endpoint 1 with groupID 0
setting fan speed 40
fanspeed 40
SET IN STATE 7
fromZigbee: state='{"fan_speed":7,"fan_mode":"7","fan_state":"ON"}' data='{"fanSpeed":40,"fanMode":40}'
PUBLISH PUBLISH { fan_mode: '7', fan_state: 'ON' }
payload { fan_mode: '7', fan_state: 'ON' }
newstate {
  fan_state: 'ON',
  fan_speed: 1,
  pm25: 0,
  air_quality: 'good',
  led_enable: true,
  child_lock: 'UNLOCK',
  replace_filter: false,
  fan_mode: '7'
}
Zigbee2MQTT:info  2021-11-15 19:26:27: MQTT publish: topic 'zigbee2mqtt/airpurifier/bedroom', payload '{"air_quality":"good","child_lock":"UNLOCK","fan_mode":"7","fan_speed":1,"fan_state":"ON","led_enable":true,"linkquality":63,"pm25":0,"replace_filter":false}'
publish {
  fan_state: 'ON',
  fan_speed: 1,
  pm25: 0,
  air_quality: 'good',
  led_enable: true,
  child_lock: 'UNLOCK',
  replace_filter: false,
  fan_mode: '7',
  linkquality: 63
}
payload {}
newstate {
  fan_state: 'ON',
  fan_speed: 1,
  pm25: 0,
  air_quality: 'good',
  led_enable: true,
  child_lock: 'UNLOCK',
  replace_filter: false,
  fan_mode: '7'
}
Zigbee2MQTT:info  2021-11-15 19:26:27: MQTT publish: topic 'zigbee2mqtt/airpurifier/bedroom', payload '{"air_quality":"good","child_lock":"UNLOCK","fan_mode":"7","fan_speed":1,"fan_state":"ON","led_enable":true,"linkquality":63,"pm25":0,"replace_filter":false}'
publish {
  fan_state: 'ON',
  fan_speed: 1,
  pm25: 0,
  air_quality: 'good',
  led_enable: true,
  child_lock: 'UNLOCK',
  replace_filter: false,
  fan_mode: '7',
  linkquality: 63
}
```

Koenkk, 2021-11-16:
Are you using debounce? Otherwise I would expect this line to be logged: https://github.com/Koenkk/zigbee2mqtt/blob/fdae2607bffef3e439a14e7965db5ce42095aec3/lib/extension/receive.ts#L145

sjorge, 2021-11-16:
Yes, debounce is also used

~ sjorge

Koenkk, 2021-11-16:
Is it fixed after applying the latest changes from https://github.com/Koenkk/zigbee2mqtt/pull/9682/files ?

sjorge, 2021-11-16:
```
Zigbee2MQTT:debug 2021-11-16 20:15:40: Received MQTT message on 'zigbee2mqtt/airpurifier/bedroom/set' with data '{"fan_mode":"5"}'
Zigbee2MQTT:debug 2021-11-16 20:15:40: Publishing 'set' 'fan_mode' to 'airpurifier/bedroom'
Zigbee2MQTT:debug 2021-11-16 20:15:41: Received Zigbee message from 'airpurifier/bedroom', type 'attributeReport', cluster 'manuSpecificIkeaAirPurifier', data '{"fanMode":30,"fanSpeed":30}' from endpoint 1 with groupID 0
PUBLISH PUBLISH { fan_mode: '5', fan_state: 'ON' }
payload { fan_mode: '5', fan_state: 'ON' }
newstate {
  fan_state: 'ON',
  fan_speed: 8,
  pm25: 1,
  air_quality: 'good',
  led_enable: false,
  child_lock: 'UNLOCK',
  replace_filter: false,
  fan_mode: '5'
}
Zigbee2MQTT:info  2021-11-16 20:15:41: MQTT publish: topic 'zigbee2mqtt/airpurifier/bedroom', payload '{"air_quality":"good","child_lock":"UNLOCK","fan_mode":"5","fan_speed":8,"fan_state":"ON","led_enable":false,"linkquality":96,"pm25":1,"replace_filter":false}'
publish {
  fan_state: 'ON',
  fan_speed: 8,
  pm25: 1,
  air_quality: 'good',
  led_enable: false,
  child_lock: 'UNLOCK',
  replace_filter: false,
  fan_mode: '5',
  linkquality: 96
}
payload { fan_speed: 5 }
newstate {
  fan_state: 'ON',
  fan_speed: 5,
  pm25: 1,
  air_quality: 'good',
  led_enable: false,
  child_lock: 'UNLOCK',
  replace_filter: false,
  fan_mode: '5'
}
Zigbee2MQTT:info  2021-11-16 20:15:41: MQTT publish: topic 'zigbee2mqtt/airpurifier/bedroom', payload '{"air_quality":"good","child_lock":"UNLOCK","fan_mode":"5","fan_speed":5,"fan_state":"ON","led_enable":false,"linkquality":96,"pm25":1,"replace_filter":false}'
```

That seems to have fixed it.

Koenkk, 2021-11-17:
Great, merged #9682

closed 2021-11-17 by Koenkk