Koenkk/zigbee2mqtt #9659: PIR Sensor (ZM-35ZH-Q) - OW-ZPS2

bobendylon, 2021-11-14:
## Information about the device + link
Hello, hope you're well!
Should it be possible to have a look on this demand please?

PIR Sensor (ZM-35ZH-Q)
https://fr.aliexpress.com/item/1005002419717912.html?spm=a2g0s.9042311.0.0.27426c37kuFC7W

## data/database.db entry of the device
{« id »:7,« type »:« Unknown »,« ieeeAddr »:« 0x00124b000b452149 »,« nwkAddr »:6371,« epList »:[1],« endpoints »:{« 1 »:{« epId »:1,« inClusterList »:[],« outClusterList »:[],« clusters »:{« genPowerCfg »:{« attributes »:{« batteryVoltage »:30,« batteryPercentageRemaining »:200}},« ssIasZone »:{« attributes »:{« zoneStatus »:0}}},« binds »:[],« configuredReportings »:[],« meta »:{}}},« interviewCompleted »:false,« meta »:{},« lastSeen »:1635082951458}

{"id":7,"type":"Unknown","ieeeAddr":"0x00124b000b452149","nwkAddr":6371,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":200}},"ssIasZone":{"attributes":{"zoneStatus":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1635523218520}
[image]
[image]

Faustino-ai, 2021-11-15:
I would suggest that you try it yourself according to the guide and post anything that you do not understand. There are already many PIR sensors available so it should be relatively straight forward to add support yourself.

ulinieto, 2021-11-22:
Are there any update about this sensor?? I don't know how to make the converter.js to incluide it.

Faustino-ai, 2021-11-23:
> Are there any update about this sensor?? I don't know how to make the converter.js to include it.

Again, try to follow the documentation and try for yourself. Ask what you don't understand in more details to show that you've tried. I was pretty clueless as well few months back but now I think I am ok thanks to the community's help. Seriously try it yourself. The community is only as helpful as how much effort you put in to understand the basics, or else there will be bunch of words that you won't know the meaning of flying around.

slavik73, 2021-12-20:
I have something that looks the same and dev branch already supports it ( {modelID: 'TS0202', manufacturerName: '_TZ3000_msl6wxk9'},). Except I'm not seeing a battery percentage, but it may be something that does not exists

Faustino-ai, 2021-12-21:
> I have something that looks the same and dev branch already supports it ( {modelID: 'TS0202', manufacturerName: '_TZ3000_msl6wxk9'},). Except I'm not seeing a battery percentage, but it may be something that does not exists

I have similar issues as well and I suspect that it has something to do with the different DPID that this unit uses that is different for others for the battery reporting. I think it is good to follow this guide https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html
From this you can find out the data points and try it one by one to see which one is actually reporting the battery. Mostly likely it will not be in percentage, but high, medium and low.

closed 2022-01-28 by github-actions[bot]