Koenkk/zigbee2mqtt #9665: YSR-MINI-01-rgbcct recognized as a Zemismart bulb

csszokolai, 2021-11-15:
## What happened
YSR-MINI-01-rgbcct recognized as a Zemismart bulb (ZB-RGBCW)
## What did you expect to happen
After normal pairing, the device recognized as a Zemismart bulb
`Friendly name
0xa4c138b40c233cac
Last seen
N/A
Device type
Router
Zigbee Model
ZB-CL01
Zigbee Manufacturer
eWeLight
Description
Zigbee 3.0 LED-bulb, RGBW LED
Support status
Supported
IEEE Address
0xa4c138b40c233cac
Network address
0xDF44
Firmware build date
20210907
Firmware version
2.0.1
Manufacturer
Lonsonho
Model
ZB-RGBCW
Power
Interview completed
True`
## How to reproduce it (minimal and precise)
removing and pairing again
## Debug info
Zigbee2MQTT version: 1.22.0
Adapter hardware: CC2652P (https://github.com/Gio-dot/Z-Bee-Duo)
Adapter firmware version: 20210319 (zStack3x0)
`Info Device '0xa4c138b40c233cac' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138b40c233cac","ieee_address":"0xa4c138b40c233cac"},"type":"device_joined"}'
Info Configuring '0xa4c138b40c233cac'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xa4c138b40c233cac"},"type":"device_connected"}'
Info MQTT publish: topic 'homeassistant/light/0xa4c138b40c233cac/light/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"brightness":true,"brightness_scale":254,"color_mode":true,"command_topic":"zigbee2mqtt/0xa4c138b40c233cac/set","device":{"identifiers":["zigbee2mqtt_0xa4c138b40c233cac"],"manufacturer":"Lonsonho","model":"Zigbee 3.0 LED-bulb, RGBW LED (ZB-RGBCW)","name":"0xa4c138b40c233cac","sw_version":"Zigbee2MQTT 1.22.0"},"json_attributes_topic":"zigbee2mqtt/0xa4c138b40c233cac","max_mireds":500,"min_mireds":153,"name":"0xa4c138b40c233cac","schema":"json","state_topic":"zigbee2mqtt/0xa4c138b40c233cac","supported_color_modes":["xy","color_temp"],"unique_id":"0xa4c138b40c233cac_light_zigbee2mqtt"}'
Info MQTT publish: topic 'homeassistant/sensor/0xa4c138b40c233cac/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138b40c233cac"],"manufacturer":"Lonsonho","model":"Zigbee 3.0 LED-bulb, RGBW LED (ZB-RGBCW)","name":"0xa4c138b40c233cac","sw_version":"Zigbee2MQTT 1.22.0"},"enabled_by_default":false,"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0xa4c138b40c233cac","name":"0xa4c138b40c233cac linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138b40c233cac","unique_id":"0xa4c138b40c233cac_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Debug Device '0xa4c138b40c233cac' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138b40c233cac","ieee_address":"0xa4c138b40c233cac"},"type":"device_announce"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xa4c138b40c233cac"},"type":"device_announced"}'
Debug Received MQTT message on 'homeassistant/light/0xa4c138b40c233cac/light/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"brightness":true,"brightness_scale":254,"color_mode":true,"command_topic":"zigbee2mqtt/0xa4c138b40c233cac/set","device":{"identifiers":["zigbee2mqtt_0xa4c138b40c233cac"],"manufacturer":"Lonsonho","model":"Zigbee 3.0 LED-bulb, RGBW LED (ZB-RGBCW)","name":"0xa4c138b40c233cac","sw_version":"Zigbee2MQTT 1.22.0"},"json_attributes_topic":"zigbee2mqtt/0xa4c138b40c233cac","max_mireds":500,"min_mireds":153,"name":"0xa4c138b40c233cac","schema":"json","state_topic":"zigbee2mqtt/0xa4c138b40c233cac","supported_color_modes":["xy","color_temp"],"unique_id":"0xa4c138b40c233cac_light_zigbee2mqtt"}'
Debug Received MQTT message on 'homeassistant/sensor/0xa4c138b40c233cac/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138b40c233cac"],"manufacturer":"Lonsonho","model":"Zigbee 3.0 LED-bulb, RGBW LED (ZB-RGBCW)","name":"0xa4c138b40c233cac","sw_version":"Zigbee2MQTT 1.22.0"},"enabled_by_default":false,"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0xa4c138b40c233cac","name":"0xa4c138b40c233cac linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138b40c233cac","unique_id":"0xa4c138b40c233cac_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Debug Retrieving state of '0xa4c138b40c233cac' after reconnect
Debug Received Zigbee message from '0xa4c138b40c233cac', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0xa4c138b40c233cac', payload '{"linkquality":108,"state":"ON"}'
Debug Received Zigbee message from '0xa4c138b40c233cac', type 'readResponse', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0xa4c138b40c233cac', payload '{"brightness":254,"linkquality":102,"state":"ON"}'
Debug Received Zigbee message from '0xa4c138b40c233cac', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorMode":2,"colorTemperature":454,"currentSaturation":0,"currentX":24939,"currentY":24701}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0xa4c138b40c233cac', payload '{"brightness":254,"color":{"h":27,"hue":27,"s":92,"saturation":92,"x":0.5056,"y":0.4152},"color_mode":"color_temp","color_temp":454,"linkquality":102,"state":"ON"}'
Error Failed to read state of '0xa4c138b40c233cac' after reconnect (Read 0xa4c138b40c233cac/1 lightingColorCtrl(["colorMode","currentX","currentY","enhancedCurrentHue","currentSaturation","colorTemperature"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE'))
Debug Received Zigbee message from '0xa4c138b40c233cac', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorCapabilities":25}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0xa4c138b40c233cac', payload '{"brightness":254,"color":{"h":27,"hue":27,"s":92,"saturation":92,"x":0.5056,"y":0.4152},"color_mode":"color_temp","color_temp":454,"linkquality":99,"state":"ON"}'
Debug Received Zigbee message from '0xa4c138b40c233cac', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorTempPhysicalMax":454,"colorTempPhysicalMin":250}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0xa4c138b40c233cac', payload '{"brightness":254,"color":{"h":27,"hue":27,"s":92,"saturation":92,"x":0.5056,"y":0.4152},"color_mode":"color_temp","color_temp":454,"linkquality":105,"state":"ON"}'
Info Successfully configured '0xa4c138b40c233cac'`

Martin004, 2021-11-22:
to solve the problem you will need the `database.db` info from your device.
See my Post https://github.com/Koenkk/zigbee2mqtt/issues/9545
.
Your Fingerprint manufacturerName is wrong.
manufacturerName  from your device is 'Lonsonho' .
The fingerprint is set to 'YSRSAI'.

csszokolai, 2021-11-23:
Thank you for your answer. I saw your other post. Maybe the solution is same?

Here is my database.db entry from the device:

`{"id":22,"type":"Router","ieeeAddr":"0xa4c138b40c233cac","nwkAddr":44660,"manufId":4107,"manufName":"eWeLight","powerSource":"Mains (single phase)","modelId":"ZB-CL01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"ZB-CL01","manufacturerName":"eWeLight","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":2,"hwVersion":0,"dateCode":"20210907","swBuildId":"2.0.1"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":250,"colorTempPhysicalMax":454}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":0,"dateCode":"20210907","swBuildId":"2.0.1","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1637650000837}`

Thank you for your help.

Martin004, 2021-11-23:
The fingerprint should be the problem.

Please try to change the file ysrsai.js like this.

`const extend = require('../lib/extend');

module.exports = [
    {
        fingerprint: [{inClusterList: [0,3,4,5,6,8,768,4096]}],
        zigbeeModel: ['ZB-CL03', 'FB56-ZCW20FB1.2','ZB-CL01'],
        model: 'YSR-MINI-01_rgbcct',
        vendor: 'YSRSAI',
        description: 'Zigbee LED controller (RGB+CCT)',
        extend: extend.light_onoff_brightness_colortemp_color({disableColorTempStartup: true, colorTempRange: [160, 370]}),
    },
    {
		fingerprint: [{inClusterList: [0,4,3,6,8,5,768,4096]}],
		zigbeeModel: ['ZB-CT01', 'FNB56-ZCW25FB2.1'],
        model: 'YSR-MINI-01_wwcw',
        vendor: 'YSRSAI',
        description: 'Zigbee LED controller (WW/CW)',
        extend: extend.light_onoff_brightness_colortemp_color({disableColorTempStartup: true}),
    },
    {
        zigbeeModel: ['ZB-DL01'],
        model: 'YSR-MINI-01_dimmer',
        vendor: 'YSRSAI',
        description: 'Zigbee LED controller (Dimmer)',
        extend: extend.light_onoff_brightness(),
    },
];
`

csszokolai, 2021-11-23:
i changed the code in the ysrsai.js, after that i reconfigured the device.  same result. After that i force removed the device, and added it again but nothing happened. I rebooted the entire system, and i tried several version of adding, reconfiguring, deleting, etc, but the device recognisedas the same as first time

Martin004, 2021-11-25:
Are you sure the file change has been applied?
Has the change been made in a Docker container or outside of it? Changes in a Docker image are lost when you restart.

see:
https://phoenixnap.com/kb/how-to-commit-changes-to-docker-image

csszokolai, 2021-11-27:
I do not know :( i changed the file via samba share. thanks for the help. i am not smart... i cannot test it.

csszokolai, 2021-11-29:
Here is my another rgbcct, it works perfectly
```
{"id":17,"type":"Router","ieeeAddr":"0x00124b0022eced92","nwkAddr":12201,"manufId":0,"manufName":"YSRSAI","powerSource":"Mains (single phase)","modelId":"ZB-CL01","epList":[1,13],"endpoints":{"1":{"profId":49246,"epId":1,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"ZB-CL01","manufacturerName":"YSRSAI","powerSource":1,"zclVersion":1,"appVersion":7,"stackVersion":2,"hwVersion":1,"dateCode":"20200914","swBuildId":"1.0.7"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":160,"colorTempPhysicalMax":370,"colorMode":2,"currentX":20480,"currentY":21558,"enhancedCurrentHue":34816,"currentSaturation":5,"colorTemperature":370}},"genLevelCtrl":{"attributes":{"currentLevel":15}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":7,"stackVersion":2,"hwVersion":1,"dateCode":"20200914","swBuildId":"1.0.7","zclVersion":1,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1638175058217}
```

Martin004, 2021-11-29:
please send your  database.db entry with both YSR-MINI-01_rgbcct 's . We will see the different. 
Perhaps it will help to modify the ysrsai.js file in the next dev.

Martin004, 2021-11-29:
Koenkk please add the non working device into your dev.
csszokolai can test the functionality.
I belive it is only a brandig problem.

Koenkk, 2021-11-29:
I do not think we can distinguish these, it is only a cosmetic branding. If it is really a problem for you:
- stop z2m
- change `manufName` to `YSRSAI` for the non working one in data/database.db
- start z2m

csszokolai, 2021-11-29:
> I do not think we can distinguish these, it is only a cosmetic branding. If it is really a problem for you:
> 
> * stop z2m
> * change `manufName` to `YSRSAI` for the non working one in data/database.db
> * start z2m

It works! Thank you!

closed 2021-11-29 by Koenkk

csszokolai, 2021-11-29:
There are some additional restrictions.
Debug Publishing 'set' 'color_temp' to '0xa4c138b40c233cac'
Debug Requested color_temp 160 is lower than minimum supported 250, using minimum!
And i found this:
```
Info MQTT publish: topic 'zigbee2mqtt/0xa4c138b40c233cac', payload '{"brightness":254,"color":{"h":40,"hue":40,"s":49,"saturation":49,"x":0.3804,"y":0.3767},"color_mode":"color_temp","color_temp":250,"linkquality":144,"state":"ON"}'
Error Publish 'get' 'color' to '0xa4c138b40c233cac' failed: 'Error: Read 0xa4c138b40c233cac/1 lightingColorCtrl(["colorMode","currentX","currentY","enhancedCurrentHue","currentSaturation","colorTemperature"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug Error: Read 0xa4c138b40c233cac/1 lightingColorCtrl(["colorMode","currentX","currentY","enhancedCurrentHue","currentSaturation","colorTemperature"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:293:23) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:413:22) at processTicksAndRejections (internal/process/task_queues.js:95:5) at Object.convertGet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:1062:13) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:273:21)
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'color' to '0xa4c138b40c233cac' failed: 'Error: Read 0xa4c138b40c233cac/1 lightingColorCtrl([\"colorMode\",\"currentX\",\"currentY\",\"enhancedCurrentHue\",\"currentSaturation\",\"colorTemperature\"], {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'","meta":{"friendly_name":"0xa4c138b40c233cac"},"type":"zigbee_publish_error"}'
```

csszokolai, 2021-11-29:
I changed the db again. :) I set the colorTempPhysicalMin attribute to 160. So now I can control the led. Thank you for your help all!