Koenkk/zigbee2mqtt #9671: Curtain switch doesn'r work fullt with HA

razzi71, 2021-11-15:
Dear Team,

My installation is HA in docker on Debian. I have a zigbee2mqtt the newest current version.
I have paired curtain switch with HA. I have a message like below:
Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_yenbr4om'
I tried to add this manufacturer to the new *.js file, next to configuration.yaml file, like desrcibed here: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html , but I cannot setup position / callibration and so on.
I have a possibility only to open/close or stop curtain. It is a chance to make this curtain supported ?

Many thanks in advance,
Best Regards,
Razzi71

FelipeSanchezCalzada, 2021-11-24:
I'm having exactly the same problem with _TZE200_nogaemzt (Tuya Smart Curtain Motor)

Opening and closing is useless if the real position cannot be read, the motor and the curtain will end up broken!

razzi71, 2021-11-25:
Hi,
I didn't find unfortunatly. I tried to edit tuya.js file, but without success. Only UP, DOWN, STOP controlling was possible...
Best Regards

FelipeSanchezCalzada, 2021-12-07:
I will expose what has happened to me with the Tuya Curtain Motor:

This Tuya Curtain Motor configures itself and you do not need to set the maximum and minimum. When it detects that it cannot continue to open / close, the motor stops and automatically sets the limits.
Set position in percentage remains locked until the curtain has set the maximum and minimum.

Additionally, I have not found a way to reverse the direction of rotation, for this I paired the motor to a Tuya HUB and from the official app I reversed the direction (There is an option to do it from the app). Later I have re-paired with zigbee2mqtt and the direction of rotation was already correct.

razzi71, 2021-12-09:
Hi,
I have a conbee II, zigbee2mqtt and Home Assistant. I am not using any cloud solution at home. Producer told me, that this particular curtain switch has no option to percentage setup. In my case hardware is useless. Will be changed for different one with options, what I need - posiotion, callibratrion and reversal mode. This is not true, that this automatically detects when curtain is fully open or closed. There is hardcoded time for opening and closing.

closed 2022-01-16 by github-actions[bot]

GitHubHAuser, 2024-06-22:
> Hi, I have a conbee II, zigbee2mqtt and Home Assistant. I am not using any cloud solution at home. Producer told me, that this particular curtain switch has no option to percentage setup. In my case [...]

Hello,
Same problem with this device.  Did the manufacturer not entrust you with a fw update that would solve the problem?  Exactly what device are you looking for/choosing that is suitable for all functions?