Koenkk/zigbee2mqtt #9677: SMSZB-120 missing temperature

Inrego, 2021-11-15:
# Bug Report
## What happened
I've just set up this device https://www.zigbee2mqtt.io/devices/SMSZB-120.html#develco-smszb-120
And I'm not getting any temperature readings. Here's the state json:

```
{
    "battery": 100,
    "battery_low": false,
    "linkquality": 255,
    "restore_reports": true,
    "smoke": false,
    "supervision_reports": true,
    "test": false,
    "update": {
        "state": "idle"
    },
    "update_available": false,
    "voltage": 3200,
    "alarm": null,
    "fault": null,
    "max_duration": null,
    "reliability": null,
    "temperature": null
}
```

## What did you expect to happen
Showing temperature

## How to reproduce it (minimal and precise)
Set up that specific device.

## Debug Info
Zigbee2MQTT version: 1.22.0 commit: 0a1fccf
Adapter hardware: Conbee II
Adapter firmware version: Not sure, but z2m about page shows Coordinate revision: 0x26680700

rindlerblabla, 2021-11-15:
I think there's some compatibility issue with conbee and those smoke detectors. Or, maybe it is just a coincidence? There are other users with conbee reporting missing temperature values here as well https://github.com/Koenkk/zigbee2mqtt/issues/9235.

Inrego, 2021-11-15:
Hmm, didn't find that when searching, but it does indeed look like the same issue!

closed 2021-11-15 by Inrego