Koenkk/zigbee2mqtt #9678: Is 2-second latency between Zigbee button and Zigbee bulb normal?

radicalmachine, 2021-11-15:
## Bug Report
Approximately 2 second delay between a Zigbee Smart Button (Konke 2AJZ4KPKEY) and a Philips Hue Bulb (model 929019531). Is this normal and expected, or should I expect better performance? No problem with this being closed if this is expected performance.

Bulb has instant response when toggling from HA directly

Have tried:

- Switching zigbee channel
- removing all devices so just the smart button, and one bulb

## What did you expect to happen
Expected there to be les latency when pressing a Zigbee button to toggle a Zigbee light bulb. Want to understand what in my critical path may be causing this delay

## How to reproduce it (minimal and precise)
Using single-click action from the button to toggle the Philips bulb via simple HA automation

## Debug Info
Raspberry Pi4 running Home Assistant OS 6.6
Supervisor-2021.10.8
Core-2021.11.3
Zigbee2MQTT version: 1.22.0-2
Adapter hardware: Raspbee II
Adapter firmware version: RaspBeeII_0x26720700

mmegaromm, 2021-11-15:
I also had delays, the kernel update helped #8663

radicalmachine, 2021-11-15:
> 
> 
> I also had delays, the kernel update helped #8663

Thanks fpr the link, interesting, but the Raspbee II doesn't use a CH340/341 controller and I'm on a later HAOS version

closed 2021-12-23 by github-actions[bot]