Koenkk/zigbee2mqtt #9679: Slaesh's CC2652RB network got unresponsive after ~1.5 days

mmakaresz, 2021-11-15:
## What happened
After  ~1.5 days all of the devices stop working, fully unresponsive. only z2mqtt restart makes the network "alive" again.
Herdsman log: https://pastebin.com/7tgdenj8
Last mqtt message: 021-11-15T09:57:33.019Z. after that the whole network is unresnponsive. 
## What did you expect to happen
Network work without restart 24/7
## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.12.0-2
Adapter hardware: Slaesh's CC2652RB
Adapter firmware version:
Debian11.
USB extension cable is used. 12 devices. Ikea and xiaomi only

RPiSpy, 2021-11-17:
I was experiencing a very similar issue with my zzh dongle. Restarting Zigbee2MQTT tended to work but not for more than a few hours. Given the logs were fairly clean I started to suspect it was a power issue. I had recently switched to using an SSD with my Pi 3B+. It used an official Pi 2.5A power supply but my theory was that the power draw on the USB was causing the zzh dongle to turn-off/restart/stop.

So today I swapped the Pi 3B+ for a Pi 4 and everything is fine. (I re-imaged the SSD with the Pi 4 Home Assistant image and restored a backup). The Pi 4 can provide more power to the USB ports and I suspect this solved the issue.

This may not be your issue but I thought I would mention it in case others find this post while fault-finding.

mmakaresz, 2021-11-18:
Hi! Thanks for the tip!
I am using an N3150 PC with an FSP DC/DC PSU, and a relatively new  adapter. I have an SSD for the system and a WD red for data and thats it. Before this adapter i have used an EZSP one with ZHA flawlessly...

changed the channel to 26, changed the pan id, the network key. moved the 2.4Ghz wifi to channel 1... i dont know what should be wrong..

Koenkk, 2021-11-18:
@mmakaresz could you try the fw from: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

mmakaresz, 2021-11-18:
@Koenkk thanks for the tip. I have updated the stick with this firmware. lets see

mmakaresz, 2021-11-19:
@Koenkk sadly tha same happened :(

Koenkk, 2021-11-19:
What is the output of `dmesg` right after the crash?

mmakaresz, 2021-11-19:
https://pastebin.com/X8iWkpfg
it is dead since ~10:00

Koenkk, 2021-11-19:
@mmakaresz not sure what happens but this is indeed strange. Looks like something in the Linux kernel to me, not a zigbee2mqtt issue for sure.

mmakaresz, 2021-11-19:
@Koenkk yes. But with ZHA and the EZSP stick it wasn't an issue to use the printer concurrently. Anyhow I have removed the printer. let's see. Thanks for your support and your work

mmakaresz, 2021-12-08:
Hi!
its stable since then! can be closed
thanks for your work

closed 2021-12-08 by Koenkk