Koenkk/zigbee2mqtt #9683: Does not work debounce

mmegaromm, 2021-11-15:
## What happened
Parameter does not work `debounce`

I install for 
`debounce: 1`
my electricity meter as it sends about 6-7 publications every 15 seconds
the problem appeared after updating to version `1.22.0`
[log 10 minutes.txt](https://github.com/Koenkk/zigbee2mqtt/files/7540905/log.10.minutes.txt)
in this log file you can see that in 10 minutes half of the messages from the electricity meter =(

On`1.21.0` worked well
sorry, I don’t have an old log file left where you can see that the parameter is working correctly there is such a screenshot
[image]

## Debug info
Zigbee2MQTT - 1.22.0

Adapter hardware:  СС2652

robertalexa, 2021-11-17:
I think I am experiencing a similar issue, but I have never used debounce before so I do not now what to expect. Maybe @sjorge or @Koenkk might be able to validate.

As of the latest release I have added debounce for a smart socket (https://www.zigbee2mqtt.io/devices/SA-003-Zigbee.html#ewelink-sa-003-zigbee) which is spitting out a lot of messages one after the other, but can't really understand why. I suspect either a crappy device or the device actually has more info than the converter is exposing (but I do not have a tuya gateway to sniff the traffic). The debounce: 1 however had no effect. I have restarted z2m, the setting was persistant, and same behaviour after the restart. Here is the log of the device output, when you turn in OFF

```
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office Marble Lamp Switch', payload '{"last_seen":1637141907586,"linkquality":75,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office Marble Lamp Switch', payload '{"last_seen":1637141907591,"linkquality":24,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office Marble Lamp Switch', payload '{"last_seen":1637141907598,"linkquality":24,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office Marble Lamp Switch', payload '{"last_seen":1637141907602,"linkquality":24,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office Marble Lamp Switch', payload '{"last_seen":1637141907609,"linkquality":24,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office Marble Lamp Switch', payload '{"last_seen":1637141907614,"linkquality":24,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office Marble Lamp Switch', payload '{"last_seen":1637141907619,"linkquality":24,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office Marble Lamp Switch', payload '{"last_seen":1637141907624,"linkquality":24,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office Marble Lamp Switch', payload '{"last_seen":1637141907631,"linkquality":24,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office Marble Lamp Switch', payload '{"last_seen":1637141907631,"linkquality":24,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office Marble Lamp Switch', payload '{"last_seen":1637141907631,"linkquality":24,"state":"OFF"}'
```

You can see from the timestamp these messages are either overlapping or fractions of a second after another.

Any feedback would be appreciated

sjorge, 2021-11-17:
There is another debounce related issue Koenkk is looking at https://github.com/Koenkk/zigbee2mqtt/pull/9682, don't think they are related though.

As for your device, debug should tell you whats triggering it, if it's a attributeReport you can probably tweak that on the reporting tab.

robertalexa, 2021-11-17:
Thanks Jorge, yeah I have seen that issue (in your Ikea PR) but do not know the ins and outs of it.

Here is the same logs but with debug level
```
Debug Received MQTT message on 'zigbee2mqtt/Rob's Office Marble Lamp Switch/set' with data 'ON'
Debug Publishing 'set' 'state' to 'Rob's Office Marble Lamp Switch'
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office Marble Lamp Switch', payload '{"last_seen":1637143181888,"linkquality":75,"state":"OFF"}'
Debug Received Zigbee message from 'Rob's Office Marble Lamp Switch', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office Marble Lamp Switch', payload '{"last_seen":1637143181896,"linkquality":51,"state":"OFF"}'
Debug Received Zigbee message from 'Rob's Office Marble Lamp Switch', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office Marble Lamp Switch', payload '{"last_seen":1637143181903,"linkquality":51,"state":"OFF"}'
Debug Received Zigbee message from 'Rob's Office Marble Lamp Switch', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office Marble Lamp Switch', payload '{"last_seen":1637143181912,"linkquality":51,"state":"OFF"}'
Debug Received Zigbee message from 'Rob's Office Marble Lamp Switch', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office Marble Lamp Switch', payload '{"last_seen":1637143181918,"linkquality":51,"state":"OFF"}'
Debug Received Zigbee message from 'Rob's Office Marble Lamp Switch', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office Marble Lamp Switch', payload '{"last_seen":1637143181925,"linkquality":51,"state":"OFF"}'
Debug Received Zigbee message from 'Rob's Office Marble Lamp Switch', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office Marble Lamp Switch', payload '{"last_seen":1637143181937,"linkquality":51,"state":"OFF"}'
Debug Received Zigbee message from 'Rob's Office Marble Lamp Switch', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office Marble Lamp Switch', payload '{"last_seen":1637143181943,"linkquality":51,"state":"OFF"}'
Debug Received Zigbee message from 'Rob's Office Marble Lamp Switch', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office Marble Lamp Switch', payload '{"last_seen":1637143181962,"linkquality":51,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office Marble Lamp Switch', payload '{"last_seen":1637143181962,"linkquality":51,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/Rob's Office Marble Lamp Switch', payload '{"last_seen":1637143181962,"linkquality":51,"state":"ON"}
```

And here are Bind and Reporting
[image]
[image]

And the converter
```javascript
    {
        zigbeeModel: ['SA-003-Zigbee'],
        model: 'SA-003-Zigbee',
        vendor: 'eWeLink',
        description: 'Zigbee smart plug',
        extend: extend.switch(),
        fromZigbee: [fz.on_off_skip_duplicate_transaction],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        },
        onEvent: async (type, data, device) => {
            device.skipDefaultResponse = true;
        },
    },
```

So with the above in mind, not sure what i can tweak, as technically there is no reporting...

sjorge, 2021-11-17:
Well it is receiving a report though... probably with some defaults.

```
Debug Received Zigbee message from 'Rob's Office Marble Lamp Switch', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
```

You can try manually adding one for genOnOff -> onOff, maybe something like:
Min=0, Max=3600, change=1

If that helps, we should probably do more than just bind it in the configure() and also set it up properly.

mmegaromm, 2021-12-01:
Fixed 1.22.1

noderat, 2021-12-24:
@mmegaromm What commit fixed this? I am still having this issue with the same device on 1.22.1

mmegaromm, 2022-01-01:
> @mmegaromm Какой коммит исправил это? У меня все еще проблема с тем же на 1.22.1

I thought they fixed it here    they discussed it there `debounce`
#9658

mmegaromm, 2022-01-01:
#9658

I thought they fixed it here they discussed it there

mmegaromm, 2022-01-01:
> @mmegaromm Какой коммит исправил это? У меня все еще проблема с тем же на 1.22.1

#9658

Я думал, они исправили это здесь, они обсудили это там

sgtwilko, 2022-01-23:
Maybe I'm using debounce wrong, but I'm also seeing multiple messages when I'm expecting only one.

The documentation suggests that debug is in seconds, I've tried setting it to 1, to 10 to 60, doesn't seem to matter what the value is, the messages arrive just as fast as before.

I've attached a debug log:
[zigbe2mqtt_notDebouncing.txt](https://github.com/Koenkk/zigbee2mqtt/files/7921255/zigbe2mqtt_notDebouncing.txt)

antbarney, 2022-02-06:
I also am noticing this behavior. I don't recall ever having to set the debounce before, but even after enabling it I have a number of outlets plugs that report repeatedly I have tried setting debounce to 1 thru 5 and no difference. 

Debug level logs showing the inbound zigbee messages as well as the multiple publishes.

[z2mqtt_debounce.log](https://github.com/Koenkk/zigbee2mqtt/files/8009394/z2mqtt_debounce.log)

closed 2022-03-17 by github-actions[bot]

kevincaradant, 2022-04-04:
> 

Hi 

Same issue with the same device 
Modèle Zigbee     TS0601
Constructeur Zigbee       _TZE200_dwcarsat

 I also have other one device which spamming so I'm interesting to know how this works because once that I use debounce to '1' as value, I got any message after that :(

Before to debounce '1'

```
Zigbee2MQTT:info  2022-04-04 12:04:07: MQTT publish: topic 'zigbee2mqtt/quality_air', payload '{"co2":758,"formaldehyd":84,"humidity":48,"linkquality":40,"pm25":8,"temperature":19,"voc":472}'
Zigbee2MQTT:info  2022-04-04 12:04:07: MQTT publish: topic 'zigbee2mqtt/quality_air', payload '{"co2":758,"formaldehyd":84,"humidity":48,"linkquality":43,"pm25":8,"temperature":19,"voc":472}'
Zigbee2MQTT:info  2022-04-04 12:04:09: MQTT publish: topic 'zigbee2mqtt/quality_air', payload '{"co2":758,"formaldehyd":87,"humidity":48,"linkquality":36,"pm25":8,"temperature":19,"voc":472}'
Zigbee2MQTT:info  2022-04-04 12:04:09: MQTT publish: topic 'zigbee2mqtt/quality_air', payload '{"co2":758,"formaldehyd":87,"humidity":48,"linkquality":36,"pm25":8,"temperature":19,"voc":493}'
Zigbee2MQTT:info  2022-04-04 12:04:09: MQTT publish: topic 'zigbee2mqtt/quality_air', payload '{"co2":515,"formaldehyd":87,"humidity":48,"linkquality":36,"pm25":8,"temperature":19,"voc":493}'
Zigbee2MQTT:info  2022-04-04 12:04:09: MQTT publish: topic 'zigbee2mqtt/quality_air', payload '{"co2":515,"formaldehyd":87,"humidity":48,"linkquality":36,"pm25":2,"temperature":19,"voc":493}'
Zigbee2MQTT:info  2022-04-04 12:04:15: MQTT publish: topic 'zigbee2mqtt/quality_air', payload '{"co2":515,"formaldehyd":87,"humidity":48,"linkquality":43,"pm25":2,"temperature":19,"voc":493}'
```

saspol, 2022-11-12:
Hello, i still experience same problem on 1.28.0
Does not matter  i set `debounce` to 1/0.1/0.001 - always got this 14 messages from device. Maybe i dont understand something how debouncing suppose to works?

```
MQTT publish: topic 'zigbee2mqtt/0x00124b002459eea6', payload 
'{"last_seen":1668229596992,"linkquality":72,"state":"OFF"}'
info 2022-11-12 13:06:36MQTT publish: topic 'zigbee2mqtt/0x00124b002459eea6', payload '{"last_seen":1668229597000,"linkquality":72,"state":"OFF"}'
info 2022-11-12 13:06:36MQTT publish: topic 'zigbee2mqtt/0x00124b002459eea6', payload '{"last_seen":1668229597000,"linkquality":72,"state":"OFF"}'
info 2022-11-12 13:06:36MQTT publish: topic 'zigbee2mqtt/0x00124b002459eea6', payload '{"last_seen":1668229597008,"linkquality":72,"state":"OFF"}'
info 2022-11-12 13:06:36MQTT publish: topic 'zigbee2mqtt/0x00124b002459eea6', payload '{"last_seen":1668229597008,"linkquality":72,"state":"OFF"}'
info 2022-11-12 13:06:36MQTT publish: topic 'zigbee2mqtt/0x00124b002459eea6', payload '{"last_seen":1668229597015,"linkquality":72,"state":"OFF"}'
info 2022-11-12 13:06:36MQTT publish: topic 'zigbee2mqtt/0x00124b002459eea6', payload '{"last_seen":1668229597015,"linkquality":72,"state":"OFF"}'
info 2022-11-12 13:06:36MQTT publish: topic 'zigbee2mqtt/0x00124b002459eea6', payload '{"last_seen":1668229597021,"linkquality":72,"state":"OFF"}'
info 2022-11-12 13:06:36MQTT publish: topic 'zigbee2mqtt/0x00124b002459eea6', payload '{"last_seen":1668229597021,"linkquality":72,"state":"OFF"}'
info 2022-11-12 13:06:36MQTT publish: topic 'zigbee2mqtt/0x00124b002459eea6', payload '{"last_seen":1668229597030,"linkquality":72,"state":"OFF"}'
info 2022-11-12 13:06:36MQTT publish: topic 'zigbee2mqtt/0x00124b002459eea6', payload '{"last_seen":1668229597030,"linkquality":72,"state":"OFF"}'
info 2022-11-12 13:06:36MQTT publish: topic 'zigbee2mqtt/0x00124b002459eea6', payload '{"last_seen":1668229597030,"linkquality":72,"state":"OFF"}'
info 2022-11-12 13:06:37MQTT publish: topic 'zigbee2mqtt/0x00124b002459eea6', payload '{"last_seen":1668229597030,"linkquality":72,"state":"OFF"}'
```

thargy, 2023-04-18:
I'm still seeing this issue too

richardcrabtree, 2023-05-09:
I'm seeing this issue. Some recent network issues are causing my button messages to become duplicated and debounce is not filtering them. Instead, It seems like it is delaying the original message by the specified time in seconds. That's not the expected behavior.

vindolin, 2023-11-28:
I'm having the same issue with [some Ikea bulbs](https://www.zigbee2mqtt.io/devices/LED2003G10.html#ikea-led2003g10).
When I toggle the state I get two messages, one with the old state and one with the new one.
My workaround is to use a delay node in Node-RED.

treemonkey83, 2023-12-22:
Im have the same issue with a zigbee plug https://www.zigbee2mqtt.io/devices/BSD29.html#woolley-bsd29
its constantly reporting its state although only when the switch/relay state is "on", in the "off" state i get nothing??

hackepeterOli, 2024-01-14:
Same issue here with a TS0601

elenril2, 2024-02-24:
Debounce does not work for the device BHT-002-GCLZB either. The [documentation ](https://www.zigbee2mqtt.io/devices/BHT-002-GCLZB.html#stop-message-flooding)advises to use this function, but the device still sends several dozen messages per second.

nonamejuju, 2024-03-27:
Still same issue here with a TS0601 power clamp

gras55, 2024-09-02:
> Same issue here with a TS0601

> Debounce does not work for the device BHT-002-GCLZB either. The [documentation ](https://www.zigbee2mqtt.io/devices/BHT-002-GCLZB.html#stop-message-flooding)advises to use this function, but the device still sends several dozen messages per second.

 Confirmed still an issue

Sortberg, 2025-01-02:
Anyone who has solved the problem?

foddy-foo, 2025-02-03:
I'm also still seeing this problem. I have a Tuya PJ-1203A which is reporting multiple times per second and setting the debounce to 1,2 3 or 5 for example (and restarting Zigbee2mqtt) does not help.