Koenkk/zigbee2mqtt #9691: temperature_precision not working for RTCGQ11LM

simonedimarzio, 2021-11-16:
# Bug Report
## What happened
The parameter "**temperature_precision**" is not taken into account using _RTCGQ11LM_ (Aqara motion sensor).
Decimal places are not sent, the temperature is only available as integer.

## What did you expect to happen
Setting "**temperature_precision**" to 1 or 2, temperature should include decimals.

## How to reproduce it (minimal and precise)
Set "**temperature_precision**" in configuration.yaml (common or device specific options).
Start z2m and wait for RTCGQ11LM message.
The temperature is an integer even though the "options" object contains the _temperature_precision_ parameter.

## Debug Info
Zigbee2MQTT version: v.1.22.0
Adapter hardware: CC2652P (ZigStar Stick v4)
Adapter firmware version: zStack3x0 20210708

`Nov 16 10:31:30 raspberrypi npm[28675]: Zigbee2MQTT:info  2021-11-16 10:31:30: MQTT publish: topic 'zigbee2mqtt/0x00158d0003f3c595', payload '{"battery":100,"illuminance":7,"illuminance_lux":7,"linkquality":168,"occupancy":true,"temperature":25,"voltage":3025}'`

ReneTode, 2021-11-16:
i got the same with an efekta_pws sensor.

ReneTode, 2021-12-17:
the problem is still active for me.

simonedimarzio, 2021-12-18:
Also for me the problem is still present, running last beta version.
May be the case that temp sensor in RTCGQ11LM is crappy and doesn’t have decimal precision? 
But in this case why the example about temperature_precision in z2m documentation is done on RTCGQ11LM?
Thanks
Simone

atomic7777, 2021-12-18:
I have the same problem with RTCGQ11LM. But at the same time I saw on the internet that it works for others (on the home assistant forum, also using zigbee2mqtt). So it could be a hardware issue in some series, or maybe there are actually different models under the same name

ReneTode, 2021-12-19:
i got the efekta_pws sensor.
which is a DIY sensor.
the settings dont work for temp, but they do for other things.

i also got other temp sensors, but they give xx.x by default.

ReneTode, 2022-01-24:
its still there.
is there anything changed for this?

Sequadon, 2022-02-09:
Same with the RTCGQ11LM. 
I changed temperature_precision 1 (was 0) in WebGUI, also confirmed in devices.yaml.
Zigbee2mqtt has been restarted, after several days, reported temperature is still integer.
Zigbee2mqtt: 1.22.2-1
RTCGQ11LM firmware version: 3000-0001 (20170627)

closed 2022-03-19 by github-actions[bot]

mikkelsiggaard, 2022-07-20:
Same issue with lumi.remote.b1acn01. 
Firmware build date: 20180525
Firmware version: 3000-0001

> "battery": 100,
    "device_temperature": 33,
    "linkquality": 131,
    "power_outage_count": 59,
    "voltage": 3035

As seen above, in "State" in Z2M, the "device_temperature" is only reported without decimals. 
However, I don't know if this is after it has been parsed through Z2M, so maybe the problem is that these devices simply report "device_temperature" without decimals. 

I do have an Aqara Weather - WSDCGQ11LM (temp, hum and pressure) that reports temp with 2 decimals, and I didn't have to adjust the "temperature_precision". Actually it is set to "0" as default, but reports with 2 decimals. 

[image]