Koenkk/zigbee2mqtt #9692: CR Smart Home TS0218 now recognized as TuYa TS0207_water_leak_detector

groenmarsmannetje, 2021-11-16:
## What happened
I have a [CR Smart Home TS0218](https://www.zigbee2mqtt.io/devices/TS0218.html) water leak sensor which is all of a sudden recognized as a [TuYa TS0207_water_leak_detector](TuYa TS0207_water_leak_detector). As you can see these are 2 totally different devices. I also notice that after water leak detection the device keeps on reporting leakage while it is already totally dry for a couple of days. My assumption is that this might be caused by the fact that it is using the incorrect device setup.

## What did you expect to happen

The correct devices is configured after setup and water leakage sensor turns off when it is dry.

## How to reproduce it (minimal and precise)

It just happened, not sure when. 

## Debug info
Zigbee2MQTT version: 1.22.0-dev commit: fa1775d
Adapter hardware: ZZH
Adapter firmware version: 20210708

sjorge, 2021-11-16:
Looks like the CR Smart Home TS0218 is marked as a white label of the TuYA branded one. So thats to be expected I think.

Koenkk, 2021-11-16:
This is indeed just a whitelabel device, see https://github.com/Koenkk/zigbee-herdsman-converters/pull/3005 for more info why this has been changed. Note that https://www.zigbee2mqtt.io/devices/TS0218.html is a button not a water leak sensor, you probably mean the TS0207

groenmarsmannetje, 2021-11-16:
Indeed my mistake. My water sensor looks similar as the button, like this:

  [image]

It now has a totally different picture and also the device/brand name has changed. This was not the case earlier.

Koenkk, 2021-11-16:
It's just the picture, the thing is that whitelabeled device cannot be distinguished from each other so the picture might not be fully correct.

groenmarsmannetje, 2021-11-16:
Okay, then I will check if it happens again that the sensor keeps on reporting leakage while it is dry. I was hoping that his could explain that unexpected behavior.

groenmarsmannetje, 2021-11-18:
The sensor automatically adjusted it state from wet to dry. Therefor I close this issue.

closed 2021-11-18 by groenmarsmannetje