Koenkk/zigbee2mqtt #9700: continuous ping to unplugged device

ster1um, 2021-11-16:
## What happened
Get continuous ping to unplugged device
```
Zigbee2MQTT:error 2021-11-16 20:15:28: Failed to ping 'Luci 2' (attempt 1/1, Read 0x60a423fffe7b5391/1 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Zigbee2MQTT:error 2021-11-16 20:18:32: Failed to ping 'Luci 2' (attempt 1/1, Read 0x60a423fffe7b5391/1 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Zigbee2MQTT:error 2021-11-16 20:21:36: Failed to ping 'Luci 2' (attempt 1/1, Read 0x60a423fffe7b5391/1 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Zigbee2MQTT:error 2021-11-16 20:24:40: Failed to ping 'Luci 2' (attempt 1/1, Read 0x60a423fffe7b5391/1 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Zigbee2MQTT:error 2021-11-16 20:27:44: Failed to ping 'Luci 2' (attempt 1/1, Read 0x60a423fffe7b5391/1 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
tResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Zigbee2MQTT:error 2021-11-16 20:33:52: Failed to ping 'Luci 2' (attempt 1/1, Read 0x60a423fffe7b5391/1 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Zigbee2MQTT:error 2021-11-16 20:36:56: Failed to ping 'Luci 2' (attempt 1/1, Read 0x60a423fffe7b5391/1 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Zigbee2MQTT:error 2021-11-16 20:40:00: Failed to ping 'Luci 2' (attempt 1/1, Read 0x60a423fffe7b5391/1 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Zigbee2MQTT:error 2021-11-16 20:43:04: Failed to ping 'Luci 2' (attempt 1/1, Read 0x60a423fffe7b5391/1 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Zigbee2MQTT:error 2021-11-16 20:46:09: Failed to ping 'Luci 2' (attempt 1/1, Read 0x60a423fffe7b5391/1 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Zigbee2MQTT:error 2021-11-16 20:49:13: Failed to ping 'Luci 2' (attempt 1/1, Read 0x60a423fffe7b5391/1 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Zigbee2MQTT:error 2021-11-16 20:52:17: Failed to ping 'Luci 2' (attempt 1/1, Read 0x60a423fffe7b5391/1 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Zigbee2MQTT:error 2021-11-16 20:55:21: Failed to ping 'Luci 2' (attempt 1/1, Read 0x60a423fffe7b5391/1 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Zigbee2MQTT:error 2021-11-16 20:58:25: Failed to ping 'Luci 2' (attempt 1/1, Read 0x60a423fffe7b5391/1 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Zigbee2MQTT:error 2021-11-16 21:01:29: Failed to ping 'Luci 2' (attempt 1/1, Read 0x60a423fffe7b5391/1 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Zigbee2MQTT:error 2021-11-16 21:04:33: Failed to ping 'Luci 2' (attempt 1/1, Read 0x60a423fffe7b5391/1 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Zigbee2MQTT:error 2021-11-16 21:07:37: Failed to ping 'Luci 2' (attempt 1/1, Read 0x60a423fffe7b5391/1 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Zigbee2MQTT:error 2021-11-16 21:10:41: Failed to ping 'Luci 2' (attempt 1/1, Read 0x60a423fffe7b5391/1 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
...
```
## What did you expect to happen
This behaviour is normal? 
is not possible set a number of count to system try to ping the device?

## How to reproduce it (minimal and precise)
always
 
## Debug info
Zigbee2MQTT version: 1.22.0
Adapter hardware: CC2652P
Adapter firmware version: 20210708

sjorge, 2021-11-16:
If you have availability set to true, this is normal.

You can change some of the defaults though: https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#availability-advanced-configuration

bartplessers, 2021-11-21:
Is there a way to avoid error messages for this?
Now my log is flooded with messages like:
`Zigbee2MQTT:error 2021-11-21 06:44:55: Failed to ping '0x842e14fffe22146f' (attempt 1/1, Read 0x842e14fffe22146f/1 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))`

It's good to now, but it's not really an error. Maybe move this event log level 'warning"?
 grtz
B

Koenkk, 2021-11-21:
@bartplessers make sense, will be logged as a warning now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2021-12-30 by github-actions[bot]