Koenkk/zigbee2mqtt #9706: Philips Gradient Light Strip no colors available

cbjerg, 2021-11-17:
## What happened
I have added the Philips Hue Gradient Light Strip, and it was correctly recognized as Philips 929002422801
On https://www.zigbee2mqtt.io/devices/929002422801.html#philips-929002422801 i can see that it only supports color temperature, and not color. Is there anything to do about this, or do I need to connect it back to the Hue Bridge

## What did you expect to happen
I've also added a regular Philips hue Light Strip (Philips 915005106701) and it has colors. I expected at least the same behaviour from the more powerful Gradient Strip

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.22.0-2
Adapter hardware: Conbee II
Adapter firmware version:

sjorge, 2021-11-17:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/8c1a1dc9b3ae2a0360cce352828eaf0306ae0345/devices/philips.js#L2101-L2109

Looks like it is added as a device that only supports color_temp.

From the description on the site it seems like it should support color, but I wonder if it's somehow different because it meant to sync to a TV.

@Koenkk Do you know if this should just have color_xy support like regular RGB strips form hue? Or was this omitted because issues?

Koenkk, 2021-11-17:
This device was added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/2733 by @Raspatat  , can you confirm this device does only support color temperature (and not RGB colors)?

Raspatat, 2021-11-17:
No, I cannot as it for sure supports RGB, works as such in my setup as well.

Koenkk, 2021-11-17:
@Raspatat thanks.

Added color support.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

cbjerg, 2021-11-18:
Thank you guys for the quick response - awesome

closed 2021-11-18 by cbjerg