Koenkk/zigbee2mqtt #9707: Last seen status is N/A on all devices

burnworld3d, 2021-11-17:
## What happened

On all devices, the last seen status is N/A

## What did you expect to happen

Get a proper last seen status, indicating how many seconds or minutes 

## How to reproduce it (minimal and precise)

https://puu.sh/IpKhv/819e5535bb.png

## Debug info
Zigbee2MQTT version: latest stable one
Adapter hardware: Conbee II
Adapter firmware version: latest stable one

francisp2, 2021-11-17:
You need to set ‘last_seen’ under the advanced section, the default is disabled.

burnworld3d, 2021-11-17:
> You need to set ‘last_seen’ under the advanced section, the default is disabled.

Can you please specify where is that option?
I can't see it, searched on all tabs, like https://puu.sh/IpKnP/1cb798f6d6.png and i don't find that option

burnworld3d, 2021-11-17:
ok i reply myself:

  # Optional: Add a last_seen attribute to MQTT messages, contains date/time of last Zigbee message
  # possible values are: disable (default), ISO_8601, ISO_8601_local, epoch (default: disable)
  last_seen: 'disable'

needs to be added in advanced inside zigbee2mqtt config file

closed 2021-11-17 by burnworld3d