Koenkk/zigbee2mqtt #9711: Sending update to unplugged device prevent other devices to communicate

vbambuch, 2021-11-17:
## What happened
Hi z2m community,

when I turn off (unplug from a power source) my Philips Hue bulb and try to change its state by sending `brightness, color_temp, hue_power_on_behavior` attributes the controller starts to produce `zigbee_publish_error` (viz [logs](https://pastebin.com/mt2m1M2Y)).
It takes several seconds the controller produces the sequence of errors while not processing any requests (publishing any updates) from other devices making the whole network irresponsible. It looks like the controller waits for a reply from the device and times out and it tries for each attribute separately.

Right now I "solved" it by checking the error message and turning the device off in software as well so my node-red stops sending the update requests. The thing is I see it as a dirty workaround because when I had the device connected via Philips Hue bridge it worked fine - not sure how the bridge handles such situations though.

Do you think running multiple z2m instances would solve it?
## What did you expect to happen
I would expect the behaviour to be asynchronous so the error handling doesn't block following requests. Honestly, I'm not sure why the controller waits for a reply if this should be a "publish/subscribe" model, shouldn't it?

I didn't find any option how to change the response timeout so it produces errors but quickly and without blocking the communication.
## How to reproduce it (minimal and precise)
* Switch off (unplug power source) a Philips Hue bulb
* Try to change its state
* Check the z2m logs for errors
* While seeing the errors try to change the state of connected devices
* The state change should take a couple of seconds 

## Debug info
Zigbee2MQTT version: 1.22.0
Adapter hardware: CC2531
Adapter firmware version: 20201127

Bascht74, 2021-11-19:
That is a problem for me, too. 
Thx to other people in my home that remove plugs...

vbambuch, 2021-12-20:
Hey @Koenkk, have you came across this issue before?

Let me know in case you need more information or any kind of help.

Koenkk, 2021-12-20:
@vbambuch the concurrency for the CC2531 is set to 2 by default, meaning that 2 commands can be active at a time. This is set so low in order to prevent the CC2531 from crashing (since it is not very powerful). You can experiment with increasing that value, see the `adapter_concurrent` option for more info: https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#transmitter-power

vbambuch, 2021-12-21:
Thanks for the reply. The setting of `adapter_concurrent: 3` did the trick. 🎉 

The errors are still in logs and appear regularly based on some internal timeout but other commands work parallelly. 

I think we can close the issue. 👍

closed 2021-12-21 by Koenkk