Koenkk/zigbee2mqtt #9712: Eurotronic SPZB0001 - System Mode not changeable

trashmaster9000, 2021-11-17:
# Bug Report
## What happened
Cant change "System Mode" of the Eurotronic Spirit Zigbee SPZB0001 Thermostat. After Setting it to "off" or "heat" i get the following Error message:

`Publish 'set' 'system_mode' to '0x00158d00053d36a6' failed: 'Error: Write 0x00158d00053d36a6/1 hvacThermostat({"16392":{"value":33,"type":34}}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4151,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_VALUE')'`

Worked Last Winter, did change the Coordinator/Stick and made a clean install of zigbee2mqtt between last winter and now.

## Debug Info
Zigbee2MQTT version: 1.22.0 (Home Assistant)
Adapter hardware: zzh!
Adapter firmware version: zStack3x0 20210708

Morphy99, 2021-11-22:
Do you have legacy mode set to false in the device settings (Specific)? I noticed that if I had it set to false as is "highly recommended" it stops the mode from changing to off.

trashmaster9000, 2021-11-25:
It doesnt make a difference if i set it to true or false :/

chino-lu, 2021-12-14:
so what I have identified is that the heat or off mode is not setting the system_mode but only occupiedHeatingSetpoint to either 30 (when heat pressed) or 0 (when off pressed)

in my opinion not how it should be

closed 2022-01-21 by github-actions[bot]