Koenkk/zigbee2mqtt #9715: Vimar 16943 actuator device

Nik71git, 2021-11-17:
create 16943.js file next to configuration.yaml, restart z2m but it doesn't work, below the main section:
 ```
 const definition = {
    zigbeeModel: ['myZigbeeModel'],
    model: '16493',
    vendor: 'Vimar',
    description: 'attuatore',
    fromZigbee: [fz.on_off],
    toZigbee: [tz.on_off],
    exposes: [e.switch()],
         
```

## Information about the device + link
..https://www.vimar.com/it/it/catalog/product/index/code/16493 it is an actuator for normal L+N 220VAC 

## data/database.db entry of the device
..https://download.vimar.com/irj/go/km/docs/z_catalogo/DOCUMENT/ZIS_20593EN_FI.97496.pdf

danieledwardgeorgehitchcock, 2021-11-18:
Hi, Is this the zigbeeModel that is reported by Zigbee2MQTT?

Please post the device entry from your database.db file

Nik71git, 2021-11-18:
No, It isn't. It's similar. I've also the device that you mean and It works
very well.
asap i Will post db

Nik71git, 2021-12-06:
here what I cane read from debug log: 
`"friendlyName":"presa bagno","ieeeAddr":"0xf4ce364d31442f50","lastSeen":1638609938309,"manufacturerName":"Vimar","modelID":"Mains_Power_Outlet_v1.0","networkAddress":5854,"type":"Router"}]}},"status":"ok","transaction":"x74p0-1"}'`

closed 2021-12-08 by Nik71git