Koenkk/zigbee2mqtt #9723: Invalue value for number - related to the boost time for a ZigBee TRV

gavyb, 2021-11-18:
Bug report?
- If applicable, provide steps how to reproduce the problem.
- Provide the herdsman debug logging: https://www.zigbee2mqtt.io/guide/usage/debug.html

-->

## What happened
My home-asistant.log file is filling up with the following errors:

2021-11-18 13:11:13 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.master_bedroom_trv_boost_time: 300 (range 0.0 - 100.0)

## What did you expect to happen

No such error messages. The TRV should be able to be set to Boost for 5 mins.

## How to reproduce it (minimal and precise)

There is nothing specific I have done to reproduce the error.
I am on HA 2021.11.3, using zigbee2mqtt Edge integration with Sonoff Zigbee Hub running Tasmota 9.5.0.1

## Debug info
Zigbee2MQTT version: zigbee2mqtt edge (1.22.0-dev commit: b2eaa31)
Adapter hardware: Sonoff zigbee bridge
Adapter firmware version: Tasmota 9.5.0.1

Koenkk, 2021-11-18:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

gavyb, 2021-11-18:
thanks @Koenkk - will look out for the update. I am already on the edge version.

gavyb, 2021-11-24:
thanks, that seems to have resolved that problem...