Koenkk/zigbee2mqtt #9735: Starting Zigbee2MQTT stops at  "Using experimental new availability feature"

Propland, 2021-11-18:
Hello,

I have installed Ubuntu Server 20.04.3 LTS 64bit on my RPI4 and use the CC2531.
I have flashed the stick and then followed https://www.zigbee2mqtt.io/guide/installation/01_linux.

I had two small problems but could fix both:
 - as i ran "node-pre-gyp info ok" the output was "/usr/bin/env: ‘nodejs’: No such file or directory" but i fixed it with "sudo ln -s /usr/bin/node /usr/bin/nodejs"
 - as I tried to start Zigbee2MQTT in /opt/zigbee2mqtt with "npm start" I  had an "Error whie starting zigbee-herdsman" but fixed it with https://www.zigbee2mqtt.io/guide/usage/debug.html#enabling-logging

My current problem is when I try to start the application in /opt/zigbee2mqtt with "npm start" it tries to use an experimental new feature.. it stops there and nothing will happen.
[image]

I even added "availability: false" to the configuration.yaml hopefully, but this doesnt solve the problem.
This change makes the startup-output-code (or however you call it) very long, but stops at the same line:
[image]
[image]
sorry for the long pictures..
This long output doesnt go away, even if i delete the added line in the configuration.yaml or set it in true.

I really hope you can help me.
Thanks in advance.

Propland, 2021-11-19:
I even tried https://www.zigbee2mqtt.io/guide/installation/01_linux on my laptop with ubuntu and on the other with mint.
On both the startup hangs up at "using an experimental new feature".
Am I the only one? Do you know this Problem?

Koenkk, 2021-11-19:
This issue may already have been fixed in the dev branch, could you check that? (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Propland, 2021-11-19:
This doesnt help.
It keeps trying to use the availability feature and stops there

Koenkk, 2021-11-19:
Could you provide the herdsman debug logging when trying to start?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Propland, 2021-11-20:
I start z2m with DEBUG=zigbee-herdsman* npm start, but it stops again at:
(sorry again for these long pictures, but theres much code. Can i send the pictures next time in a better way e.g. minimize the pictures somehow?)
[image]
Every Minute or so a new one of the similar code blocks appear.

When I Strg+c:
[image]
[image]
I had to press Strg+c again that ubuntu@ubuntu:/opt/zigbee2mqtt$ show in the last line

In the state.json file theres just:   {}

Koenkk, 2021-11-20:
Could you try flashing the source routing fw on your cc2531? https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

Propland, 2021-11-21:
Unfortunately it still stops at the same line..
[image]

Koenkk, 2021-11-21:
Could you provide your data/database.db and configuration.yaml?

Propland, 2021-11-22:
My configuration.yaml
[image]

I think my database.db is empty
[image]

francisp2, 2021-11-22:
I don't think it is a sqlite database. Just use nano to open database.db.

Propland, 2021-11-22:
oh.. sry:
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018e19044","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,10,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null}

Koenkk, 2021-11-22:
I just checked but this is actually the last thing being logged once starting is completed

> My current problem is when I try to start the application in /opt/zigbee2mqtt with "npm start" it tries to use an experimental new feature.. it stops there and nothing will happen.

What do you expect to happen after this?

Propland, 2021-11-22:
It won't start.
[image]

Koenkk, 2021-11-23:
What if you enable the frontend in your configuration: https://www.zigbee2mqtt.io/guide/configuration/frontend.html and execute `npm start` in the z2m install dir, can you open the frontend after this?

Propland, 2021-11-23:
Yes I can open the frontend. Thank you very much.

But I still cant start the deamon:
[image]

this is my systemctl file in /etc/systemd/system/zigbee2mqtt.service
[image]

Koenkk, 2021-11-24:
This probably means the permissions of the user you are running the deamon as are incorrect, try fixing that or change `User` to `root` in `zigbee2mqtt.service`.

Propland, 2021-11-25:
I tried to run the daemon as root but it failed again:
[image]

closed 2022-01-02 by github-actions[bot]