Koenkk/zigbee2mqtt #9742: Device sporadically switches ON

ChankyTheConqueror, 2021-11-19:
# Bug Report
## What happened
I have a smart switch module of type: **MS-104BZ** 
[image]

This device mysteriously switches itself ON without any MQTT request (there are no sets with retained flags or any other requests, the device is simply powered ON). I have been observing it for some days and now the only clue i have is this:
[image]

When the above messages come in (They always come in pair) both switches are set ON.
[image]

## What did you expect to happen
The device should only switch on when there is a request.
## How to reproduce it (minimal and precise)
Switches on sporadically
## Debug Info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1 or CC1352P-2
im using the zzh from electrolama

Adapter firmware version:
[image]

Any help in finding the issue will be highly appreciated

Timminater, 2021-11-22:
Check the timestamps the light turns on. A friend of me had the same problem and after weeks of searching we noticed this device switching on/off at the same time as the street lights switching. Maybe you have the same problem.

ChankyTheConqueror, 2021-11-26:
Hi Timminater,
mine is completely random since i posted the device hasnt switched on again by itself again.

ChankyTheConqueror, 2021-12-07:
The devices still switches on sporadically, anyone out here, who can help me debug the issue?

ChankyTheConqueror, 2022-01-07:
Is it really impossible to debug such an issue?

ChankyTheConqueror, 2022-02-07:
come on guys a little debugging help is all i need.

Koenkk, 2022-02-07:
Given that there is nothing in the logs besides `Received zigbee message..`, it seems z2m itself doesn't send this message. That leaves two other root causes:
- The firmware on the device is bugged, turning itself on
- Other devices in the network control this switch, to see if this is the case sniff the traffic while the switch switches on. https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

ChankyTheConqueror, 2022-02-13:
Hi @Koenkk,
i have captured the zigbee traffic where it is sporadically going ON,  how can i tell if another device is switching on? see attachment thanks in advance. The device Network ID is 0xcdbd

[zigbee_debug.zip](https://github.com/Koenkk/zigbee2mqtt/files/8056406/zigbee_debug.zip)

Koenkk, 2022-02-13:
@ChankyTheConqueror do you know at what +- No. (first column) the bulb turns off?

closed 2022-02-21 by github-actions[bot]