Koenkk/zigbee2mqtt #9747: Temp offset not working RTCGQ11LM

tungmeister, 2021-11-19:
I've just setup some new RTCGQ11LM's. Everything appears to be working however setting a temperature offset is having no impact on the reported temp.
the temperature is being reported as 5 degrees over the correct value:
[image]
I've adjusted to compensate and confirmed this is in the config file:
[image]
[image]
however, the temperature is still being reported as 30.

I've restarted zigbee2mqtt and applied a bit of heat to the unit however it's still reading 5 degrees over.

Koenkk, 2021-11-19:
This will be applied on the next temperature report of the sensor (could take some time, few hours)

tungmeister, 2021-11-19:
ah - I figured hitting it with the hairdryer would force it to report a new value - does it not provide the value every time it reports a new motion detection?

Koenkk, 2021-11-19:
No, only on an interval basis (the use case of its temperature sensor is therefore limited)

lux73, 2021-11-20:
the Temperature Value from the Aqara Motion Sensors are not reliable - even the Brightness Reports are ridiculous...

therefore using dedicated Xiaomi GZCGQ01LM Light Sensor for Brightness

some of my Aqara Door Contacts measure Temperatures 6-8 Degrees higher than my Aqara Temperatur Sensors - i would not trust them anymore ;-)

tungmeister, 2021-11-20:
Yeah, I'm not going to rely on it for anything, but it's nice to get an approximate temperature in a couple of rooms that didn't have sensors prevously. After 5 hours or so it started updating and adopting the offset value so working as expected.

closed 2021-11-20 by tungmeister