Koenkk/zigbee2mqtt #9748: TS0601_dimmer dropped off the network and will not re-pair

elgerg, 2021-11-19:
I have a TS0601_dimmer that seems to refuse to re-pair to my network. It was working fine for months then seemed to disappear.

Each time I try to re-pair it (using warning only logs) I get:
Warning Device 'SpareRoomLight' left the network
Warning Device 'SpareRoomLight' left the network
Warning Device 'SpareRoomLight' left the network

## What happened
The device "left" the network multiple times

## What did you expect to happen
I expected the device to re-join the network

## How to reproduce it (minimal and precise)
Press the power button 5 times then hold on the 6th
I'm not sure why this one dimmer will not re-pair. I can see it in the devices db:

            "0x5c0272fffec61638": {
                "friendly_name": "SpareRoomLight"
            },

## Debug info
Zigbee2MQTT version: 1.22.0
Adapter hardware: Slaesh's CC2652RB stick
Adapter firmware version: I cant remember, how do I find out?

elgerg, 2021-12-17:
Any ideas?

elgerg, 2021-12-17:
@Koenkk 
What is the best way of removing the device from the db file?
Thanks
Alex

elgerg, 2021-12-20:
@Koenkk  Any ideas?
Thanks

Koenkk, 2021-12-20:
Could you make a sniff when trying to pair this device?

https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

When sharing the `.pcapng` file please also share your network key

elgerg, 2022-01-17:
Sure will do when I have some time..

elgerg, 2022-02-17:
Still trying to get time to do the sniffing..

closed 2022-03-27 by github-actions[bot]