Koenkk/zigbee2mqtt #9752: WSDCGQ11LM seen as unsuported

Tonio16, 2021-11-19:
## What happened
I want to add some aqara WSDCGQ11LM to my zigbee network. I have already a bunch of them which were added months ago. Today, the two new I tried to add are seen as unsupported!

See log:
> [32mZigbee2MQTT:info [39m 2021-11-19 20:34:45: Removing device '0x00158d0005405537' (block: false, force: true)
[32mZigbee2MQTT:info [39m 2021-11-19 20:34:45: MQTT publish: topic 'zigbee2mqtt/0x00158d0005405537', payload ''
[32mZigbee2MQTT:info [39m 2021-11-19 20:34:45: Successfully removed device '0x00158d0005405537' (block: false, force: true)
[32mZigbee2MQTT:info [39m 2021-11-19 20:34:45: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{"block":false,"force":true,"id":"0x00158d0005405537"},"status":"ok","transaction":"9oavq-4"}'
2021-11-19T19:36:03.072Z zigbee-herdsman:adapter:zigate:driver:log { shortAddress: 47742, ieee: '0x00158d0005405537', MACcapability: { alternatePanCoordinator: false, fullFunctionDevice: false, mainsPowerSource: false, receiverOnWhenIdle: false, reserved: 0, securityCapability: false, allocateAddress: true } }
2021-11-19T19:36:03.073Z zigbee-herdsman:controller:log Device '0x00158d0005405537' joined
2021-11-19T19:36:03.074Z zigbee-herdsman:controller:log Device '0x00158d0005405537' accepted by handler
2021-11-19T19:36:03.088Z zigbee-herdsman:controller:log Delete device '0x00158d0005405537' joined, undeleting
[32mZigbee2MQTT:info [39m 2021-11-19 20:36:03: Device '0x00158d0005405537' joined
[32mZigbee2MQTT:info [39m 2021-11-19 20:36:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0005405537","ieee_address":"0x00158d0005405537"},"type":"device_joined"}'
[32mZigbee2MQTT:info [39m 2021-11-19 20:36:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00158d0005405537"},"type":"device_connected"}'
2021-11-19T19:36:03.698Z zigbee-herdsman:controller:log Device '0x00158d0005405537' is already in database with different networkAddress, updating networkAddress
2021-11-19T19:36:03.712Z zigbee-herdsman:controller:log Interview '0x00158d0005405537' start
[32mZigbee2MQTT:info [39m 2021-11-19 20:36:03: Starting interview of '0x00158d0005405537'
[32mZigbee2MQTT:info [39m 2021-11-19 20:36:04: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0005405537","ieee_address":"0x00158d0005405537","status":"started"},"type":"device_interview"}'
[32mZigbee2MQTT:info [39m 2021-11-19 20:36:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00158d0005405537"},"type":"pairing"}'
2021-11-19T19:36:04.125Z zigbee-herdsman:controller:device:log Interview - start device '0x00158d0005405537'
2021-11-19T19:36:14.139Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00158d0005405537', attempt 1
2021-11-19T19:36:24.165Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00158d0005405537', attempt 2
2021-11-19T19:36:34.171Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00158d0005405537', attempt 3
2021-11-19T19:36:44.178Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00158d0005405537', attempt 4
2021-11-19T19:36:54.184Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00158d0005405537', attempt 5
2021-11-19T19:37:04.190Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00158d0005405537', attempt 6
2021-11-19T19:37:04.193Z zigbee-herdsman:controller:device:log Interview - failed for device '0x00158d0005405537' with error 'Error: Interview failed because can not get node descriptor ('0x00158d0005405537')
2021-11-19T19:37:04.218Z zigbee-herdsman:controller:error Interview failed for '0x00158d0005405537 with error 'Error: Interview failed because can not get node descriptor ('0x00158d0005405537')'
[31mZigbee2MQTT:error[39m 2021-11-19 20:37:04: Failed to interview '0x00158d0005405537', device has not successfully been paired
[32mZigbee2MQTT:info [39m 2021-11-19 20:37:04: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0005405537","ieee_address":"0x00158d0005405537","status":"failed"},"type":"device_interview"}'
[32mZigbee2MQTT:info [39m 2021-11-19 20:37:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x00158d0005405537"},"type":"pairing"}'

## What did you expect to happen
To have a successfull pairing as in the past.
## How to reproduce it (minimal and precise)
I supposed to do the pairing of this device with z2m 1.22.0 and a zigate.
## Debug info
Version Zigbee2MQTT
1.22.0 commit: b3e9afaf
Type de Coordinateur
zigate
Révision du Coordinateur
1d03
Version Interface
0.6.30

danieledwardgeorgehitchcock, 2021-11-21:
Hi,

This looks as though the interview is not being completed.

Please make sure you keep the device awake during pairing, and nice and close to your coordinator

Tonio16, 2021-11-21:
> Hi,
> 
> This looks as though the interview is not being completed.
> 
> Please make sure you keep the device awake during pairing, and nice and close to your coordinator

I did it, of course. 

Antoine

fast0490f, 2021-11-22:
#9791 
Hi
a similar problem, I do everything quickly, the device and the zigbee stick are on the table
If you restart the Z2M service immediately after the interview, then everything works right away
Otherwise, you need to wait 1-2 minutes until the device_anonce message arrives.

after rebooting, it turns out to complete interviews immediately, permanently

Tonio16, 2021-11-22:
I will give a try.

Antoine

Tonio16, 2021-12-03:
Still no success to pair these devices though I have already done that in the past for the realy same one (same order).
Antoine

lrodorigo, 2021-12-17:
Exactly same problem here, brand new sensors, Ebyte CC2652P coordinator. The first time the device quickly joins but it fails to get descriptors (failed to interview device error), then it become quite impossible to retry....
really frustrating.

Tonio16, 2021-12-23:
I changed of key for a sonoff 3.0.0 tead. No trouble.

Tx

Antoine

closed 2021-12-23 by Tonio16

lrodorigo, 2021-12-30:
Also moving the firmware to Egony solved the issue on the CC2652P