Koenkk/zigbee2mqtt #9756: TypeError: Cannot read property 'ID' of undefined

paavokurkela, 2021-11-20:
Hey.
My network has been unchanged for a while and everything worked fine.
Just out of the blue one of my Ikea buttons didn't work anymore. Assumed battery and replaced it without success. I gave a look to logs and found this out:

https://pastebin.com/G8pmJTst

Effectively I can't add o delete any devices from web frontend or change log level but I can use most of my devices and my automations at Home Assistant / Nodered works.

I have TI2531 reflashed to latest source routing firmware. Home assistant runs in container under Debian and Z2M is on latest version as well.

Please see my config here: https://pastebin.com/5SKzwe5b

Cheers,
Paavo

Koenkk, 2021-11-20:
Could you provide your data/database.db?

paavokurkela, 2021-11-20:
Absolutely, please fin it here: https://pastebin.com/BF4PZC8E

Koenkk, 2021-11-20:
Thanks, could you also post your data/configuration.yaml? (not the one from HA)

paavokurkela, 2021-11-20:
Not sure do I follow. Would you like to see Z2M's config.yaml? 
https://pastebin.com/5SKzwe5b

Koenkk, 2021-11-20:
Could you:
- stop z2m
- remove this line from data/database.db: `{"id":14,"type":"Group","groupID":null,"members":[],"meta":{}}`
- start z2m
- see if they problem has been fixed

paavokurkela, 2021-11-20:
Yes, that does it!

I suspected it as well but didnnt realize z2m should be stop before editing database.

Thank you for your help!

closed 2021-11-20 by Koenkk