Koenkk/zigbee2mqtt #9757: Zigbee model 'TS0202' and manufacturer name '_TZ3000_mcxw5ehu'

chrisrock1984, 2021-11-20:
there seems to be several variants of the TS0202 motion sensor.  I already own 2 of them, which look the same but have a red led instead of a blue. I tried with existing converters but were not able to get it running. Pairing works but I do not get any data.

## Information about the device + link
https://de.aliexpress.com/item/1005003320192145.html?spm=a2g0s.9042311.0.0.27424c4dH1kHCA

## data/database.db entry of the device
..

chrisrock1984, 2021-11-23:
added database.db entry

GimpArm, 2021-11-24:
I just got a couple of these. What you have for the fromZigbee seems to work for me. The device does not stay on and I can trigger automations with it.

toZigbee mostly works too except keep_time. I get in the logs.

```
2021-11-24 21:17:14: Publish 'set' 'keep_time' to 'Hall 1 Movement' failed: 'Error: Write 0xa4c138843d6f5f00/1 ssIasZone({"61441":{"value":2,"type":32}}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 5517 - 1 - 181 - 1280 - 4 after 10000ms)'
```

chrisrock1984, 2021-11-25:
> I just got a couple of these. What you have for the fromZigbee seems to work for me. The device does not stay on and I can trigger automations with it.
>
> toZigbee mostly works too except keep_time. I [...]

And you really have the same manufacturer name: _TZ3000_mcxw5ehu' ?
I have a same looking one, which already is supported but ordered several months ago.
The new one - _TZ3000_mcxw5ehu' - is not supported, and with the external converter I only get some values.

GimpArm, 2021-11-25:
I have
Zigbee Model: TS0202
Zigbee Manufacturer: _TZ3000_mcxw5ehu

It shows up as unsupported on z2m 1.22.0-3 but when I add your converter it reads all the values.
I got them from the same Aliexpress seller as you although a different listing. https://de.aliexpress.com/item/1005003033573539.html?spm=a2g0s.9042311.0.0.72f74c4dklEhGx

chrisrock1984, 2021-11-26:
doesn't work for me ;-/

chrisrock1984, 2021-11-26:
the following converter works for me (except e.battery - maybe it only has battery_low)

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0202', manufacturerName: '_TZ3000_mcxw5ehu'}],
    model: 'TS0202',
    vendor: 'TuYa',
    description: 'Motion sensor',
    fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery, fz.ignore_basic_report, fz.ias_occupancy_alarm_1_report],
    toZigbee: [],
    exposes: [e.occupancy(), e.battery_low(), e.tamper(), e.battery()],
};

module.exports = definition;

@Koenkk Please add

GimpArm, 2021-11-26:
This also works for me except I didn't realize that the battery wasn't work for me either. I think you should remove `e.battery()` maybe it isn't supported. The rest look good and it detect motion. I thought sensitivity was working but now I'm not sure.

chrisrock1984, 2021-11-26:
its somehow not working stable - this is weird....

GimpArm, 2021-11-26:
For me, occupancy detection is stable. I have 5 of these and I've been testing with them, they do no miss anything.

Even though the device doesn't respond the keep_time is definitely doing something. Fresh out of the box there is a 60 second keep_time. Send `{"61441":{"value":0,"type":32}}`  and the keep_time is now 30 seconds. Send `{"61441":{"value":1,"type":32}}` and the keep_time is back to 60 seconds. So the UI is wrong and value 2 = 120 seconds, but values 3 and above do not do anything. This also has a tendency to crash the device. So sometimes it is sent and then the device is unresponsive for a few minutes but when it comes back then the keep_time is set correctly. So it is definitely doing something, just the values are not quite right. Maybe there shouldn't be an expectation of a response?

I have no idea if sensitivity does anything. That's a little harder to test and it results in the same timeout as keep_time. I'll try placing the sensor at various distances and see what happens

GimpArm, 2021-11-26:
I can't find a difference with the sensitivity. It doesn't change the detection distance. It also didn't change whether my cat was detected (she wasn't).

chrisrock1984, 2021-11-27:
can you send your external converter please?

GimpArm, 2021-11-27:
I just copied the ZB003X that you used but removed the parts that weren't needed and change the keep_time values.

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
  fingerprint: [{modelID: 'TS0202', manufacturerName: '_TZ3000_mcxw5ehu'}],
  model: 'TS0202',
  vendor: 'TuYa',
  description: 'Motion sensor',
  fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery, fz.ignore_basic_report, fz.ias_occupancy_alarm_1_report],
  toZigbee: [{
      key: ['keep_time'],
      convertSet: async (entity, key, value, meta) => {
            switch (key) {
            case 'keep_time':
                await entity.write('ssIasZone', {61441: {value: {'30': 0, '60': 1, '120': 2}[value], type: 0x20}});
                break;
            default: // Unknown key
                throw new Error(`Unhandled key ${key}`);
            }
        },
  }],
  exposes: [e.occupancy(), e.battery_low(), e.tamper(),
    exposes
      .enum("sensitivity", ea.STATE_SET, ["low", "medium", "high"])
      .withDescription("PIR sensor sensitivity"),
    exposes
      .enum("keep_time", ea.STATE_SET, ["30", "60", "120"])
      .withDescription("PIR keep time in seconds"),
  ],
};

module.exports = definition;`

GimpArm, 2021-11-30:
A little more playing with it. They seem inconsistent although in operation I have no issues with them detecting occupancy.

This is my current configuration.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
  fingerprint: [{modelID: 'TS0202', manufacturerName: '_TZ3000_mcxw5ehu'}],
  model: 'TS0202',
  vendor: 'TuYa',
  description: 'Motion sensor',
  configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['ssIasZone']);
    },
  fromZigbee: [fz.ias_occupancy_alarm_1, fz.ias_occupancy_alarm_1_report, fz.ignore_basic_report,
    {
        cluster: 'ssIasZone',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            if (msg.data.hasOwnProperty('currentZoneSensitivityLevel')) {
                const sensitivityLookup = {'low': 0, 'medium': 1, 'high': 2};
                result.sensitivity = sensitivityLookup[msg.data['currentZoneSensitivityLevel']];
            }
            if (msg.data.hasOwnProperty('61441')) {
                const keepTimeLookup = {'30': 0, '60': 1, '120': 2};
                result.keep_time = keepTimeLookup[msg.data['61441']];
            }
            return result;
        },
    }],
  toZigbee: [{
      key: ['sensitivity', 'keep_time'],
      convertSet: async (entity, key, value, meta) => {
            switch (key) {
            case 'sensitivity':
                const sensitivityLookup = {'low': 0, 'medium': 1, 'high': 2};
                await entity.write('ssIasZone', {currentZoneSensitivityLevel: sensitivityLookup[value]}, {disableResponse: true});
                return {readAfterWriteTime: 1000, state: {sensitivity: value}};
            case 'keep_time':
                const keepTimeLookup = {'30': 0, '60': 1, '120': 2};
                await entity.write('ssIasZone', {61441: {value: keepTimeLookup[value], type: 0x20}}, {disableResponse: true});
                return {readAfterWriteTime: 1000, state: {keep_time: value}};
            default: // Unknown key
                throw new Error(`Unhandled key ${key}`);
            }
        },
  }],
  exposes: [e.occupancy(), e.battery_low(), e.tamper(),
    exposes
      .enum('sensitivity', ea.STATE_SET, ['low', 'medium', 'high'])
      .withDescription('PIR sensor sensitivity'),
    exposes
      .enum("keep_time", ea.STATE_SET, ["30", "60", "120"])
      .withDescription("PIR keep time in seconds"),
  ],
};

module.exports = definition;
```

thenakuls, 2021-12-01:
Just received this device today and the manufacturer name is _TYZB01_hqbdru35
Will try adding it later tonight

thenakuls, 2021-12-03:
Changing manufacturer name in the converter posted in [https://github.com/Koenkk/zigbee2mqtt/issues/9757#issuecomment-982699090](url) worked.

chrisrock1984, 2021-12-04:
> Changing manufacturer name in the converter posted in [#9757 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/9757#issuecomment-982699090) worked.

now this one works for me too ;-)
thx!!

@Koenkk can you please add this one? thx!

Koenkk, 2021-12-04:
Added both _TZ3000_mcxw5ehu and _TYZB01_hqbdru35, assuming this can be closed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2021-12-04 by Koenkk

jvandenbroek, 2021-12-14:
Another one: _TZ3000_msl6wxk9
Bought here: https://nl.aliexpress.com/item/1005003518834102.html

Koenkk, 2021-12-14:
@jvandenbroek added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Roelski, 2021-12-14:
Hi all. Just stumbled on this topic as I was exploring how to add a new entry.
I have sensor TZ3000_msl6wxk9 which is bought here: https://nl.aliexpress.com/item/1005003639654220.html?spm=a2g0o.productlist.0.0.78595f2e2pC5Rp&algo_pvid=be990d25-4693-435f-bb81-8e33ea155ca3&algo_exp_id=be990d25-4693-435f-bb81-8e33ea155ca3-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000026604863571%22%7D

Box say's model ZM-RT20, and sellers site say's model is 40ZH-O.
Looks a bit different than previous post, but all settings and template stuff apply.
Have not yet tried changing keep time, but according to debug log, and picture on vendor site from tuya app both keep_sime and sensitivity should be able to be changed.
Hope this helps someone!

Jarda-cz, 2022-01-11:
Hello,
here is another similar type.
Zigbee Model:   TS0202
Zigbee Manufacturer:   _TZ3000_bsvqrxru

https://www.aliexpress.com/item/1005003500671455.html?spm=a2g0o.9042311.0.0.27424c4dJELF5r

THX

Koenkk, 2022-01-12:
@Jarda-cz added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

JohnMcLear, 2022-07-19:
I'm on 1.26.0-1  and I still needed to include this converter so I think this issue is persisting with some variants available on Aliexpress.  With or without this converter no values are ever shown in "exposes"

With this converter I successfully pair and see the values but occupancy is never updated from "Clear" despite the item flashing to show it has detected occupancy.

```
info 2022-07-19 11:42:36Accepting joining not in blocklist device '0xa4c138daf6cd1e25'
debug 2022-07-19 11:42:36Device '0xa4c138daf6cd1e25' announced itself
info 2022-07-19 11:42:36MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138daf6cd1e25","ieee_address":"0xa4c138daf6cd1e25"},"type":"device_announce"}'
info 2022-07-19 11:42:36MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xa4c138daf6cd1e25"},"type":"device_announced"}'
debug 2022-07-19 11:42:38Retrieving state of '0xa4c138daf6cd1e25' after reconnect
info 2022-07-19 11:43:01Accepting joining not in blocklist device '0xa4c138daf6cd1e25'
debug 2022-07-19 11:43:02Device '0xa4c138daf6cd1e25' announced itself
info 2022-07-19 11:43:02MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138daf6cd1e25","ieee_address":"0xa4c138daf6cd1e25"},"type":"device_announce"}'
info 2022-07-19 11:43:02MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xa4c138daf6cd1e25"},"type":"device_announced"}'
debug 2022-07-19 11:43:04Retrieving state of '0xa4c138daf6cd1e25' after reconnect
```

Note that it never seems to get ``state``

brianmay, 2022-08-03:
This 'appears' to work with https://www.aliexpress.com/item/1005003041905466.html, at least for occupancy. No idea if `battery_low` and `tamper` are correct.

Incoming data looks like:

```
Aug 04 09:05:36 zigbee2mqtt npm[23708]: Zigbee2MQTT:debug 2022-08-04 09:05:36: Received Zigbee message from 'Brian's PIR', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
```

I only use the `zonestatus`, no idea what `extendedstatus`, always appears to be 0.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0202', manufacturerName: '_TZ3000_lltemgsf'}],
    model: 'TS0202',
    vendor: 'TUYA',
    description: 'Tuya Smart ZigBee 3.0 Battery Alexa PIR Motion Sensor',
    fromZigbee: [fz.ias_occupancy_alarm_1],
    toZigbee: [],
    exposes: [e.occupancy(), e.battery_low(), e.tamper()], 
};

module.exports = definition;
```

I think it should be possible to program some stuff in this sensor, such as delays. At least there is a video on the page that implies it should be possible. But no idea what the requests should look like. And the example above didn't appear to work for me. I got errors like the following. Plus no change in the delay.

```
Aug 04 07:47:42 zigbee2mqtt npm[21189]: Zigbee2MQTT:debug 2022-08-04 07:47:42: Publishing 'set' 'keep_time' to 'Brian's PIR'
Aug 04 07:47:54 zigbee2mqtt npm[21189]: Zigbee2MQTT:error 2022-08-04 07:47:54: Publish 'set' 'keep_time' to 'Brian's PIR' failed: 'Error: Write 0xa4c1383aaf96fb69/1 ssIasZone({"61441":{"value":1,"type":32}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'
Aug 04 07:47:54 zigbee2mqtt npm[21189]: Zigbee2MQTT:debug 2022-08-04 07:47:54: Error: Write 0xa4c1383aaf96fb69/1 ssIasZone({"61441":{"value":1,"type":32}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))
Aug 04 07:47:54 zigbee2mqtt npm[21189]:     at ZStackAdapter.sendZclFrameToEndpointInternal (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:414:23)
Aug 04 07:47:54 zigbee2mqtt npm[21189]:     at Queue.executeNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
```

sebastian-tennant, 2022-09-10:
I think there's another type - recently bought one and it's come with the following :)

**Zigbee Model** TS0202
**Zigbee Manufacturer** _TZ3000_mg4dy6z6
**Support status** Not supported

sebastian-tennant, 2022-09-10:
Looks like I bought an new model

**Zigbee Model** TS0202
**Zigbee Manufacturer** _TZ3000_mg4dy6z6
**Support status** Not supported

Koenkk, 2022-09-11:
@sebastian-tennant  

Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0202', manufacturerName: '_TYZB01_jytabjkb'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_lltemgsf'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_mg4dy6z6'}],
    model: 'TS0202_1',
    vendor: 'TuYa',
    description: 'Motion sensor',
    // Requires alarm_1_with_timeout https://github.com/Koenkk/zigbee2mqtt/issues/2818#issuecomment-776119586
    fromZigbee: [fz.ias_occupancy_alarm_1_with_timeout, fz.battery, fz.ignore_basic_report],
    toZigbee: [],
    exposes: [e.occupancy(), e.battery_low(), e.linkquality(), e.battery(), e.battery_voltage()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        await reporting.batteryPercentageRemaining(endpoint);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if device works

sebastian-tennant, 2022-09-11:
> @sebastian-tennant
> Could you check if the issue is fixed with the following external converter:

Thank you, that does make it work. Apologies for posting in two locations, I figured this would remain closed.

kubbur, 2022-09-16:
Any idea if the light in the sensor can be disabled ?
It flashes blue when it detects occupansy

hangon1234, 2022-09-16:
> @sebastian-tennant
>
> Could you check if the issue is fixed with the following external converter:
>
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = [...]

I'm using latest Docker image. When this device will be supported by default? Of course, I already pulled latest arm7 image from official docker image. Thanks!

fvillena, 2022-10-19:
I bought a new device on 2022-10-06 and is reporting itself as _TZ3040_6ygjfyll

I bought it here: https://www.aliexpress.com/item/1005003187399208.html?spm=a2g0o.order_list.0.0.49281802L6i21h

cawidtu, 2022-10-27:
I seem to have another clone of a Tuya motion sensor, bought on 2022-10-25 from proshop.dk. The device has modelID TS0202 and manufacturerID _TZ3000_mg4dy6z6 .

cawidtu, 2022-10-27:
The _TZ3000_mg4dy6z6 variant works fine for me with the changes in https://github.com/Koenkk/zigbee2mqtt/issues/9757#issuecomment-1242909728

martindell, 2022-10-31:
Recent purchase from Ali Express, seems to work OK with updated converter
added: {modelID: 'TS0202', manufacturerName: '_TZ3040_6ygjfyll'},

Zigbee Model: TS0202
Zigbee Manufacturer: _TZ3040_6ygjfyll

> @sebastian-tennant
>
> Could you check if the issue is fixed with the following external converter:
>
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = [...]

patrick-foubert-56770, 2022-11-21:
Hi same issue with device from aliexpress

Friendly name
0x8cf681fffec16d44
Description
Last seen
N/A
Availability
[Disabled](https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#availability-advanced-configuration)
Device type
EndDevice
Zigbee Model
TS0202
Zigbee Manufacturer
_TZ3210_wuhzzfqg
Support status
Not supported

IEEE Address
0x8cf681fffec16d44
Network address
0x5A5C
Power
Battery 
Interview completed
True

patrick-foubert-56770, 2022-11-21:
After adding converter above detection (PIR) ok. but i got some errors:

Error 2022-11-21 15:08:44Failed to configure '0x8cf681fffec16d44', attempt 1 (Error: ConfigureReporting 0x8cf681fffec16d44/1 genPowerCfg([{"attribute":"batteryPercentageRemaining","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUP_GENERAL_COMMAND') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:694:22) at Object.batteryPercentageRemaining (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:61:9) at Object.configure (/app/dist/util/externally-loaded.js:23:9) at Configure.configure (/app/lib/extension/configure.ts:115:13))
Error 2022-11-21 15:09:07Failed to configure '0x8cf681fffec16d44', attempt 2 (Error: ConfigureReporting 0x8cf681fffec16d44/1 genPowerCfg([{"attribute":"batteryPercentageRemaining","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUP_GENERAL_COMMAND') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:694:22) at Object.batteryPercentageRemaining (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:61:9) at Object.configure (/app/dist/util/externally-loaded.js:23:9) at Configure.configure (/app/lib/extension/configure.ts:115:13))
Error 2022-11-21 15:09:15Failed to configure '0x8cf681fffec16d44', attempt 3 (Error: ConfigureReporting 0x8cf681fffec16d44/1 genPowerCfg([{"attribute":"batteryPercentageRemaining","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUP_GENERAL_COMMAND') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:694:22) at Object.batteryPercentageRemaining (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:61:9) at Object.configure (/app/dist/util/externally-loaded.js:23:9) at Configure.configure (/app/lib/extension/configure.ts:115:13))

a2nt, 2022-12-10:
> Hi same issue with device from aliexpress
>
> Friendly name 0x8cf681fffec16d44 Description Last seen N/A Availability [...]

Hey Patrick, try this:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0202', manufacturerName: '_TZ3210_wuhzzfqg'}],
        model: 'ZB003-X',
        vendor: 'Fantem',
        description: '4 in 1 multi sensor',
        fromZigbee: [fz.battery, fz.ignore_basic_report, fz.illuminance, fz.ZB003X, fz.ZB003X_attr, fz.ZB003X_$
        toZigbee: [tz.ZB003X],
        exposes: [e.occupancy(), e.tamper(), e.illuminance_lux(), e.illuminance(), e.temperature(), e.humidity$
            e.battery(), e.battery_voltage(),
            exposes.numeric('battery2', ea.STATE).withUnit('%').withDescription('Remaining battery 2 in %'),
            exposes.numeric('illuminance_calibration', ea.STATE_SET).withDescription('Illuminance calibration $
                .withValueMin(-20).withValueMax(20),
            exposes.numeric('temperature_calibration', ea.STATE_SET).withDescription('Temperature calibration $
                .withValueMin(-2).withValueMax(2).withValueStep(0.1),
            exposes.numeric('humidity_calibration', ea.STATE_SET).withDescription('Humidity calibration')
                .withValueMin(-15).withValueMax(15),
            exposes.binary('reporting_enable', ea.STATE_SET, true, false).withDescription('Enable reporting'),
            exposes.numeric('reporting_time', ea.STATE_SET).withDescription('Reporting interval in minutes')
                .withValueMin(0).withValueMax(1440).withValueStep(5),
            exposes.binary('led_enable', ea.STATE_SET, true, false).withDescription('Enable LED'),
            exposes.binary('pir_enable', ea.STATE_SET, true, false).withDescription('Enable PIR sensor'),
            exposes.enum('sensitivity', ea.STATE_SET, ['low', 'medium', 'high']).withDescription('PIR sensor s$
            // eslint-disable-next-line
            exposes.enum('keep_time', ea.STATE_SET, ['0', '30', '60', '120', '240', '480'])
                .withDescription('PIR keep time in seconds')],
};

module.exports = definition;
```

zigamilek, 2022-12-13:
@Koenkk I have another unsupported variant. Can confirm using this `{ modelID: 'TS0202', manufacturerName: '_TZ3000_6ygjfyll' }` in the external converter file works.

Koenkk, 2022-12-13:
@zigamilek added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Peetz0r, 2022-12-22:
> I bought a new device on 2022-10-06 and is reporting itself as _TZ3040_6ygjfyll
> 
> I bought it here: https://www.aliexpress.com/item/1005003187399208.html?spm=a2g0o.order_list.0.0.49281802L6i21h

This is exactly the same product listing as the one I ordered, on 2022-12-10

> @Koenkk I have another unsupported variant. Can confirm using this `{ modelID: 'TS0202', manufacturerName: '_TZ3000_6ygjfyll' }` in the external converter file works.

This is exactly how mine identifies.

> @zigamilek added!
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

This sensor looks more like the TuYa IH012-RT01 and less like the Fantem ZB003-X. It works, but it's showing N/A fields because it's not a 4-in-1 sensor but just a plain old PIR sensor.

x-magic, 2022-12-30:
> This sensor looks more like the TuYa IH012-RT01 and less like the Fantem ZB003-X. It works, but it's showing N/A fields because it's not a 4-in-1 sensor but just a plain old PIR sensor.

I agree. Just got this exact sensor delivered. Works with IH012-RT01 profile instead of ZB003-X. I think I better send a pull request to make this change happen. Not sure the consequence of keeping the entry in both manufacturer files so let me give that a try...

cloudbr34k84, 2023-01-05:
I have been trying to follow this thread as I have 
_TZ3000_mcxw5ehu
Model: TuYa IH012-RT01

How do i add the Converter for this??

x-magic, 2023-01-05:
> I have been trying to follow this thread as I have _TZ3000_mcxw5ehu Model: TuYa IH012-RT01
> 
> How do i add the Converter for this??

I can see your model in the source already. 

https://github.com/x-magic/zigbee-herdsman-converters/blob/2c11201ac414f514735849d645b40d2eb94427ce/devices/tuya.js#L1147

Looks like been there for a while. If you upgrade Z2M it should work out of the box.

cloudbr34k84, 2023-01-05:
Hey, yeah it works but I'm.tryinh to get the power monitoring entities to work or power, current.

gu33mis, 2023-01-09:
I bought few TS0202 _TZ3000_6ygjfyll sensors from Aliexpress and it look like battery and voltage information are always showing Null. Is this expected or am I missing something?

I'm running the latest zigbee2mqtt Docker image, 1.29.1 commit: 7d67ffc

destman, 2023-02-15:
I bought TS0202 device with manufacturer set to "_TYZB01_5nr7ncpl". After adding to definition array - it works fine.
Thx :)

> > @sebastian-tennant
> > Could you check if the issue is fixed with the following external converter:
> > ```js
> > const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> > const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> > const exposes = require('zigbee-herdsman-converters/lib/exposes');
> > const reporting = require('zigbee-herdsman-converters/lib/reporting');
> > const extend = require('zigbee-herdsman-converters/lib/extend');
> > const ota = require('zigbee-herdsman-converters/lib/ota');
> > const tuya = require('zigbee-herdsman-converters/lib/tuya');
> > const e = exposes.presets;
> > const ea = exposes.access;
> > 
> > const definition = {
> >     fingerprint: [{modelID: 'TS0202', manufacturerName: '_TYZB01_jytabjkb'},
> >         {modelID: 'TS0202', manufacturerName: '_TZ3000_lltemgsf'},
> >         {modelID: 'TS0202', manufacturerName: '_TZ3000_mg4dy6z6'}],
> >     model: 'TS0202_1',
> >     vendor: 'TuYa',
> >     description: 'Motion sensor',
> >     // Requires alarm_1_with_timeout https://github.com/Koenkk/zigbee2mqtt/issues/2818#issuecomment-776119586
> >     fromZigbee: [fz.ias_occupancy_alarm_1_with_timeout, fz.battery, fz.ignore_basic_report],
> >     toZigbee: [],
> >     exposes: [e.occupancy(), e.battery_low(), e.linkquality(), e.battery(), e.battery_voltage()],
> >     configure: async (device, coordinatorEndpoint, logger) => {
> >         const endpoint = device.getEndpoint(1);
> >         await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
> >         await reporting.batteryPercentageRemaining(endpoint);
> >     },
> > };
> > 
> > module.exports = definition;
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 
> > * save this as file next to `configuration.yaml` as `ext_converter.js`
> > * add it to `configuration.yaml`:
> > 
> > ```yaml
> > external_converters:
> >   - ext_converter.js
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 
> > * start z2m, check if device works
> 
> I'm using latest Docker image. When this device(TZ3000_mg4dy6z6) will be supported by default? Of course, I already pulled latest arm7 image from official docker image. Thanks!

Koenkk, 2023-02-15:
@destman added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

dsz79, 2023-02-28:
Zigbee Model: TS0202
Zigbee Manufacturer: _TZ3000_6ygjfyll

z2mqtt runs inside a synology docker, tested with koenkk/zigbee2mqtt:latest-dev and koenkk/zigbee2mqtt images, and occupancy sensor is always on, any solution?

wr420, 2023-03-06:
HI - I have another variant of the Fantem - fingerprint: [{modelID: 'TS0202', manufacturerName: '_TZ3210_wuhzzfqg'}],
        model: 'ZB003-X',     vendor: 'Fantem',

I am able to get motion to work with this code

Koenkk, 2023-03-07:
`_TZ3210_wuhzzfqg` is supported in z2m 1.30.2

wr420, 2023-03-07:
Thanks, I was on edge and didn’t realize edge did not offer updates.

Thanks
Matt

The content of this email is confidential and intended for the recipient specified in message only. It is strictly forbidden to share any part of this message with any third party, without a written consent of the sender. If you received this message by mistake, please reply to this message and follow with its deletion, so that we can ensure such a mistake does not occur in the future.

alexvaltchev, 2023-03-27:
IT detects every 1 min 5 sec with ZHA and every 10-12 sec with Z2M. I have both running together and want to move everything to ZHA. Thanks

yundin2000, 2023-06-06:
_TZ3000_zk2xi5uz working with converter

Blacksli, 2023-06-11:
> 

Can you give a working converter for _TZ3000_zk2xi5uz?

Jurgen1980, 2023-08-25:
Can't get it to work with _TZ3000_o4mkahkc
Keeps saying it is unsupported

`
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{ modelID: 'TS0202', manufacturerName: '_TZ3000_o4mkahkc' }],
    model: 'TS0202_1',
    vendor: 'TuYa',
    description: 'Motion sensor',
    fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery, fz.ignore_basic_report, fz.ias_occupancy_alarm_1_report],
    toZigbee: [{
        key: ['keep_time'],
        convertSet: async (entity, key, value, meta) => {
            switch (key) {
                case 'keep_time':
                    await entity.write('ssIasZone', { 61441: { value: { '30': 0, '60': 1, '120': 2 }[value], type: 0x20 } });
                    break;
                default: // Unknown key
                    throw new Error(`Unhandled key ${key}`);
            }
        },
    }],
    exposes: [e.occupancy(), e.battery_low(), e.tamper(),
    exposes
        .enum("keep_time", ea.STATE_SET, ["30", "60", "120"])
        .withDescription("PIR keep time in seconds"),
    ],
};

module.exports = definition;
`
[image]

paulbaird87, 2023-08-28:
> 

Hi @Jurgen1980 could you please explain how I can implement this "external converter" I have never setup up anything like this before and am fairly new to zigbee2mqtt. I have the same model or sensor as you, the _TZ3000_o4mkahkc.

Cmajda, 2023-08-28:
Hi @paulbaird87 
1. Install visual studio code for HA [Install visual studio code](https://github.com/hassio-addons/addon-vscode/blob/main/vscode/DOCS.md)
2. in folder ./zigbee2mqtt create a file for example pir1.js (next to configuration.yaml) with content:
```
const fz = require("zigbee-herdsman-converters/converters/fromZigbee");
const tz = require("zigbee-herdsman-converters/converters/toZigbee");
const exposes = require("zigbee-herdsman-converters/lib/exposes");
const reporting = require("zigbee-herdsman-converters/lib/reporting");
const extend = require("zigbee-herdsman-converters/lib/extend");
const e = exposes.presets;
const ea = exposes.access;

const definition = {
  fingerprint: [{ modelID: "TS0202", manufacturerName: "_TZ3000_o4mkahkc" }],
  model: "TS0202_1",
  vendor: "TuYa",
  description: "Motion sensor",
  fromZigbee: [
    fz.ias_occupancy_alarm_1,
    fz.battery,
    fz.ignore_basic_report,
    fz.ias_occupancy_alarm_1_report,
  ],
  toZigbee: [
    {
      key: ["keep_time"],
      convertSet: async (entity, key, value, meta) => {
        switch (key) {
          case "keep_time":
            await entity.write("ssIasZone", {
              61441: { value: { 30: 0, 60: 1, 120: 2 }[value], type: 0x20 },
            });
            break;
          default: // Unknown key
            throw new Error("Unhandled key ${key}");
        }
      },
    },
  ],
  exposes: [
    e.occupancy(),
    e.battery_low(),
    e.tamper(),
    exposes
      .enum("keep_time", ea.STATE_SET, ["30", "60", "120"])
      .withDescription("PIR keep time in seconds"),
  ],
};
```
[image]

3. edit file ./zigbee2mqtt/configuration.yaml add this code:
```
external_converters:
  - pir1.js

```
[image]

4. go to zigbee2MQTT > Settings > tools > restart Zigbee2MQTT or Restart home assistant

Alexey-Tsarev, 2023-09-11:
Can someone help us with the following device?

TS0202
_TZ3000_zk2xi5uz

[image]

xeoMe015, 2024-11-25:
> > 
> 
> Hi @Jurgen1980 could you please explain how I can implement this "external converter" I have never setup up anything like this before and am fairly new to zigbee2mqtt. I have the same model or sensor as you, the _TZ3000_o4mkahkc.

Pretty sure you already have the knowledge and experience at this time but for future people seeing this message and having the same question:

Within zigbee2mqtt go to the "cogwheel" right next to the "Extentions" setting at the top of the screen and click settings.
There go to the tab "External converters" fill in "ext_converter.js" and press submit.
On the top right a red button will appear "Restart" click on it and restart zigbee2mqtt.

Done.

Koenkk, 2024-11-27:
Added the device for out of the box support! Will be supported in the 1 December release.